lwfinger/rtl8188eu #386: don't boot kernel 5.15 with this module

ptah-alexs, 2021-12-03:
Debian sid, kernel 5.15, module build from commit a69d6361ef0185aa7d2e4c774bc2de36fe83d81e
[messages.txt](https://github.com/lwfinger/rtl8188eu/files/7650966/messages.txt)
with 5.10 work fine

lwfinger, 2021-12-03:
I am unable to duplicate your problem running kernel 5.15.5 from openSUSE Tumbleweed. I get no messages and LibreSpeed throughput rates of roughly 30 Mbps up and down.

georgmu, 2022-01-14:
I am running into the same issue and I am trying to debug this.
Multiple processes hang on `rtnl_lock()`

lwfinger, 2022-01-14:
What branch are you running? That information is always needed if a repo has more than one.

georgmu, 2022-01-14:
Sorry for the missing info:
I am testing with **v5.2.2.4** because I need cfg80211 to run an AP with NetworkManager.

lwfinger, 2022-01-14:
I have learned a few things. It brole between v5.10 and v5.11. I am currently bisecting to see what changed in the kernel at that time. I hope that knowledge will help me find the problem with rtl8188eu. A similar problem was just reported for rtl8192ee using the Realtek driver. If all goes well, the same fix should apply to both.

georgmu, 2022-01-14:
Maybe [51d62f2f2c501a93d9a6a46f43731f984e227764](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=51d62f2f2c501a93d9a6a46f43731f984e227764)
This is the lock where I get stuck

lwfinger, 2022-01-15:
Reverting that commit does avoid the lockup. Now, I need to find the offending lock in the rtl8188eu code.

georgmu, 2022-01-15:
I will try to build a kernel with CONFIG_DEBUG_MUTEXES=y and rerun my tests

georgmu, 2022-01-15:
```
[  188.583739] ============================================
[  188.583746] WARNING: possible recursive locking detected
[  188.583754] 5.15.0-0.bpo.2.debug-amd64 #1 Tainted: G           OE    
[  188.583763] --------------------------------------------
[  188.583769] NetworkManager/531 is trying to acquire lock:
[  188.583777] ffff9ed7d57cff90 (pmutex){+.+.}-{3:3}, at: usbctrl_vendorreq+0xa1/0x502 [8188eu]
[  188.583891] 
               but task is already holding lock:
[  188.583898] ffff9ed7d57cf140 (pmutex){+.+.}-{3:3}, at: netdev_open+0x2f/0x84 [8188eu]
[  188.583999] 
               other info that might help us debug this:
[  188.584006]  Possible unsafe locking scenario:

[  188.584013]        CPU0
[  188.584017]        ----
[  188.584021]   lock(pmutex);
[  188.584027]   lock(pmutex);
[  188.584033] 
                *** DEADLOCK ***

[  188.584041]  May be due to missing lock nesting notation

[  188.584049] 2 locks held by NetworkManager/531:
[  188.584057]  #0: ffffffff97f2b510 (rtnl_mutex){+.+.}-{3:3}, at: rtnetlink_rcv_msg+0x140/0x520
[  188.584079]  #1: ffff9ed7d57cf140 (pmutex){+.+.}-{3:3}, at: netdev_open+0x2f/0x84 [8188eu]
[  188.584181] 
               stack backtrace:
[  188.584188] CPU: 0 PID: 531 Comm: NetworkManager Tainted: G           OE     5.15.0-0.bpo.2.debug-amd64 #1  Debian 5.15.5-2~bpo11+1.1
[  188.584204] Hardware name: Supermicro SYS-E50-9AP-N5-AG050/A2SAP-H, BIOS 1.2c 05/29/2019
[  188.584215] Call Trace:
[  188.584221]  <TASK>
[  188.584227]  dump_stack_lvl+0x57/0x72
[  188.584239]  __lock_acquire.cold+0xc3/0x2b4
[  188.584252]  ? desc_read_finalized_seq+0x2f/0x90
[  188.584263]  lock_acquire+0xc8/0x2d0
[  188.584273]  ? usbctrl_vendorreq+0xa1/0x502 [8188eu]
[  188.584368]  ? usbctrl_vendorreq+0xa1/0x502 [8188eu]
[  188.584460]  __mutex_lock+0x7c/0x870
[  188.584471]  ? usbctrl_vendorreq+0xa1/0x502 [8188eu]
[  188.584566]  usbctrl_vendorreq+0xa1/0x502 [8188eu]
[  188.584662]  ? _printk+0x58/0x6f
[  188.584671]  usb_write8+0x3c/0x5a [8188eu]
[  188.584758]  _rtw_write8+0x1b/0x26 [8188eu]
[  188.584827]  rtl8188eu_hal_init+0x12f/0xc0e [8188eu]
[  188.584913]  ? _printk+0x58/0x6f
[  188.584922]  rtw_hal_init+0x22/0xc7 [8188eu]
[  188.585015]  _netdev_open+0x78/0x2aa [8188eu]
[  188.585107]  netdev_open+0x37/0x84 [8188eu]
[  188.585199]  __dev_open+0xe7/0x1b0
[  188.585209]  __dev_change_flags+0x1ec/0x270
[  188.585220]  dev_change_flags+0x21/0x60
[  188.585229]  do_setlink+0x285/0x1030
[  188.585240]  ? __lock_acquire+0x3df/0x1e70
[  188.585249]  ? __nla_validate_parse+0x5f/0xc00
[  188.585264]  __rtnl_newlink+0x5f1/0x9b0
[  188.585276]  ? lock_acquire+0xc8/0x2d0
[  188.585285]  ? sock_def_readable+0x5/0x2a0
[  188.585294]  ? lock_is_held_type+0xaa/0x120
[  188.585304]  ? find_held_lock+0x32/0x90
[  188.585313]  ? sched_clock_cpu+0x9/0xb0
[  188.585322]  ? lock_release+0x296/0x400
[  188.585335]  ? __lock_acquire+0x3df/0x1e70
[  188.585356]  ? rcu_read_lock_sched_held+0x3f/0x80
[  188.585369]  ? kmem_cache_alloc_trace+0x274/0x300
[  188.585381]  rtnl_newlink+0x44/0x70
[  188.585389]  rtnetlink_rcv_msg+0x169/0x520
[  188.585399]  ? find_held_lock+0x32/0x90
[  188.585407]  ? sched_clock_cpu+0x9/0xb0
[  188.585416]  ? rtnl_newlink+0x70/0x70
[  188.585424]  netlink_rcv_skb+0x4e/0xf0
[  188.585436]  netlink_unicast+0x18e/0x250
[  188.585446]  netlink_sendmsg+0x245/0x480
[  188.585458]  sock_sendmsg+0x5b/0x60
[  188.585467]  ____sys_sendmsg+0x231/0x270
[  188.585476]  ? import_iovec+0x2d/0x30
[  188.585485]  ? sendmsg_copy_msghdr+0x7c/0xa0
[  188.585496]  ___sys_sendmsg+0x75/0xb0
[  188.585506]  ? lock_acquire+0xc8/0x2d0
[  188.585515]  ? lock_is_held_type+0xaa/0x120
[  188.585524]  ? find_held_lock+0x32/0x90
[  188.585534]  ? lock_release+0x296/0x400
[  188.585544]  ? __fget_files+0x166/0x200
[  188.585556]  __sys_sendmsg+0x59/0xa0
[  188.585567]  ? lockdep_hardirqs_on_prepare+0xff/0x180
[  188.585578]  do_syscall_64+0x38/0xc0
[  188.585588]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[  188.585599] RIP: 0033:0x7f4a87986e4d
[  188.585608] Code: 28 89 54 24 1c 48 89 74 24 10 89 7c 24 08 e8 ca ee ff ff 8b 54 24 1c 48 8b 74 24 10 41 89 c0 8b 7c 24 08 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 33 44 89 c7 48 89 44 24 08 e8 fe ee ff ff 48
[  188.585629] RSP: 002b:00007ffe4eb2ab40 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
[  188.585643] RAX: ffffffffffffffda RBX: 000055e5af51b480 RCX: 00007f4a87986e4d
[  188.585652] RDX: 0000000000000000 RSI: 00007ffe4eb2ab90 RDI: 000000000000000c
[  188.585661] RBP: 00007ffe4eb2ab90 R08: 0000000000000000 R09: 0000000000000000
[  188.585670] R10: 0000000000000000 R11: 0000000000000293 R12: 000055e5af51b480
[  188.585679] R13: 00007ffe4eb2ad38 R14: 00007ffe4eb2ad2c R15: 0000000000000000
[  188.585693]  </TASK>
```

georgmu, 2022-01-15:
I have replaced the mutex wrapper functions in `osdep_service.h` and `osdep_service_linux.h` to get a better output of the lock debug stuff:

```
#define _rtw_mutex_init(x) mutex_init(x)
#define _rtw_mutex_free(x) mutex_destroy(x)

#define _enter_critical_mutex(x,y) mutex_lock(x)
#define _exit_critical_mutex(x,y) mutex_unlock(x)
```

Now it gives the following output:

```
[  132.861245] ============================================
[  132.861252] WARNING: possible recursive locking detected
[  132.861259] 5.15.0-0.bpo.2.debug-amd64 #1 Tainted: G           OE    
[  132.861268] --------------------------------------------
[  132.861275] NetworkManager/528 is trying to acquire lock:
[  132.861283] ffffffff93d2b510 (rtnl_mutex){+.+.}-{3:3}, at: wiphy_apply_custom_regulatory+0xa0/0x110 [cfg80211]
[  132.861386] 
               but task is already holding lock:
[  132.861393] ffffffff93d2b510 (rtnl_mutex){+.+.}-{3:3}, at: rtnetlink_rcv_msg+0x140/0x520
[  132.861413] 
               other info that might help us debug this:
[  132.861421]  Possible unsafe locking scenario:

[  132.861428]        CPU0
[  132.861432]        ----
[  132.861436]   lock(rtnl_mutex);
[  132.861442]   lock(rtnl_mutex);
[  132.861449] 
                *** DEADLOCK ***

[  132.861457]  May be due to missing lock nesting notation

[  132.861465] 2 locks held by NetworkManager/528:
[  132.861472]  #0: ffffffff93d2b510 (rtnl_mutex){+.+.}-{3:3}, at: rtnetlink_rcv_msg+0x140/0x520
[  132.861490]  #1: ffff88b0983f4140 (&pdvobj->hw_init_mutex){+.+.}-{3:3}, at: netdev_open+0x2f/0x84 [8188eu]
[  132.861600] 
               stack backtrace:
[  132.861608] CPU: 0 PID: 528 Comm: NetworkManager Tainted: G           OE     5.15.0-0.bpo.2.debug-amd64 #1  Debian 5.15.5-2~bpo11+1.1
[  132.861624] Hardware name: Supermicro SYS-E50-9AP-N5-AG050/A2SAP-H, BIOS 1.2c 05/29/2019
[  132.861635] Call Trace:
[  132.861642]  <TASK>
[  132.861648]  dump_stack_lvl+0x57/0x72
[  132.861660]  __lock_acquire.cold+0xc3/0x2b4
[  132.861674]  lock_acquire+0xc8/0x2d0
[  132.861686]  ? wiphy_apply_custom_regulatory+0xa0/0x110 [cfg80211]
[  132.861761]  ? lock_acquire+0xc8/0x2d0
[  132.861770]  ? fs_reclaim_acquire+0x4d/0xd0
[  132.861781]  ? lock_is_held_type+0xaa/0x120
[  132.861792]  ? wiphy_apply_custom_regulatory+0xa0/0x110 [cfg80211]
[  132.861867]  __mutex_lock+0x7c/0x870
[  132.861876]  ? wiphy_apply_custom_regulatory+0xa0/0x110 [cfg80211]
[  132.861951]  ? rcu_read_lock_sched_held+0x3f/0x80
[  132.861962]  ? __kmalloc+0x291/0x350
[  132.861973]  wiphy_apply_custom_regulatory+0xa0/0x110 [cfg80211]
[  132.862053]  rtw_regd_init+0x3c/0x4b [8188eu]
[  132.862147]  rtw_cfg80211_init_wiphy+0x22c/0x29a [8188eu]
[  132.862244]  _netdev_open+0xbc/0x2aa [8188eu]
[  132.862338]  netdev_open+0x37/0x84 [8188eu]
[  132.862430]  __dev_open+0xe7/0x1b0
[  132.862440]  __dev_change_flags+0x1ec/0x270
[  132.862451]  dev_change_flags+0x21/0x60
[  132.862460]  do_setlink+0x285/0x1030
[  132.862471]  ? __lock_acquire+0x3df/0x1e70
[  132.862480]  ? __nla_validate_parse+0x5f/0xc00
[  132.862495]  __rtnl_newlink+0x5f1/0x9b0
[  132.862508]  ? lock_acquire+0xc8/0x2d0
[  132.862517]  ? sock_def_readable+0x5/0x2a0
[  132.862526]  ? lock_is_held_type+0xaa/0x120
[  132.862535]  ? find_held_lock+0x32/0x90
[  132.862544]  ? sched_clock_cpu+0x9/0xb0
[  132.862555]  ? lock_release+0x296/0x400
[  132.862567]  ? __lock_acquire+0x3df/0x1e70
[  132.862589]  ? rcu_read_lock_sched_held+0x3f/0x80
[  132.862599]  ? kmem_cache_alloc_trace+0x274/0x300
[  132.862610]  rtnl_newlink+0x44/0x70
[  132.862618]  rtnetlink_rcv_msg+0x169/0x520
[  132.862628]  ? find_held_lock+0x32/0x90
[  132.862637]  ? sched_clock_cpu+0x9/0xb0
[  132.862645]  ? rtnl_newlink+0x70/0x70
[  132.862653]  netlink_rcv_skb+0x4e/0xf0
[  132.862666]  netlink_unicast+0x18e/0x250
[  132.862676]  netlink_sendmsg+0x245/0x480
[  132.862687]  sock_sendmsg+0x5b/0x60
[  132.862697]  ____sys_sendmsg+0x231/0x270
[  132.862706]  ? import_iovec+0x2d/0x30
[  132.862714]  ? sendmsg_copy_msghdr+0x7c/0xa0
[  132.862725]  ___sys_sendmsg+0x75/0xb0
[  132.862736]  ? lock_acquire+0xc8/0x2d0
[  132.862745]  ? lock_is_held_type+0xaa/0x120
[  132.862754]  ? find_held_lock+0x32/0x90
[  132.862763]  ? lock_release+0x296/0x400
[  132.862774]  ? __fget_files+0x166/0x200
[  132.862786]  __sys_sendmsg+0x59/0xa0
[  132.862796]  ? lockdep_hardirqs_on_prepare+0xff/0x180
[  132.862808]  do_syscall_64+0x38/0xc0
[  132.862816]  entry_SYSCALL_64_after_hwframe+0x44/0xae
[  132.862828] RIP: 0033:0x7f2a345aae4d
[  132.862838] Code: 28 89 54 24 1c 48 89 74 24 10 89 7c 24 08 e8 ca ee ff ff 8b 54 24 1c 48 8b 74 24 10 41 89 c0 8b 7c 24 08 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 33 44 89 c7 48 89 44 24 08 e8 fe ee ff ff 48
[  132.862859] RSP: 002b:00007ffed4678f50 EFLAGS: 00000293 ORIG_RAX: 000000000000002e
[  132.862872] RAX: ffffffffffffffda RBX: 0000564f3586d480 RCX: 00007f2a345aae4d
[  132.862881] RDX: 0000000000000000 RSI: 00007ffed4678fa0 RDI: 000000000000000c
[  132.862890] RBP: 00007ffed4678fa0 R08: 0000000000000000 R09: 0000000000000000
[  132.862899] R10: 0000000000000000 R11: 0000000000000293 R12: 0000564f3586d480
[  132.862908] R13: 00007ffed4679148 R14: 00007ffed467913c R15: 0000000000000000
[  132.862922]  </TASK>
```

georgmu, 2022-01-15:
I am trying to use a solution as suggested here:
https://lore.kernel.org/all/768b8297801ed3987d97efd6d55ed6a9ad1c7598.camel@sipsolutions.net/
https://p.sipsolutions.net/235c352b8ae5db88.txt

georgmu, 2022-01-15:
The following diff fixes the recursive mutex deadlock for me:
```
diff --git a/ioctl_cfg80211.c b/ioctl_cfg80211.c
index 1d7ff70..0e629a2 100644
--- a/ioctl_cfg80211.c
+++ b/ioctl_cfg80211.c
@@ -6514,9 +6514,6 @@ void rtw_cfg80211_init_wiphy(_adapter *padapter)
                        rtw_cfg80211_init_ht_capab(padapter, &bands->ht_cap, NL80211_BAND_5GHZ, rf_type);
        }
 #endif
-       /* init regulary domain */
-       rtw_regd_init(padapter);
-
        /* copy mac_addr to wiphy */
        memcpy(wiphy->perm_addr, adapter_mac_addr(padapter), ETH_ALEN);
 
@@ -6928,6 +6925,9 @@ int rtw_wdev_alloc(_adapter *padapter, struct wiphy *wiphy)
        ATOMIC_SET(&pwdev_priv->switch_ch_to, 1);
 #endif
 
+       /* init regulary domain */
+       rtw_regd_init(padapter);
+
 exit:
        return ret;
 }
```

georgmu, 2022-01-15:
> Debian sid, kernel 5.15, module build from commit [a69d636](https://github.com/lwfinger/rtl8188eu/commit/a69d6361ef0185aa7d2e4c774bc2de36fe83d81e) wifi stick TP-LINK TL-WN727N [messages.txt](https://github.com/lwfinger/rtl8188eu/files/7650966/messages.txt) with 5.10 work fine

the message.txt also references branch v5.2.2.4, not master, so this is the same issue

lwfinger, 2022-01-15:
Thanks for chasing this down. Strangely, I thought that lockdep testing was enabled in my kernel. Wonder how it got turned off. Last night, I decided to see if the regulatory domain registration could be moved, and when I got up, the solution was waiting in my mailbox.
Your fix is applied and pushed. As you will see in the commit message, you get credit for the fix.

closed 2022-01-15 by lwfinger

georgmu, 2022-01-15:
Good to have it fixed. Thanks for all your work!