lwfinger/rtl8188eu #388: Help ?

Suehta732, 2021-12-10:
Is it normal is showing 0bda 0179 ?? (It is usually 0bda f179) I'm using this tutorial, it says 0bda 0179 works:https://linuxforums.org.uk/index.php?topic=11261.0
[image]
[image]
[image]

Suehta732, 2021-12-10:
Also wanted to ask if monitor mode works?? I assume that being cheap doesn't work does it ?? it sucks to be poor. The funny thing is that the other ones that work are the size of my cock, very small, and this big one doesn't work. Ps: I've seen the same error on the other guy over there, but updating doesn't work (make), will it be because I'm out of memory to blame ??

Suehta732, 2021-12-10:
[image]
This is my wifi adapter. I'm on my cell phone anyway.

Suehta732, 2021-12-10:
[image]

Suehta732, 2021-12-10:
https://user-images.githubusercontent.com/95923470/145586916-4385dd50-08bf-47cb-8eb9-9ceb70abdda4.jpg

lwfinger, 2021-12-10:
The 8188etv (0bda:0179) is in the tables. Your first problem is that you have not installed the kernel firmware. In addition, I am not sure that the driver will build on kernels as old as 3.18! Monitor mode will likely not work.

Suehta732, 2021-12-10:
> The 8188etv (0bda:0179) is in the tables. Your first problem is that you have not installed the kernel firmware. In addition, I am not sure that the driver will build on kernels as old as 3.18! Monitor mode will likely not work.

So my way out is to give up. If you can recommend a rom that makes it all automatic ?? Thanks for answering.

Suehta732, 2021-12-10:
My problem is so unique that even the person who made the code responded, wow. Well I think I have an old Samsung cell phone that already has everything I need. I'll wait until someone answers and close. As this problem is unique, I think that no one will ever have it.

Suehta732, 2021-12-11:
To solve this problem, you need to have an old Samsung cell phone and that's it.

Suehta732, 2021-12-11:
To solve this problem, just have an old Samsung cell phone and that's it. Old Samsung cell phones have monitor mode.

closed 2021-12-11 by Suehta732