lwfinger/rtl8188eu #390: Don't work with TP-Link TL-WN727N adapter

Arsoban, 2021-12-16:
Hello! I'm beginner at linux. I installed Manjaro Linux, but unfortunately my pc doesn't have embedded wifi. On windows i installed driver from TP-Link. But they drivers for linux are outdated. And i found this repository, but driver doesn't work. When i'm connecting to the net, it say something like "This net haven't internet", or infinity connecting(no result after installing driver, the same situation was before installation). This driver is compatible with TP-Link TL-WN 727N v5.20? If yes, what i need to do? Sorry about my bad english, this text may contain errors.

lwfinger, 2021-12-16:
My search says that the USB ID for that device is 0x2357:0x0111. Can you confirm that?

As you are new to Linux, what version of Manjaro are you using, and what kernel is it running? A driver for this device has been in the kernel since v5.5, thus you probably do not need to use this external driver.

If you did install this driver, then the in-kernel version and the external one are likely conflicting. Please post the output of the command

lsmod | grep 88

Arsoban, 2021-12-16:
Hello again. I'm logged in from linux, and i have internet connection. It was happen 2 times, but internet connection was not stable, and disconnected very soon. I'm using manjaro 21.2.0. Kernel 5.13(i installed manjaro when this kernel was actual).
USB ID is `2357:0111`, you're right(i checked it with `lsusb` command, correct me if i'm wrong).
[image]
I think this connection will disconnect very soon.

lwfinger, 2021-12-16:
You do not need to post a photo to verify that the USB ID matches the one I stated.

When it disconnects, it should leave a record in the output of dmesg. What is it. Note: If you post the entire dmesg output, I will ignore it. I just want the part concerning the disconnect.

Arsoban, 2021-12-17:
Hello, connection didn't disconnect yesterday, but today when i've logged in from linux, and i haven't internet connection.
Now i can't connect, and send disconnection logs. But there are logs while connecting:
```
[   16.611488] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[   16.634185] usbcore: registered new interface driver r8188eu
[   16.651198] r8188eu 1-4:1.0 wlp0s20f0u4: renamed from wlan0
[   22.520603] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[   22.878647] R8188EU: INFO indicate disassoc
[   24.812432] R8188EU: INFO indicate disassoc
[   43.009424] R8188EU: INFO assoc success
[   45.998782] R8188EU: INFO ap recv disassoc reason code(2) sta:98:de:d0:f7:58:7e
[   46.001324] R8188EU: INFO indicate disassoc
[   56.433443] R8188EU: INFO indicate disassoc
[   56.872870] R8188EU: INFO ap recv disassoc reason code(2) sta:98:de:d0:f7:58:7e
[   60.943458] R8188EU: INFO indicate disassoc
[   68.356819] R8188EU: INFO indicate disassoc
[   68.788804] R8188EU: INFO ap recv disassoc reason code(2) sta:98:de:d0:f7:58:7e
[   75.076771] R8188EU: INFO indicate disassoc
[   95.260101] R8188EU: INFO indicate disassoc
[  109.473438] R8188EU: INFO indicate disassoc
[  125.660147] R8188EU: INFO indicate disassoc
[  140.246812] R8188EU: INFO indicate disassoc
[  151.766751] R8188EU: INFO indicate disassoc
[  165.530094] R8188EU: INFO indicate disassoc
[  176.353484] R8188EU: INFO indicate disassoc
[  192.836752] R8188EU: INFO indicate disassoc
[  255.606750] R8188EU: INFO indicate disassoc
[  272.806772] R8188EU: INFO indicate disassoc
[  279.366791] R8188EU: INFO indicate disassoc
[  293.606779] R8188EU: INFO indicate disassoc
[  303.400654] R8188EU: INFO assoc success
[  351.555193] R8188EU: INFO sta recv deauth reason code(6) sta:98:de:d0:f7:58:7e, ignore = 0
```
There are some things below, but its repeating, i think you don't need that.
I think i posted the part that you need, i got this from command `sudo dmesg | grep 8188`, correct me if i'm wrong.

lwfinger, 2021-12-17:
That was correct. Reason 2 for disconnect means "Previous authentication no longer valid". I have no idea how that happens.

How close to your AP is the computer? Is it in the same room?

To get the signal strengths, run and post the output of

sudo iw dev wlp0s20f0u4 scan | egrep "signal|ssid"

In the output, indicate which AP you are using.

Arsoban, 2021-12-17:
When i logged in from linux, i haven't internet. I pasted command, and i have "No such device" error (-19). And after 2-3 minutes. It successfully connected, but internet is very slow. When i'm go on github, it says something like "Can't find github.com ip". When i'm hovering on network widget, it saying "This connection is limited". In dmesg when disconnect is appearing it logs this string
```
INFO sta recv deauth reason code(6) sta:98:de:d0:f7:58:7e, ignore = 0
```