lwfinger/rtl8188eu #393: RTW: rtw_sta_mstatus_disc_rpt(wlan0) invalid macid:255

jimmyw, 2022-01-05:
version v5.2.2.4 kernel 5.4.169

```
RTNETLINK answers: Cannot assign requested address
Flask starting
STATE: start -> t1 -> show_state
STATE: show_state -> net_has_internet -> show_state_has_internet
STATE: show_state_has_internet -> bt_long_press -> setup_mode_start
Configuration file: /tmp/hostapd.conf
[39510.320922] RTW: rtw_sta_mstatus_disc_rpt(wlan0) invalid macid:255
[39510.328415] ------------[ cut here ]------------
Using interface wlan0 with hwaddr 5c:f3:70:5b:28:61 and ssid "Wa[39510.333834] WARNING: CPU: 3 PID: 897 at /usr/src/debug/rtl8188eu/git-r0/git/rtw_mlme.c:2330 rtw_sta_mstatus_disc_rpt+0x60/0xdc [8188eu]
ttyBox-16554153080895d6"
[39510.351664] Modules linked in: ip6t_REJECT ipt_REJECT ip6table_filter ip6_tables xt_MASQUERADE nfnetlink br_netfilter xt_owner 8188eu(O) sha256_generic libsha256 cfg80211 smsc95xx usbnet mii rfkill spidev sun6i_dma_spi(O) nls_cp437 vfat fat
[39510.375282] CPU: 3 PID: 897 Comm: RTW_CMD_THREAD Tainted: G     U  W  O      5.4.169 #1
[39510.383299] Hardware name: Allwinner sun8i Family
[39510.388057] [<c010e4b8>] (unwind_backtrace) from [<c010b5f4>] (show_stack+0x10/0x14)
[39510.395821] [<c010b5f4>] (show_stack) from [<c07df720>] (dump_stack+0x84/0x98)
[39510.403067] [<c07df720>] (dump_stack) from [<c07da234>] (__warn+0xbc/0xd8)
[39510.409964] [<c07da234>] (__warn) from [<c07da2b4>] (warn_slowpath_fmt+0x64/0xc8)
[39510.418057] [<c07da2b4>] (warn_slowpath_fmt) from [<bf0d3b10>] (rtw_sta_mstatus_disc_rpt+0x60/0xdc [8188eu])
[39510.428543] [<bf0d3b10>] (rtw_sta_mstatus_disc_rpt [8188eu]) from [<bf0d8810>] (rtw_stadel_event_callback+0x38/0x394 [8188eu])
[39510.441025] [<bf0d8810>] (rtw_stadel_event_callback [8188eu]) from [<bf0eadd8>] (mlme_evt_hdl+0x78/0x90 [8188eu])
[39510.452100] [<bf0eadd8>] (mlme_evt_hdl [8188eu]) from [<bf0bc57c>] (rtw_cmd_thread+0x488/0x580 [8188eu])
[39510.462054] [<bf0bc57c>] (rtw_cmd_thread [8188eu]) from [<c013c588>] (kthread+0x174/0x18c)
[39510.470350] [<c013c588>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[39510.477576] Exception stack(0xecca1fb0 to 0xecca1ff8)
[39510.482640] 1fa0:                                     00000000 00000000 00000000 00000000
[39510.490822] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[39510.499001] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[39510.506591] ---[ end trace cb8ac88edaaad2f9 ]---
[39510.529945] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: STA 70:56:81:c0:53:49 IEEE 802.11: disassociated
STATE: setup_mode_start -> net_has_internet -> setup_mode
wlan0: STA 70:56:81:c0:53:49 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 70:56:81:c0:53:49
wlan0: STA 70:56:81:c0:53:49 RADIUS: starting accounting session 13453B8B0D7C3074
```

rlelchev, 2022-01-07:
It seems, that it successfully connects you to the internet. What happens if you run "iwconfig" from wireless-tools?

jimmyw, 2022-01-07:
Everything seems to work, i have no idea of what's going on and why there is a stacktrace.

lwfinger, 2022-01-09:
Your system has failed the test "if (mac_id >= 0 && mac_id < macid_ctl->num)", which triggers a warning. I have no idea what that means other than warnings are usually not too serious.