lwfinger/rtl8188eu #394: Make install fails on Armbian, insmod need to be used.

2010019970909, 2022-01-07:
Hello,

I built an Armbian minimal image (`Linux pcduino3 5.15.13-sunxi #trunk.0004 SMP Wed Jan 5 17:53:11 UTC 2022 armv7l GNU/Linux`), but the Wi-Fi driver for my board (pcduino3) is not working properly (it fails when trying to connect).

The Wi-Fi chip seems to be the “RTL8188EUS”.
```bash
root@pcduino3:~# lsusb | grep RTL
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```

So, I installed the header (https://forum.armbian.com/topic/8141-building-kernel-modul-install-kernel-headers/?do=findComment&comment=61388) and after a while I found this repo.

I then proceeded to build and install the driver:
```bash
root@pcduino3:~# git clone https://github.com/lwfinger/rtl8188eu.git rtl8188eu
[...]
root@pcduino3:~# cd rtl8188eu
root@pcduino3:~/rtl8188eu# make all
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.13-sunxi/build M=/root/lwfinger  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.13-sunxi'
[...]
make[1]: Leaving directory '/usr/src/linux-headers-5.15.13-sunxi'
```

## The issue
```bash
root@pcduino3:~/rtl8188eu# make install
install -p -m 644 8188eu.ko  /lib/modules/5.15.13-sunxi/kernel/drivers/net/wireless
install: cannot create regular file '/lib/modules/5.15.13-sunxi/kernel/drivers/net/wireless': No such file or directory
make: *** [Makefile:161: install] Error 1
```

Make install is not working properly. I tried creating the directory, but I get the following messages:
```bash
[...]
depmod: WARNING: could not open modules.builtin at /lib/modules/5.15
depmod: WARNING: could not open modules.order at /lib/modules/5.15
[...]
```

## The solution
I then looked at Armbian's documentation (https://docs.armbian.com/User-Guide_Advanced-Features/#how-to-build-a-wireless-driver). I was able to install the driver using:
```bash
root@pcduino3:~/rtl8188eu# insmod 8188eu.ko
```
Using `lsmod`, I confirmed that the module was loaded and `ip a` showed the wireless adapter.

The last step was to connect the adapter:
```bash
nmcli device wifi connect <ssid> password <password>
```

rlelchev, 2022-01-07:
It seems, that your kernel sources do not support wireless connections. You enabled it as a module, but you better get a kernel, that does support it.

2010019970909, 2022-01-07:
Yes, I just realized that the installation was not persistent when using `insmod`… I think the installation in the Makefile could start with  `mkdir -p $(MODDESTDIR)` (which seems to solve the issue).

lwfinger, 2022-01-10:
Yes, the Makefile should have the mkdir command. It is only needed when the kernel does not support any other wireless devices. For distro-generated kernels, this will never be the case, which is why this bug was not discovered before.

One thing: You should NEVER clone or build any external driver while logged in as root. At one point, the Makefile for the kernel had a bug that was harmless for ordinary users, but it destroyed /dev/NULL and crashed the system if you did the build as root. There are not likely to be this kind of error for most Makefiles, but be safe. You will need to have privilege for commands that write files into protected areas of the system. That is what "sudo" is for. In this case "sudo make install", "sudo insmod <fname>" or :sudo modprobe <driver_name>".

closed 2022-01-10 by lwfinger