lwfinger/rtl8188eu #399: 8188eu.ko often freezes my linux for ~1 sec

neva-blyad, 2022-02-05:
Hello,

When I connect and start using Wi-Fi, system lags.
I work normally ~5 secs, then LEDs on my dongle blink and everything freezes for ~1 sec.
I get new kernel message in dmesg after that:
`[22193.014524] r8188eu 5-1:1.0: firmware: direct-loading firmware rtlwifi/rtl8188eufw.bin`
Then everything occurs again: LEDs blink, 1 sec freeze and that message with the firmware reloading.

Do you have any thoughts? Why firmware is constantly reloading?

**Some information**
My device is TP-Link TL-WN722N,
my driver is 8188eu.ko from official kernel repository,
my kernel is custom based on Debian testing:
`Linux laptop 5.9.1-custom-3-amd64 #3 SMP neva_blyad 5.9.1 (2020-11-15 19:34:48) x86_64 GNU/Linux`

lwfinger, 2022-02-05:
This problem has not previously been reported.

You say your driver is from the official kernel sources, but the name you give, 8188eu.ko, is from here. We carefully name the drivers such that 8188eu comes from GitHub, and r8188eu comes from the kernel driver. Older kernels will have rtl8188eu. Is your system loading both drivers? What does 'lsmod | grep 8188' show?

neva-blyad, 2022-02-06:
**Modules which are loaded:**
`lsmod | grep 8188`
`r8188eu               471040  0`
`lib80211               16384  1 r8188eu`
`cfg80211              774144  3 rtlwifi,mac80211,r8188eu`
`usbcore               278528  11 xhci_hcd,ohci_hcd,ehci_pci,usb_storage,uvcvideo,ehci_hcd,btusb,xhci_pci,ohci_pci,uas,r8188eu`

**It is located at:**
`cd /lib/modules/5.9.1-custom-3-amd64/`
`find ./ -name '*8188*'`
`./drivers/staging/rtl8188eu`
`./drivers/staging/rtl8188eu/r8188eu.ko`
`./drivers/net/wireless/realtek/rtlwifi/rtl8188ee`
`./drivers/net/wireless/realtek/rtlwifi/rtl8188ee/rtl8188ee.ko`

**And what I have built:**
`cat /boot/config-5.9.1-custom-3-amd64 | grep 8188`
`CONFIG_RTL8188EE=m`
`CONFIG_R8188EU=m`

And this is information about my driver:
`sudo modinfo r8188eu`
`filename:       /lib/modules/5.9.1-custom-3-amd64/kernel/drivers/staging/rtl8188eu/r8188eu.ko`
`firmware:       rtlwifi/rtl8188eufw.bin`
`version:        v4.1.4_6773.20130222`
`author:         Realtek Semiconductor Corp.`
`description:    Realtek Wireless Lan Driver`
`license:        GPL`
`srcversion:     95B07E454EC3FFD5E701219`
`alias:          usb:v0BDApFFEFd*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v0DF6p0076d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v2C4Ep0102d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v2357p0111d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v2001p331Bd*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v2001p3311d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v2001p3310d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v2001p330Fd*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v0B05p18F0d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v056Ep4008d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*`
`alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*`
`depends:        cfg80211,usbcore,lib80211`
`staging:        Y`
`retpoline:      Y`
`intree:         Y`
`name:           r8188eu`
`vermagic:       5.9.1-custom-3-amd64 SMP mod_unload modversions `
`parm:           rtw_ips_mode:The default IPS mode (int)`
`parm:           ifname:The default name to allocate for first interface (charp)`
`parm:           if2name:The default name to allocate for second interface (charp)`
`parm:           rtw_initmac:charp`
`parm:           rtw_channel_plan:int`
`parm:           rtw_channel:int`
`parm:           rtw_wmm_enable:int`
`parm:           rtw_vrtl_carrier_sense:int`
`parm:           rtw_vcs_type:int`
`parm:           rtw_ht_enable:int`
`parm:           rtw_cbw40_enable:int`
`parm:           rtw_ampdu_enable:int`
`parm:           rtw_rx_stbc:int`
`parm:           rtw_ampdu_amsdu:int`
`parm:           rtw_power_mgnt:int`
`parm:           rtw_smart_ps:int`
`parm:           rtw_wifi_spec:int`
`parm:           rtw_antdiv_cfg:int`
`parm:           rtw_antdiv_type:int`
`parm:           rtw_enusbss:int`
`parm:           rtw_hwpdn_mode:int`
`parm:           rtw_max_roaming_times:The max roaming times to try (uint)`
`parm:           rtw_fw_iol:FW IOL (int)`
`parm:           rtw_mc2u_disable:int`
`parm:           rtw_80211d:Enable 802.11d mechanism (int)`
`parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)`
`parm:           debug:Set debug level (1-9) (default 1) (int)`
`parm:           monitor_enable:Enable monitor interface (default: false) (bool)`

neva-blyad, 2022-02-06:
Also I have built the driver 8188eu.ko from last commit of master branch of your GitHub repo. Version and other modinfo information of this driver is the same as for my kernel driver. So they are the same.

Also the bug occurs with the GitHub driver too.

lwfinger, 2022-02-06:
There are more differences than that. Please try the v5.2.2.4 branch. Do the following:
git checkout v5.2.2.4
git pull
make
sudo modprobe -rv r8188eu
sudo modprobe -rv 8188eu
sudo make install
sudo modprobe -v 8188eu

You will likely get an error on one or both of the 'modprobe -rv' commands. Ignore that.

neva-blyad, 2022-02-08:
Sorry for delay with the testing and answering.
I've built the 5.2.2.4-branch kernel module and blacklisted the old one.
And new driver works without lags!
I think 5.2.2.4-branch should be merged with master of your repo and with official Linux kernel.
Thank you man!
I close this issue.

closed 2022-02-08 by neva-blyad