lwfinger/rtl8188eu #401: Hello I'm using archlinux

0x3n0, 2022-04-01:
I did yay -S 8188eu-dkms and getting errors on the latest kernel

```plaintext
==> dkms install --no-depmod 8188eu/v4.3.0.8_13968 -k 5.17.1-arch1-1
Error! Bad return status for module build on kernel: 5.17.1-arch1-1 (x86_64)
Consult /var/lib/dkms/8188eu/v4.3.0.8_13968/build/make.log for more information.
==> WARNING: `dkms install --no-depmod 8188eu/v4.3.0.8_13968 -k 5.17.1-arch1-1' exited 10
```

lwfinger, 2022-04-01:
Please attach /var/lib/dkms/8188eu/v4.3.0.8_13968/build/make.log  to this issue so that I can see what happened.

0x3n0, 2022-04-02:
```plaintext
DKMS make.log for 8188eu-v4.3.0.8_13968 for kernel 5.17.1-arch1-1 (x86_64)
Sat Apr  2 06:00:43 PM WIT 2022
make: Entering directory '/usr/lib/modules/5.17.1-arch1-1/build'
  CC [M]  /var/lib/dkms/8188eu/v4.3.0.8_13968/build/core/rtw_cmd.o
In file included from /var/lib/dkms/8188eu/v4.3.0.8_13968/build/include/osdep_service.h:41,
                 from /var/lib/dkms/8188eu/v4.3.0.8_13968/build/include/drv_types.h:32,
                 from /var/lib/dkms/8188eu/v4.3.0.8_13968/build/core/rtw_cmd.c:22:
/var/lib/dkms/8188eu/v4.3.0.8_13968/build/core/rtw_cmd.c: In function ‘rtw_cmd_thread’:
/var/lib/dkms/8188eu/v4.3.0.8_13968/build/include/osdep_service_linux.h:156:31: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
  156 |         #define thread_exit() complete_and_exit(NULL, 0)
      |                               ^~~~~~~~~~~~~~~~~
/var/lib/dkms/8188eu/v4.3.0.8_13968/build/core/rtw_cmd.c:637:9: note: in expansion of macro ‘thread_exit’
  637 |         thread_exit();
      |         ^~~~~~~~~~~
/var/lib/dkms/8188eu/v4.3.0.8_13968/build/core/rtw_cmd.c:639:1: error: control reaches end of non-void function [-Werror=return-type]
  639 | }
      | ^
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:288: /var/lib/dkms/8188eu/v4.3.0.8_13968/build/core/rtw_cmd.o] Error 1
make: *** [Makefile:1831: /var/lib/dkms/8188eu/v4.3.0.8_13968/build] Error 2
make: Leaving directory '/usr/lib/modules/5.17.1-arch1-1/build'
```