lwfinger/rtl8188eu #403: Concurrent Mode

LegBRe4ker, 2022-04-08:
Hello,
I am trying unsuccessfully so far to compile the driver with concurrent mode enabled.

I checked out branch v5.2.2.4 and tried to add
"EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE"
to the Makefile.

Also I tried in "autoconf.h":
- /* #define CONFIG_CONCURRENT_MODE */
+ #define CONFIG_CONCURRENT_MODE
	#ifdef CONFIG_CONCURRENT_MODE
		#define CONFIG_RUNTIME_PORT_SWITCH

		#define CONFIG_SCAN_BACKOP
		#define CONFIG_TSF_RESET_OFFLOAD /* For 2 PORT TSF SYNC. */
	#endif

Unfortunately both did not work. What am I doing wrong?
I am running Debian 11 with kernel 5.10.0-13-amd64.

Thanks and regards

lwfinger, 2022-04-08:
What do you mean they did not work? I built the driver with CONFIG_CONCURRENT_MODE enabled. It created two wireless devices, and each could connect to my AP, but not at the same time. My understanding is that mode is for one to be an AP and the other an STA. The STA device would be for the AP's link to the Internet.

LegBreaker, 2022-04-09:
Thank you for your reply. 
Unfortunately I get only one interface. The 1st interface should be operated in station mode and the other as AP. So exactly as you have described. 

Is #define CONFIG_CONCURRENT_MOD in autoconf.h enough? 

Thanks and regards

lwfinger, 2022-04-09:
Yes, that is how I enabled it for my system. Please do the command lsmod and post the results.

LegBreaker, 2022-04-20:
Hello,
sorry for the late reply.
The hint was already enough. The wrong kernel module was loaded.

I have solved the problem:

`echo "blacklist r8188eu" | tee /etc/modprobe.d/blacklist_r8818eu.conf`

Thanks a lot