lwfinger/rtl8188eu #404: TP-Link TL-WN725N does not connect to router with latest master

mcraveiro, 2022-04-09:
Hi rtl8188eu developers,

Many thanks for maintaining this driver. I just bought a new TP-Link TL-WN725N device which I would like to setup with your driver, but I haven't quite figured out how to do so. In this ticket I will outline my setup as well as the steps I have taken so far; if you can spot where I am going wrong, or provide any other pointers, I would be very grateful.

I apologise in advance for the length of the post, but it should at least save me some time if I ever need to do this again :-)

I suspect my problems have much in common with #390, but I did not want to hijack that ticket just in case its not the same issue.

# Problem Summary

The kernel module loads correctly, but I cannot connect to my router. The device appears to be in some kind of weird state.

# Linux Kernel

I am using Debian unstable, updated today.

```
$ uname -a
Linux lovelace 5.16.0-6-amd64 #1 SMP PREEMPT Debian 5.16.18-1 (2022-03-29) x86_64 GNU/Linux
```

# Networking Hardware Setup

My networking setup is sadly a bit complicated. The PCI networking devices are as follows:

```
$ lspci | grep -i netw
05:00.0 Network controller: Ralink corp. RT5392 PCIe Wireless Network Adapter
$ lspci | grep -i eth
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
``` 

After a similar effort to the present, I'm afraid I did not manage to get RT5392 to work; so I bought myself the afore-mentioned USB based WNIC:

```
# lsusb | grep -i net
Bus 001 Device 015: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```

I chose this device based on the reports of Linux compatibility. The detailed ```lsusb``` for the device is as follows:

```
# lsusb -vv -s 016

Bus 001 Device 016: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x8179 RTL8188EUS 802.11n Wireless Network Adapter
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
  iSerial                 3 00E04C0001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
[... 283 lines cut ...]
[Sat Apr  9 18:56:40 2022] usb 1-9: reset high-speed USB device number 16 using xhci_hcd
# modprobe 8188eu
# dmesg -T | tail
[Sat Apr  9 18:57:35 2022] usbcore: registered new interface driver r8188eu
[Sat Apr  9 18:57:35 2022] r8188eu 1-9:1.0 wlx14ebb6309493: renamed from wlan0
[Sat Apr  9 18:57:35 2022] r8188eu 1-9:1.0: firmware: direct-loading firmware rtlwifi/rtl8188eufw.bin
[Sat Apr  9 18:57:35 2022] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[Sat Apr  9 18:57:36 2022] MAC Address = 14:eb:b6:30:94:93
[Sat Apr  9 18:57:37 2022] R8188EU: INFO indicate disassoc
```

I will continue to investigate and update the ticket, but do let me know of any suggestions of where to look.

Many thanks for your time.

# Sources Consulted

- [8 Linux Commands: To Find Out Wireless Network Speed, Signal Strength And Other Information](https://www.cyberciti.biz/tips/linux-find-out-wireless-network-speed-signal-strength.html)
- [How to know which standard my Wi-Fi connection is currently using](https://askubuntu.com/questions/191650/how-to-know-which-standard-my-wi-fi-connection-is-currently-using)
- [Using iw to Manage Wireless LAN in Linux](http://ict.siit.tu.ac.th/help/iw)

# References

[1] https://github.com/lwfinger/rtl8188eu/blob/master/README.md
[2] https://www.reddit.com/r/openSUSE/comments/pwxs3i/btf_file_not_generated_because_vmlinux_could_not/

mcraveiro, 2022-04-09:
Actually, I wonder if I made a mistake in buying a TP-Link device and assuming it would be Linux-friendly :-) According to this source [1]:

> Note: I will not list any products made by TP-Link and D-Link. Both companies regularly change chipsets while keeping the same model number on their products. This makes it very difficult for Linux users to buy a product with a specific chipset with any degree of certainty. This also makes it difficult for me to post links and recommendations so I will not do so but that is okay because there are many good adapters available. TP-Link and D-Link's Linux support is very poor as their product support sites generally only contain very old Linux drivers that won't work with modern distros... if they post any drivers at all and neither company does a good job of supplying adapters that use in-kernel drivers. We know that Linux cannot be properly supported in the manner that both companies use. This is sad because both companies have made a lot of money from Linux by using it inside many of their products, yet they do not return the support. My recommendation is to avoid TP-Link and D-Link products.

Oh well, one can but try.

[1] https://github.com/morrownr/USB-WiFi

mcraveiro, 2022-04-09:
# Explanation for weird device names

I was wondering if my problems were connected to this message:

```
 wlx14ebb6309493: renamed from wlan0; 
```

However, it appears that the strange device names are by design [1],[2]. In particular:

> It's being renamed so that it has a consistent name regardless of what order network devices are probed/connected (otherwise, if you had two USB wifi devices, which is wlan0 and wlan1 would potentially change every boot — or when doing it by MAC address, replacing a NIC suddenly made eth0 become eth1 causing all kinds of failure). More details & rationale can be found at https://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames/

[1] [[Why is my wlan device being renamed?](https://unix.stackexchange.com/questions/491754/why-is-my-wlan-device-being-renamed)
[2] [How can I change the default "ens33" network device to old "eth0" on Fedora 19?](https://unix.stackexchange.com/questions/81834/how-can-i-change-the-default-ens33-network-device-to-old-eth0-on-fedora-19)

lwfinger, 2022-04-09:
It is your distro's user-space software that sets the device name. That one is pretty strange. On my system, I get names like wlp0s20u6, which is derived from the bus connectivity. lsusb shows
Bus 002 Device 006: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

Note that it has the same vendor and device IDs as yours.

There should not have been any interference between that rt2800 and the rtl8188eus. I frequently run several wireless devices for testing. At the moment, I have two PCI devices and one USB wireless interfaces installed.

Are you using NetworkManager to try to make a connection, or some other management software?

If you use NetworkManager, the fact that the rt2800 cannot connect suggests that something is wrong with the user-space end of your system. The rt2800 driver is solid.

To check your hardware download https://download.opensuse.org/distribution/leap/15.3/live/openSUSE-Leap-15.3-GNOME-Live-x86_64-Build9.443-Media.iso, burn it to a USB stick, and boot it. You should be able to make a connection using the rt2800. I do not suspect your hardware, but this will prove it.

mcraveiro, 2022-05-13:
Actually, after a lot of investigation I didn't make much progress on this. I decided to just get an ethernet cable and that worked fine. Thanks very much for all your help.

closed 2022-05-13 by mcraveiro

gszy, 2022-07-31:
>     # iw dev wlx14ebb6309493 connect SOME-SSID
>     command failed: No such device (-19)

IIRC, when using RTL8188-based cards, wpa_supplicant (or was it Netplan?) fails with the same error if the nl80211 driver is used instead of (or without fallback to) wext. See [`wpa_supplicant(8)`](https://manpages.debian.org/unstable/wpasupplicant/wpa_supplicant.8).