lwfinger/rtl8188eu #406: [kernel 5.10.95] stack corrupted, causes kernel panic

acostach, 2022-05-30:
Hi @lwfinger , I'm getting this panic each time I insert the 8188eu driver at the latest master HEAD with kernel 5.10.95 on a Pi3 32bit, (sorry for the picture quality, unfortunately I couldn't capture better uart logs)
[image]

Device is:
```
Bus 001 Device 004: ID 0bda:8179  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda 
  idProduct          0x8179 
  bcdDevice            0.00
  iManufacturer           1 Realtek
  iProduct                2 802.11n NIC
  iSerial                 3 (error)
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 
      bInterfaceSubClass    255 
      bInterfaceProtocol    255 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
```

Older kernel versions like 5.4.83 work fine though, tested with 60cb0b5a3cfc95e3f2e6fc294a28d987096b0559 without problems.

Any idea what might be causing this or is there other info I could provide that may help diagnose this? Thank you

lwfinger, 2022-05-30:
This problem has not been previously reported. I am not sure how many people are running this driver on 32-bit ARM. Unfortunately, I cannot read your photo at all. If you could blow up the first few lines after the "Kernel PANIC", they might help. Otherwise, it you cannot get a photo, could you transcribe about 10 lines. I do not need the first 2 columns (time and address).

Thanks

lwfinger, 2022-05-30:
One thing that just occurred to me: Make certain that the kernel you are running does not have either rtl8818eu or r8188eu built in. Use 'modinfo rtl8188eu' to test that. If one of these is present, it must be blacklisted for you to use the driver from here.

acostach, 2022-05-31:
Hi @lwfinger and thank you very much for your prompt reply. The rtl8818 and r8188eu are not present, here's what I have on the board:

```
root@02e5c28:~# find /lib/modules/ | grep 8188
/lib/modules/5.10.95-v7/extra/8188eu.ko

root@02e5c28:~# modinfo /lib/modules/5.10.95-v7/extra/8188eu.ko
filename:       /lib/modules/5.10.95-v7/extra/8188eu.ko
version:        v5.2.2.4_25483.20171222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     09579C867E0EE8769DED732
```

```
root@9fd2f14:~# modinfo rtl8188eu
modinfo: ERROR: Module rtl8188eu not found.
root@9fd2f14:~# modinfo r8188eu
modinfo: ERROR: Module r8188eu not found.
```

```
root@9fd2f14:~# zcat /proc/config.gz | grep RTL
CONFIG_BT_RTL=m
CONFIG_BT_HCIBTUSB_RTL=y
# CONFIG_BT_HCIUART_RTL is not set
CONFIG_USB_RTL8150=m
CONFIG_USB_RTL8152=m
CONFIG_RTL8187=m
CONFIG_RTL8187_LEDS=y
CONFIG_RTL_CARDS=m
CONFIG_RTL8192CU=m
CONFIG_RTLWIFI=m
CONFIG_RTLWIFI_USB=m
CONFIG_RTLWIFI_DEBUG=y
CONFIG_RTL8192C_COMMON=m
CONFIG_RTL8XXXU=m
# CONFIG_RTL8XXXU_UNTESTED is not set
# CONFIG_SERIAL_UARTLITE is not set
CONFIG_DVB_USB_RTL28XXU=m
CONFIG_DVB_RTL2830=m
CONFIG_DVB_RTL2832=m
CONFIG_DVB_RTL2832_SDR=m
# CONFIG_RTLLIB is not set
# CONFIG_RTL8723BS is not set

root@02e5c28:~# zcat /proc/config.gz  | grep R8
CONFIG_USB_MR800=m
CONFIG_MEDIA_TUNER_R820T=m
# CONFIG_USB_R8A66597_HCD is not set
# CONFIG_USB_R8A66597 is not set
CONFIG_R8712U=m
# CONFIG_R8188EU is not set
root@02e5c28:~# 

```

The other in-tree modules present are:
```
root@02e5c28:~# find /lib/modules/ | grep rtl
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtl8xxxu
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtl8xxxu/rtl8xxxu.ko
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtlwifi
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192c
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192c/rtl8192c-common.ko
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtlwifi/rtlwifi.ko
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtlwifi/rtl_usb.ko
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192cu
/lib/modules/5.10.95-v7/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192cu/rtl8192cu.ko
[... 322 lines cut ...]
[   22.063994]  r9:81d94000 r8:81cf1000 r7:00000001 r6:3c39a000 r5:81cf4340 r4:00000015
[   22.064010] [<8010ed44>] (ipi_handler) from [<8019a9f8>] (handle_percpu_devid_fasteoi_ipi+0x80/0x154)
[   22.064025] [<8019a978>] (handle_percpu_devid_fasteoi_ipi) from [<80193498>] (generic_handle_irq+0x44/0x54)
[   22.064036]  r7:00000001 r6:81cf0b18 r5:819eb988 r4:000000e0
[   22.064051] [<80193454>] (generic_handle_irq) from [<80740668>] (bcm2836_arm_irqchip_handle_ipi+0xa8/0xc8)
[   22.064067] [<807405c0>] (bcm2836_arm_irqchip_handle_ipi) from [<80193d1c>] (__handle_domain_irq+0x8c/0xe0)
[   22.064077]  r7:00000001 r6:00000000 r5:00000000 r4:818e2d90
[   22.064091] [<80193c90>] (__handle_domain_irq) from [<801012c4>] (bcm2836_arm_irqchip_handle_irq+0x60/0x64)
[   22.064104]  r9:81d94000 r8:81a4061f r7:81d95f6c r6:ffffffff r5:600f0013 r4:8010859c
[   22.064117] [<80101264>] (bcm2836_arm_irqchip_handle_irq) from [<80100abc>] (__irq_svc+0x5c/0x7c)
[   22.064123] Exception stack(0x81d95f38 to 0x81d95f80)
[   22.064131] 5f20:                                                       0028a44a 00000000
[   22.064143] 5f40: 0028a44c 80119f20 81d94000 00000002 81907898 819078e0 81a4061f 80f64948
[   22.064155] 5f60: 00000000 81d95f94 81d95f98 81d95f88 80108598 8010859c 600f0013 ffffffff
[   22.064170] [<80108554>] (arch_cpu_idle) from [<80c9a474>] (default_idle_call+0x40/0x118)
[   22.064183] [<80c9a434>] (default_idle_call) from [<8015babc>] (do_idle+0x10c/0x16c)
[   22.064197] [<8015b9b0>] (do_idle) from [<8015bdfc>] (cpu_startup_entry+0x28/0x2c)
[   22.064210]  r9:410fd034 r8:0000406a r7:81d95ff8 r6:10c0387d r5:00000002 r4:00000092
[   22.064224] [<8015bdd4>] (cpu_startup_entry) from [<8010f2ec>] (secondary_start_kernel+0x168/0x174)
[   22.064236] [<8010f184>] (secondary_start_kernel) from [<00101890>] (0x101890)
[   22.064244]  r5:00000055 r4:01d8806a

```

Thank you and please let me know what I should try next.

lwfinger, 2022-06-01:
I have learned that the "Kernel stack is corrupted" message arises when the system is built with -fstack-protector. Unfortunately, building the x86_64 version with that option does not cause a problem here.

The version of the driver in this repo has a very convoluted way to read/write USB devices. The fact that your two dumps failed from different callers points to something common to read32 and read8.

The only 32-bit system I have is an ancient PowerBook with a ppc32 CPU. It did not crash when running the kernel version from 5.18.0. I do not have GUI desktop and it will be hard to actually make a wifi connection.

I plan to copy the kernel code into a new branch on this repo. The problem will be getting the kernel-dependent changes so that it will build for 5.10.

What image are you using on your Pi3? Perhaps, I can get it to fail when running under QEMU.

lwfinger, 2022-06-01:
At this point, I have no idea what to try. When I copied the kernel source from the staging-next tree, the driver failed one of the startup routines, but did not crash. Copying the source from the mainline kernel did get a crash, but not the same as yours.

I am currently bisecting the staging tree to see when that startup problem was introduced. Once I find the problem there, I will be able to adjust the kernel version to build on older kernels and have something to test.

lwfinger, 2022-06-08:
I have not forgotten about you. There is now a new branch 'kernel' in the repo, which contains the code found in the staging driver of the kernel with the bits added needed for it to build on older kernels. Check out that branch and let me know if it works for you.

acostach, 2022-06-09:
Hi @lwfinger and thank you for helping with this. I've built the branch 'kernel' and unfortunately am still seeing similar panics:

```
[  OK  ] Finished Runtime development features management.
[  OK  ] Listening on Load/Save RF …itch Status /dev/rfkill Watch.
         Starting Load/Save RF Kill Switch Status...
[  OK  ] Started Load/Save RF Kill Switch Status.
[   19.269984] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[   19.269999] [vc_sm_connected_init]: failed to initialize shared memory service
[   19.427575] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: usb_read+0x164/0x164 [8188eu]
[   19.441644] CPU: 1 PID: 971 Comm: systemd-udevd Tainted: G         C O      5.10.95-v7 #1
[   19.453284] Hardware name: BCM2835
[   19.458366] Backtrace: 
[   19.462449] [<80c85e8c>] (dump_backtrace) from [<80c861fc>] (show_stack+0x20/0x24)
[   19.471750]  r7:ffffffff r6:00000000 r5:60070093 r4:819eb7a4
[   19.479090] [<80c861dc>] (show_stack) from [<80c8c7b4>] (dump_stack+0xc4/0xf0)
[   19.488015] [<80c8c6f0>] (dump_stack) from [<80c8660c>] (panic+0x12c/0x344)
[   19.496643]  r9:83d83200 r8:8414bb98 r7:80f619fc r6:00000000 r5:8190ed90 r4:81a59488
[   19.506065] [<80c864e0>] (panic) from [<80c92c7c>] (printk_nmi_enter+0x0/0x44)
[   19.514946]  r3:81f19f00 r2:8398ae80 r1:7f247384 r0:80f619fc
[   19.522214]  r7:84017e00
[   19.526326] [<80c92c60>] (__stack_chk_fail) from [<7f247384>] (usb_write+0x0/0x15c [8188eu])
[   19.537995] [<7f247220>] (usb_read [8188eu]) from [<7f2475ec>] (rtw_read32+0x44/0x64 [8188eu])
[   19.549774]  r8:7f27c858 r7:8301a800 r6:be875000 r5:be875000 r4:8414a000
[   19.558211] [<7f2475a8>] (rtw_read32 [8188eu]) from [<7f2422f4>] (rtl8188e_read_chip_version+0x24/0x70 [8188eu])
[   19.571588]  r4:84017e00
[   19.575710] [<7f2422d0>] (rtl8188e_read_chip_version [8188eu]) from [<7f250260>] (rtw_drv_init+0x1b8/0x2e8 [8188eu])
[   19.589354]  r8:7f27c858 r7:8301a800 r6:be875000 r5:83d83200 r4:84017e00
[   19.597690] [<7f2500a8>] (rtw_drv_init [8188eu]) from [<8090e458>] (usb_probe_interface+0xe4/0x2a0)
[   19.609795]  r9:83d83200 r8:7f27c858 r7:83951878 r6:7f27c8c4 r5:00000001 r4:83d83220
[   19.619158] [<8090e374>] (usb_probe_interface) from [<8085e20c>] (really_probe+0x104/0x4bc)
[   19.630584]  r10:0000001a r9:7f27c858 r8:81aff598 r7:00000000 r6:00000000 r5:81aff590
[   19.640033]  r4:83d83220
[   19.644070] [<8085e108>] (really_probe) from [<8085e970>] (driver_probe_device+0x6c/0xc4)
[   19.655354]  r10:8406a780 r9:00000000 r8:00000011 r7:7f27c858 r6:83d83264 r5:7f27c858
[   19.664875]  r4:83d83220
[   19.668984] [<8085e904>] (driver_probe_device) from [<8085eca4>] (device_driver_attach+0xb8/0xc0)
[   19.681106]  r5:00000000 r4:83d83220
[   19.686294] [<8085ebec>] (device_driver_attach) from [<8085ed3c>] (__driver_attach+0x90/0x120)
[   19.698165]  r7:819ff668 r6:83d83220 r5:7f27c858 r4:00000000
[   19.705487] [<8085ecac>] (__driver_attach) from [<8085be48>] (bus_for_each_dev+0x84/0xd0)
[   19.716955]  r7:819ff668 r6:8085ecac r5:7f27c858 r4:00000000
[   19.724285] [<8085bdc4>] (bus_for_each_dev) from [<8085da98>] (driver_attach+0x2c/0x30)
[   19.734007]  r6:00000000 r5:83d35e00 r4:7f27c858
[   19.740259] [<8085da6c>] (driver_attach) from [<8085d3e8>] (bus_add_driver+0x108/0x1f0)
[   19.749977] [<8085d2e0>] (bus_add_driver) from [<8085f8a8>] (driver_register+0x98/0x128)
[   19.761295]  r7:819ff668 r6:7f27c858 r5:00000000 r4:7f27c858
[   19.768635] [<8085f810>] (driver_register) from [<8090cdf0>] (usb_register_driver+0x88/0x140)
[   19.780432]  r5:7f278eb4 r4:7f27c820
[   19.785672] [<8090cd68>] (usb_register_driver) from [<7f28c050>] (rtw_drv_entry+0x50/0x1000 [8188eu])
[   19.798146]  r9:00000000 r8:00000011 r7:00000000 r6:8406a740 r5:84017a40 r4:7f27c820
[   19.807648] [<7f28c000>] (rtw_drv_entry [8188eu]) from [<8010229c>] (do_one_initcall+0x50/0x280)
[   19.819680]  r5:84017a40 r4:7f28c000
[   19.824831] [<8010224c>] (do_one_initcall) from [<801d7ae4>] (do_init_module+0x70/0x2a4)
[   19.836087]  r7:00000002 r6:8406a740 r5:84017a40 r4:7f27cbc0
[   19.843356] [<801d7a74>] (do_init_module) from [<801da4d0>] (load_module+0x269c/0x2bf4)
[   19.853018]  r6:8406a740 r5:00000002 r4:8414bf30
[   19.859217] [<801d7e34>] (load_module) from [<801dacb8>] (sys_finit_module+0xc4/0x104)
[   19.868788]  r10:0000017b r9:8414a000 r8:80100224 r7:0000017b r6:0000000f r5:76ee735c
[   19.878260]  r4:00000000
[   19.882344] [<801dabf4>] (sys_finit_module) from [<80100204>] (__sys_trace_return+0x0/0x1c)
[   19.893859] Exception stack(0x8414bfa8 to 0x8414bff0)
[   19.900514] bfa0:                   00000000 00000000 0000000f 76ee735c 00000000 00020000
[   19.911839] bfc0: 00000000 00000000 00000000 0000017b 00000000 005636eb 00000000 00d10078
[   19.923194] bfe0: 7eb216e0 7eb216d0 76ee24e7 76dae952
[   19.929846]  r6:00000000 r5:00000000 r4:00000000
[   19.936041] CPU0: stopping
[   19.940265] CPU: 0 PID: 966 Comm: systemd-udevd Tainted: G         C O      5.10.95-v7 #1
[   19.951469] Hardware name: BCM2835
[   19.956345] Backtrace: 
[... 299 lines cut ...]
parm:           rtw_led_enable:int
parm:           rtw_ht_enable:int
parm:           rtw_cbw40_enable:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
[... cut ...] div_type:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)

```

As usual, please let me know once there's something else I should test. Thank you

acostach, 2022-11-10:
Unfortunately no update for this one I guess, right?