lwfinger/rtl8188eu #407: Weird device name

bearsylla, 2022-06-30:
As per mentioned here, https://github.com/lwfinger/rtl8188eu/issues/400#issuecomment-1085079252 

I have cloned the latest from this repo, yet I can't establish a connection. Network Manager keep connects and disconnects and there is no connectivity. 

ip a

```
2: wlxd03745d66e1b: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d0:37:45:d6:6e:1b brd ff:ff:ff:ff:ff:ff
```

dmesg
```
[  437.472122] r8188eu 1-2:1.0: firmware: direct-loading firmware rtlwifi/rtl8188eufw.bin
[  437.946935] R8188EU: INFO assoc success
[  441.966741] R8188EU: INFO ap recv disassoc reason code(15) sta:ae:70:f0:33:96:90
[  441.968211] R8188EU: INFO sta recv deauth reason code(15) sta:ae:70:f0:33:96:90, ignore = 0
[  441.974095] R8188EU: INFO indicate disassoc
[  453.305120] R8188EU: INFO assoc success
[  457.318737] R8188EU: INFO ap recv disassoc reason code(15) sta:ae:70:f0:33:96:90
[  457.319837] R8188EU: INFO sta recv deauth reason code(15) sta:ae:70:f0:33:96:90, ignore = 0
[  457.326100] R8188EU: INFO indicate disassoc
[  463.548008] R8188EU: INFO assoc success
[  467.559113] R8188EU: INFO ap recv disassoc reason code(15) sta:ae:70:f0:33:96:90
[  467.561086] R8188EU: INFO sta recv deauth reason code(15) sta:ae:70:f0:33:96:90, ignore = 0
[  467.566479] R8188EU: INFO indicate disassoc
[  479.025741] R8188EU: INFO assoc success
[  483.038821] R8188EU: INFO ap recv disassoc reason code(15) sta:ae:70:f0:33:96:90
[  483.039824] R8188EU: INFO sta recv deauth reason code(15) sta:ae:70:f0:33:96:90, ignore = 0
[  483.046096] R8188EU: INFO indicate disassoc
[  488.532974] R8188EU: INFO assoc success
[  492.545745] R8188EU: INFO ap recv disassoc reason code(15) sta:ae:70:f0:33:96:90
[  492.546709] R8188EU: INFO sta recv deauth reason code(15) sta:ae:70:f0:33:96:90, ignore = 0
[  492.553729] R8188EU: INFO indicate disassoc
[  503.999497] R8188EU: INFO assoc success
[  508.012971] R8188EU: INFO ap recv disassoc reason code(15) sta:ae:70:f0:33:96:90
[  508.014594] R8188EU: INFO sta recv deauth reason code(15) sta:ae:70:f0:33:96:90, ignore = 0
[  508.020461] R8188EU: INFO indicate disassoc
[  519.007334] r8188eu 1-2:1.0: firmware: direct-loading firmware rtlwifi/rtl8188eufw.bin
```

bearsylla, 2022-06-30:
Additional info, this is a machine running debian x86_64 with kernel 5.18.0-2-amd64 #1 SMP PREEMPT_DYNAMIC Debian 5.18.5-1 (2022-06-16) x86_64 GNU/Linux.

My device is a usb wifi from TP-Link

bearsylla, 2022-06-30:
Anyway, after a few reboots and it works. I am not sure why. The device name is still remain quite long but there is no issue with the connection anymore.

closed 2022-06-30 by bearsylla

lwfinger, 2022-07-01:
That long name is your distro's doing. Mine does it as well. The name is derived from the bus name. On the two USB ports that I use most often, the name is wlp0s20u5 or wlp0s20u6.