lwfinger/rtl8188eu #408: Why my device is discarded from list,

kdm6389, 2022-07-11:
user@host:~ $ lsusb
Bus 001 Device 004: ID 0bda:ffef Realtek Semiconductor Corp. 802.11n NIC

user@host:~ $ dmesg
[    5.810272] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=ffef, bcdDevice= 0.00
[    5.827239] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    5.839069] usb 1-1.4: Product: 802.11n NIC
[    5.847979] usb 1-1.4: Manufacturer: Realtek
[    5.857024] usb 1-1.4: SerialNumber: 00E04C0001

user@host:~ $ ifconfig 
list only eth0 and lo 

please add: 
alias:          usb:v0BDApFFEFd*dc*dsc*dp*ic*isc*ip*in*

I was using this usb-wifi-dongle in my RasPi2B after dist-upgrade, it is broken

kdm6389, 2022-07-12:
https://github.com/lwfinger/rtl8188eu/blob/master/os_dep/usb_intf.c

line no 47-68

/* DID_USB_v916_20130116 */
static struct usb_device_id rtw_usb_id_tbl[] = {
	/*=== Realtek demoboard ===*/
	{USB_DEVICE(USB_VENDER_ID_REALTEK, 0x8179)}, /* 8188EUS */
	{USB_DEVICE(USB_VENDER_ID_REALTEK, 0x0179)}, /* 8188ETV */
	{USB_DEVICE(USB_VENDER_ID_REALTEK, 0xf179)}, /* 8188FU */
	/*=== Customer ID ===*/
	/****** 8188EUS ********/
	{USB_DEVICE(0x07B8, 0x8179)}, /* Abocom - Abocom */
	{USB_DEVICE(0x0DF6, 0x0076)}, /* Sitecom N150 v2 */
	{USB_DEVICE(0x2001, 0x330F)}, /* DLink DWA-125 REV D1 */
        {USB_DEVICE(0x2001, 0x3310)}, /* Dlink DWA-123 REV D1 */
	{USB_DEVICE(0x2001, 0x3311)}, /* DLink GO-USB-N150 REV B1 */
	{USB_DEVICE(0x2001, 0x331B)}, /* D-Link DWA-121 rev B1 */
	{USB_DEVICE(0x056E, 0x4008)}, /* Elecom WDC-150SU2M */
	{USB_DEVICE(0x2357, 0x010c)}, /* TP-Link TL-WN722N v2 */
	{USB_DEVICE(0x2357, 0x0111)}, /* TP-Link TL-WN727N v5.21 */
	{USB_DEVICE(0x2C4E, 0x0102)}, /* MERCUSYS MW150US v2 */
	{USB_DEVICE(0x0B05, 0x18F0)}, /* ASUS USB-N10 Nano B1 */
	{USB_DEVICE(0x7392, 0xb811)}, /* Edimax EW-7811Un V2 */
	{}	/* Terminating entry */
};

noes not contain my device Rosewill USB-N150 Nano
need to append 
	{USB_DEVICE(0x0BDA, 0xffef)}, /* Rosewill USB-N150 Nano */

lwfinger, 2022-07-12:
I have added that device to the tables and pushed the new code. Please report is this makes your device work.

I find that dev_id of ffef to be quite surprising.

kdm6389, 2022-07-13:
```
stpi@rbox2b:~ $ git clone https://github.com/lwfinger/rtl8188eu
Cloning into 'rtl8188eu'...
remote: Enumerating objects: 14104, done.
remote: Counting objects: 100% (1258/1258), done.
remote: Compressing objects: 100% (884/884), done.
remote: Total 14104 (delta 374), reused 1144 (delta 371), pack-reused 12846
Receiving objects: 100% (14104/14104), 23.13 MiB | 3.29 MiB/s, done.
Resolving deltas: 100% (10113/10113), done.

stpi@rbox2b:~ $ cd rtl8188eu/

stpi@rbox2b:~/rtl8188eu $ ls -l 
total 64
-rw-r--r-- 1 stpi stpi  4393 Jul 13 04:38 Makefile
-rw-r--r-- 1 stpi stpi  1961 Jul 13 04:38 README.md
-rwxr-xr-x 1 stpi stpi  4039 Jul 13 04:38 control_ap
drwxr-xr-x 2 stpi stpi  4096 Jul 13 04:38 core
drwxr-xr-x 3 stpi stpi  4096 Jul 13 04:38 debian
-rw-r--r-- 1 stpi stpi   199 Jul 13 04:38 dkms.conf
drwxr-xr-x 2 stpi stpi  4096 Jul 13 04:38 hal
drwxr-xr-x 4 stpi stpi  4096 Jul 13 04:38 hostapd-2.9
drwxr-xr-x 2 stpi stpi  4096 Jul 13 04:38 include
drwxr-xr-x 2 stpi stpi  4096 Jul 13 04:38 os_dep
-rw-r--r-- 1 stpi stpi 13904 Jul 13 04:38 rtl8188eufw.bin
-rw-r--r-- 1 stpi stpi  1819 Jul 13 04:38 rtl_hostapd.conf

stpi@rbox2b:~/rtl8188eu $ sudo make all
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.32-v7+/build M=/home/stpi/rtl8188eu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v7+'
  CC [M]  /home/stpi/rtl8188eu/core/rtw_ap.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_br_ext.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_cmd.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_debug.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_efuse.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_ieee80211.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_io.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_ioctl_set.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_iol.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_led.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_mlme.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_mlme_ext.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_mp.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_mp_ioctl.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_pwrctrl.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_p2p.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_recv.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_rf.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_security.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_sreset.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_sta_mgt.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_wlan_util.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_xmit.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalHWImg8188E_MAC.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalHWImg8188E_BB.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalHWImg8188E_RF.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalPhyRf.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalPhyRf_8188e.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalPwrSeqCmd.o
  CC [M]  /home/stpi/rtl8188eu/hal/Hal8188EPwrSeq.o
  CC [M]  /home/stpi/rtl8188eu/hal/Hal8188ERateAdaptive.o
  CC [M]  /home/stpi/rtl8188eu/hal/hal_intf.o
  CC [M]  /home/stpi/rtl8188eu/hal/hal_com.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_debug.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_interface.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_HWConfig.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_RegConfig8188E.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_RTL8188E.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_cmd.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_dm.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_hal_init.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_mp.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_phycfg.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_rf6052.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_rxdesc.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_sreset.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_xmit.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188eu_led.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188eu_recv.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188eu_xmit.o
  CC [M]  /home/stpi/rtl8188eu/hal/usb_halinit.o
  CC [M]  /home/stpi/rtl8188eu/hal/usb_ops_linux.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/ioctl_linux.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/mlme_linux.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/os_intfs.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/osdep_service.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/recv_linux.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/rtw_android.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/usb_intf.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/usb_ops_linux.o
  CC [M]  /home/stpi/rtl8188eu/os_dep/xmit_linux.o
  LD [M]  /home/stpi/rtl8188eu/8188eu.o
  MODPOST /home/stpi/rtl8188eu/Module.symvers
  CC [M]  /home/stpi/rtl8188eu/8188eu.mod.o
  LD [M]  /home/stpi/rtl8188eu/8188eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v7+'

stpi@rbox2b:~/rtl8188eu $ sudo make install
install -p -m 644 8188eu.ko  /lib/modules/5.15.32-v7+/kernel/drivers/net/wireless
cp rtl8188eufw.bin /lib/firmware/.
/sbin/depmod -a 5.15.32-v7+
mkdir -p /lib/firmware/rtlwifi
cp rtl8188eufw.bin /lib/firmware/rtlwifi/.
stpi@rbox2b:~/rtl8188eu $ 

stpi@rbox2b:~/rtl8188eu $ dmesg
[  545.817421] usb 1-1.2: USB disconnect, device number 4
[  552.517919] usb 1-1.2: new high-speed USB device number 5 using dwc_otg
[  552.649317] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=ffef, bcdDevice= 0.00
[  552.649369] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  552.649396] usb 1-1.2: Product: 802.11n NIC
[  552.649417] usb 1-1.2: Manufacturer: Realtek
[  552.649438] usb 1-1.2: SerialNumber: 00E04C0001
```

didn't worked so far, i m trying, i will  report if I get success, I don't know much

lwfinger, 2022-07-13:
My system shows "alias:          usb:v0BDApFFEFd*dc*dsc*dp*ic*isc*ip*in*" now.
On your system, you now have two drivers - r8188eu and 8188eu. You need to blacklist r8188eu. To verify the issue, do the following:

sudo modprobe -rv r8188eu
sudo modprobe -v 8188eu

kdm6389, 2022-07-13:
`stpi@rbox2b:~/ $ rm -rf rtl8188eu `

```
stpi@rbox2b:~ $ git clone https://github.com/lwfinger/rtl8188eu/
Cloning into 'rtl8188eu'...
remote: Enumerating objects: 14104, done.
remote: Counting objects: 100% (1258/1258), done.
remote: Compressing objects: 100% (884/884), done.
remote: Total 14104 (delta 374), reused 1144 (delta 371), pack-reused 12846
Receiving objects: 100% (14104/14104), 23.13 MiB | 3.45 MiB/s, done.
Resolving deltas: 100% (10113/10113), done.
```

```
stpi@rbox2b:~/rtl8188eu $ sudo make all && sudo modprobe -rv r8188eu && sudo modprobe -rv 8188eu && sudo make install
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.32-v7+/build M=/home/stpi/rtl8188eu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v7+'
  CC [M]  /home/stpi/rtl8188eu/core/rtw_ap.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_br_ext.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_cmd.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_debug.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_efuse.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_ieee80211.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_io.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_ioctl_set.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_iol.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_led.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_mlme.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_mlme_ext.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_mp.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_mp_ioctl.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_pwrctrl.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_p2p.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_recv.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_rf.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_security.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_sreset.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_sta_mgt.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_wlan_util.o
  CC [M]  /home/stpi/rtl8188eu/core/rtw_xmit.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalHWImg8188E_MAC.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalHWImg8188E_BB.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalHWImg8188E_RF.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalPhyRf.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalPhyRf_8188e.o
  CC [M]  /home/stpi/rtl8188eu/hal/HalPwrSeqCmd.o
  CC [M]  /home/stpi/rtl8188eu/hal/Hal8188EPwrSeq.o
  CC [M]  /home/stpi/rtl8188eu/hal/Hal8188ERateAdaptive.o
  CC [M]  /home/stpi/rtl8188eu/hal/hal_intf.o
  CC [M]  /home/stpi/rtl8188eu/hal/hal_com.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_debug.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_interface.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_HWConfig.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_RegConfig8188E.o
  CC [M]  /home/stpi/rtl8188eu/hal/odm_RTL8188E.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_cmd.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_dm.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_hal_init.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_mp.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_phycfg.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_rf6052.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_rxdesc.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_sreset.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188e_xmit.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188eu_led.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188eu_recv.o
  CC [M]  /home/stpi/rtl8188eu/hal/rtl8188eu_xmit.o
  CC [M]  /home/stpi/rtl8188eu/hal/usb_halinit.o
  CC [M]  /home/stpi/rtl8188eu/hal/usb_ops_linux.o
[... 71 lines cut ...]
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           debug:Set debug level (1-9) (default 1) (int)
```

```
[ 1432.955134] usb 1-1.4: new high-speed USB device number 5 using dwc_otg
[ 1433.096415] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=ffef, bcdDevice= 0.00
[ 1433.096464] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1433.096490] usb 1-1.4: Product: 802.11n NIC
[ 1433.096512] usb 1-1.4: Manufacturer: Realtek
[ 1433.096532] usb 1-1.4: SerialNumber: 00E04C0001
```

kdm6389, 2022-07-13:
added at line 53 to code {USB_DEVICE(0x0BDA, 0xffef)},  to ~/rtl8188eu/os_dep/usb_intf.c 
and 

```
user@raaspberry2b:~/rtl8188eu $ sudo make all && sudo modprobe -rv r8188eu && sudo modprobe -rv 8188eu && sudo make install
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.15.32-v7+/build M=/home/user/rtl8188eu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v7+'
  CC [M]  /home/user/rtl8188eu/os_dep/usb_intf.o
  LD [M]  /home/user/rtl8188eu/8188eu.o
  MODPOST /home/user/rtl8188eu/Module.symvers
  CC [M]  /home/user/rtl8188eu/8188eu.mod.o
  LD [M]  /home/user/rtl8188eu/8188eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v7+'
install -p -m 644 8188eu.ko  /lib/modules/5.15.32-v7+/kernel/drivers/net/wireless
cp rtl8188eufw.bin /lib/firmware/.
/sbin/depmod -a 5.15.32-v7+
mkdir -p /lib/firmware/rtlwifi
cp rtl8188eufw.bin /lib/firmware/rtlwifi/.

user@raaspberry2b:~/rtl8188eu $ sudo modprobe -v 8188eu

user@raaspberry2b:~/rtl8188eu $ dmesg | tail -9
[ 2004.862480] usb 1-1.4: new high-speed USB device number 6 using dwc_otg
[ 2004.993758] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=ffef, bcdDevice= 0.00
[ 2004.993806] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2004.993832] usb 1-1.4: Product: 802.11n NIC
[ 2004.993853] usb 1-1.4: Manufacturer: Realtek
[ 2004.993873] usb 1-1.4: SerialNumber: 00E04C0001
[ 2004.995554] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[ 2005.033873] EEPROM ID = 0x8129
[ 2005.672935] MAC Address = 68:ff:dd:aa:bb:3f
user@raaspberry2b:~/rtl8188eu $ ifconfig 
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether b8:27:eb:8b:b1:37  txqueuelen 1000  (Ethernet)
        RX packets 22012  bytes 25992501 (24.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 10559  bytes 1260588 (1.2 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 10  bytes 1570 (1.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 10  bytes 1570 (1.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 68:ff:dd:aa:bb:3f  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

user@raaspberry2b:~/rtl8188eu $ 
```

finally worked ...........

kdm6389, 2022-07-13:
close the topic :+1: :1st_place_medal: :bowing_man:

closed 2022-07-13 by kdm6389