lwfinger/rtl8188eu #409: Does this driver support nl80211?

izuk, 2022-07-26:
The driver seems to load OK, `5.18.13`:

```
$ dmesg
[   35.612807] 8188eu: loading out-of-tree module taints kernel.
[   35.618136] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[...]
[   35.658541] usbcore: registered new interface driver r8188eu
[...]
[   35.777840] r8188eu 1-6.4:1.0 wlp0s20f0u6u4: renamed from wlan0
[...]
[   36.119265] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[   36.454259] MAC Address = XX:XX:XX:XX:XX:XX

$ lsmod
[...]
8188eu                790528  0
usbcore               327680  4 xhci_hcd,usbhid,8188eu,xhci_pci
```

But `iw` doesn't list it:

```
$ sudo iw list
$ 
```

Before I dig further, does this driver support `iw` (ie., nl80211)?  Am I missing something obvious?

lwfinger, 2022-07-26:
The master branch does not, but v5.2.2.4 does. Do 'git checkout origin/v5.2.2.4 -b v5.2.2.4'.

izuk, 2022-07-27:
That did the trick, thanks!

closed 2022-07-27 by izuk