lwfinger/rtl8188eu #412: Module 8188eu.ko not found in directory /lib/modules/5.4.0-122-generic

rkleczek457, 2022-08-12:
is what I got from sudo modprobe 8188eu.ko . For make and sudo make install there were no errors. I can't find any existing solutions to the problem. Could anyone help?

rkleczek457, 2022-08-12:
Also the device works but the speed is very slow which i suspect is caused by this

lwfinger, 2022-08-12:
The command is 'sudo modprobe -v 8188eu'. You need the .ko part for insmod, but never for modprobe.

First do 'lsmod | grep 8188eu'. That will show you what driver is now loaded. If not 8188eu, that one will need to be blacklisted.

rkleczek457, 2022-08-12:
I think it worked but it's still very slow,around 1mb/s

lwfinger, 2022-08-12:
Being slow can be a result of having both the driver from the kernel and this one installed. Please run lsmod and post the output here.

I suspect you are using the master branch. Try
git checkout origin/v5.2.2.4 -b v5.2.2.4
git pull
make
sudo make install
sudo modprobe -rv 8188eu
sudo modprobe -v 8188eu

With the master branch, I get 8 Mbps down and 4 up. With v5.2.2.4, I get 38 Mbps down. and 25 Mbps up. Your results may be different, but please try it.

rkleczek457, 2022-08-13:
Module                  Size  Used by
xt_conntrack           16384  1
xt_MASQUERADE          20480  1
nf_conntrack_netlink    45056  0
nfnetlink              16384  2 nf_conntrack_netlink
xfrm_user              36864  1
xfrm_algo              16384  1 xfrm_user
xt_addrtype            16384  2
iptable_nat            16384  1
nf_nat                 45056  2 iptable_nat,xt_MASQUERADE
nf_conntrack          139264  4 xt_conntrack,nf_nat,nf_conntrack_netlink,xt_MASQUERADE
nf_defrag_ipv6         24576  1 nf_conntrack
nf_defrag_ipv4         16384  1 nf_conntrack
libcrc32c              16384  2 nf_conntrack,nf_nat
iptable_filter         16384  1
br_netfilter           28672  0
xt_comment             16384  0
bridge                176128  1 br_netfilter
stp                    16384  1 bridge
llc                    16384  2 bridge,stp
bpfilter               32768  0
rfcomm                 81920  4
aufs                  262144  0
cmac                   16384  2
algif_hash             16384  1
algif_skcipher         16384  1
af_alg                 24576  6 algif_hash,algif_skcipher
bnep                   24576  2
overlay               118784  0
vboxnetadp             28672  0
vboxnetflt             28672  0
vboxdrv               499712  2 vboxnetadp,vboxnetflt
binfmt_misc            24576  1
nls_iso8859_1          16384  1
intel_rapl_msr         20480  0
mei_hdcp               24576  0
wl                   6455296  0
btusb                  57344  0
btrtl                  24576  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
razermouse            114688  0
bluetooth             552960  31 btrtl,btintel,btbcm,bnep,btusb,rfcomm
snd_hda_codec_realtek   131072  1
snd_hda_codec_generic    81920  1 snd_hda_codec_realtek
snd_hda_codec_hdmi     61440  1
rtsx_usb_ms            24576  0
snd_hda_intel          53248  4
snd_intel_dspcfg       28672  1 snd_hda_intel
memstick               20480  1 rtsx_usb_ms
ecdh_generic           16384  1 bluetooth
ecc                    28672  1 ecdh_generic
snd_hda_codec         139264  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core           90112  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_hwdep              20480  1 snd_hda_codec
intel_rapl_common      24576  1 intel_rapl_msr
x86_pkg_temp_thermal    20480  0
snd_pcm               106496  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
intel_powerclamp       20480  0
snd_seq_midi           20480  0
snd_seq_midi_event     16384  1 snd_seq_midi
coretemp               20480  0
snd_rawmidi            36864  1 snd_seq_midi
kvm_intel             282624  0
snd_seq                69632  2 snd_seq_midi,snd_seq_midi_event
kvm                   663552  1 kvm_intel
crct10dif_pclmul       16384  1
ghash_clmulni_intel    16384  0
aesni_intel           372736  3
crypto_simd            16384  1 aesni_intel
[... 45 lines cut ...]
rtsx_usb_sdmmc         32768  0
rtsx_usb               28672  2 rtsx_usb_sdmmc,rtsx_usb_ms
hid_rmi                24576  0
rmi_core               81920  1 hid_rmi
videobuf2_vmalloc      20480  1 rmi_core
videobuf2_memops       20480  1 videobuf2_vmalloc
videobuf2_v4l2         24576  1 rmi_core
videobuf2_common       57344  2 rmi_core,videobuf2_v4l2
videodev              225280  3 rmi_core,videobuf2_v4l2,videobuf2_common
mc                     53248  3 videodev,videobuf2_v4l2,videobuf2_common
usbhid                 57344  0
hid_generic            16384  0
crc32_pclmul           16384  0
i2c_i801               32768  0
r8169                  90112  0
ahci                   40960  2
lpc_ich                24576  0
libahci                32768  1 ahci
realtek                24576  1
wmi                    32768  4 dell_wmi,wmi_bmof,dell_smbios,dell_wmi_descriptor
video                  53248  3 dell_wmi,dell_laptop,i915
dw_dmac                16384  0
dw_dmac_core           28672  1 dw_dmac
i2c_hid                28672  0
hid                   131072  5 i2c_hid,usbhid,hid_generic,hid_rmi,razermouse

rkleczek457, 2022-08-13:
Also the slow speed is still there

lwfinger, 2022-08-13:
In your list, cfg80211 is only used by wl. If you had version v5.2.2.4 installed, it would also be used by 8188eu. You can check that with

git branch > temp
cat temp

Post the output of the cat command.

rkleczek457, 2022-08-13:
master
* v5.2.2.4

lwfinger, 2022-08-13:
You have the branch we wanted installed. Run the following:

git pull
make
sudo make install
sudo modprobe -rv 8188eu
sudo modprobe -v 8188eu
lsmod | grep 8818

Show the output of any errors and the complete output of the lsmod command.

rkleczek457, 2022-08-13:
8188eu 1490944 0
cfg80211 708608 1 8188eu

lwfinger, 2022-08-13:
You have the right one now. Any improvement in speed?

rkleczek457, 2022-08-13:
It's 20mb/s at best.

lwfinger, 2022-08-13:
Since you are running v5.2.2.4, which is the latest, I am not able to help you any more. I did not write the driver and I have no knowledge of the chip internals, thus I have no way to improve throughput!

lwfinger, 2022-08-14:
I have some more questions about your system
1. What is the Broadcom system you are driving with "wl"?
2. If it works, what speeds do you get?
3. What signal strength do you see with the RTL8188EU?

rkleczek457, 2022-08-26:
On holiday,give me a week