lwfinger/rtl8188eu #415: rtl8188eu kernel panic with preempt_rt kernel 5.10

erichyc7035, 2022-08-23:
Hi,

Is anyone tried to run the driver within Kernel 5.10 with preempt_rt patched?
I met kernel panic as below:
[   79.141208] R8188EU: assoc success
[   79.151525] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   79.216300] R8188EU: sta recv deauth reason code(7) sta:00:fe:c8:28:12:f0
[   79.225988] Unable to handle kernel paging request at virtual address ffff80001256e054
[   79.233906] Mem abort info:
[   79.236728] printk: console [ttyS2]: printing thread stopped
[   79.243058]   ESR = 0x96000021
[   79.246158]   EC = 0x25: DABT (current EL), IL = 32 bits
[   79.251476]   SET = 0, FnV = 0
[   79.254523]   EA = 0, S1PTW = 0
[   79.257664] Data abort info:
[   79.260544]   ISV = 0, ISS = 0x00000021
[   79.264378]   CM = 0, WnR = 0
[   79.267346] swapper pgtable: 4k pages, 48-bit VAs, pgdp=0000000082e93000
[   79.274044] [ffff80001256e054] pgd=00000008fffff003, p4d=00000008fffff003, pud=00000008ffffe003, pmd=0000000881758003, pte=0068000886b1c703
[   79.286567] Internal error: Oops: 96000021 [#1] PREEMPT_RT SMP
[   79.292390] Modules linked in: r8188eu(C) lib80211 cfg80211 bluetooth ecdh_generic ecc rfkill xhci_plat_hcd xhci_hcd usbcore rpmsg_char ti_am335x_adc kfifo_buf dwc3 udc_core roles usb_common omap_rng rng_core irq_pruss_intc crct10dif_ce icssg_prueth pru_rproc icss_iep phy_can_transceiver ti_k3_r5_remoteproc virtio_rpmsg_bus m_can_platform m_can can_dev ti_am335x_tscadc ti_cal videobuf2_dma_contig v4l2_fwnode videobuf2_memops videobuf2_v4l2 videobuf2_common pruss sa2ul sha512_generic authenc phy_omap_usb2 dwc3_keystone sch_fq_codel ipv6
[   79.339883] CPU: 0 PID: 1314 Comm: RTW_CMD_THREAD Tainted: G         C O      5.10.100-rt62-g204ec708dc #24
[   79.349609] Hardware name: Texas Instruments AM654 Base Board (DT)
[   79.355777] pstate: 40000005 (nZcv daif -PAN -UAO -TCO BTYPE=--)
[   79.361774] pc : rt_spin_unlock+0xa8/0x130
[   79.365877] lr : rtw_free_xmitframe_queue+0x38/0x90 [r8188eu]
[   79.371720] sp : ffff80001308bc40
[   79.375024] x29: ffff80001308bc40 x28: ffff800008daf308
[   79.380330] x27: ffff800011a77000 x26: ffff800011a770a0
[   79.385634] x25: ffff800008dc2000 x24: ffff800011a77008
[   79.390939] x23: ffff80001256dd6c x22: ffff80001256e03c
[   79.396243] x21: ffff800011a785e8 x20: ffff80001256e02c
[   79.401547] x19: ffff800011a785e8 x18: 0000000000000010
[   79.406852] x17: 0000000000000001 x16: 0000000000000019
[   79.412156] x15: 000005bc3a0feed8 x14: 0000000000000227
[   79.417460] x13: 0000000000000227 x12: 0000000000000000
[   79.422764] x11: 0000000000001852 x10: 0000000000000a50
[   79.428068] x9 : ffff80001308b950 x8 : ffff00087f784ec0
[   79.433372] x7 : ffff000000000000 x6 : 00000000000015e8
[   79.438676] x5 : ffff80001256e054 x4 : 0000000000000800
[   79.443981] x3 : ffff80001256e03c x2 : ffff000803aec9c0
[   79.449285] x1 : 0000000000000000 x0 : ffff80001256e03c
[   79.454590] Call trace:
[   79.457029]  rt_spin_unlock+0xa8/0x130
[   79.460773]  rtw_free_xmitframe_queue+0x38/0x90 [r8188eu]
[   79.466244]  rtw_free_stainfo+0x58/0x3f0 [r8188eu]
[   79.471106]  rtw_free_assoc_resources_locked+0x7c/0x228 [r8188eu]
[   79.477267]  rtw_free_assoc_resources+0x38/0x60 [r8188eu]
[   79.482736]  rtw_stadel_event_callback+0x188/0x3a0 [r8188eu]
[   79.488465]  mlme_evt_hdl+0x5c/0x128 [r8188eu]
[   79.492980]  rtw_cmd_thread+0x1e4/0x310 [r8188eu]
[   79.497755]  kthread+0x174/0x198
[   79.500982]  ret_from_fork+0x10/0x34
[   79.504562] Code: c8047ca2 35ffff84 17ffff30 f98000b1 (c85ffca0)
[   79.510648] ---[ end trace 0000000000000002 ]---
[   79.515437] Kernel panic - not syncing:
[   79.519261] Oops: Fatal exception in interrupt
[   79.519381] SMP: stopping secondary CPUs
[   79.527752] Kernel Offset: disabled
[   79.531230] CPU features: 0x0040002,2000200c
[   79.535490] Memory Limit: none
[   79.538538] ---[ end Kernel panic - not syncing: Oops: Fatal exception in interrupt ]---

I'm not sure if it's related to kernel 5.10 or due to preempt_rt patch.
Please someone help to check.

Thanks.

lwfinger, 2022-08-23:
You are using driver r8188eu. That means it comes from the kernel, not this repo. You need to report this problem at linux-wireless@vger.org. If you blacklist r8188eu, install the driver from this repo, and still get the error, then I can look at it.

erichyc7035, 2022-08-24:
Hi Iwfinger,

Thanks for your response.
I will report this to the one you mentioned to see if it can be solved.

Thanks.

Muil1024, 2024-10-12:
> Hi Iwfinger,
> 
> Thanks for your response. I will report this to the one you mentioned to see if it can be solved.
> 
> Thanks.

Hi erichyc7035

I have the same problem with you.
Have you found a solution and can you share it?

Thanks.

erichyc7035, 2024-10-15:
@Muil1024 

No, I didn't find a solution for this.

Eric