lwfinger/rtl8188eu #417: Kernel panic for some Wifis only

lemked, 2022-08-28:
First of all, thanks for providing this driver as an open-source project, really appreciate all the effort it takes in developing and maintaining such a project.

I've done a fresh installation of Raspberry Pi OS Lite today on my Raspberry Model B+ (2014) that I'm planning to use for some home automation project. The full OS details are below:

```
Release date: April 4th 2022
System: 64-bit
Kernel version: 5.15
Debian version: 11 (bullseye)
Size: 270MB
```

After the initial installation, I've installed the drivers for the TP-Link Wifi adapter following the instructions [here](https://gist.github.com/MBing/de297a8ae5e8a191c55a67a568d20d31) (using the master branch). Everything worked fine, so I was able to setup the Wifi connection and connect to the Raspberry via wireless network.

Then the fun started. After moving the Raspberry to a different physical location, having a different Wifi Router, it crashed with a Kernel panic as shown below on startup. After a bit forth and back, I realized that it's related to the Wifi, so without the wifi adapter it started up fine, and also when removing the `network` within the `wpa_supplicant.conf` it worked fine. But as soon as I add it (either manually or via `raspi-config`, I'll receive a Kernel panic shortly.

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=DE

network={
	ssid="My Wifi"
	psk="..."
	id_str="rs5"
}
```

[image]
[image]

Any ideas why this is working for one wifi, but not for another?

lwfinger, 2022-08-29:
Obviously, the AP settings are different. Unfortunately, about all I can read from your screen shots is that you got a NULL pointer dereference crash.

I'm not sure what facilities you have for scanning for wireless networks. If you have iw, then 'sudo iw dev <net_device> scan' will show you what the AP is offering. You do not need to be connected for this to work. Please post the results for the AP that works, and the one that fails.

lemked, 2022-08-29:
Thanks @lwfinger . Am a few days on vacation now but will get back with the scan details here after. Thanks for the quick reply!

lemked, 2022-09-17:
Hi @lwfinger ,

I've done a scan from my Windows machine as `iw` on my Raspberry didn't return me any results. Here's what I get from `netsh wlan show interfaces` for the problematic AP (sorry for the German texts, but I guess the meanings should be clear):

```
    Netzwerktyp            : Infrastruktur
    Funktyp                : 802.11n
    Authentifizierung      : WPA2-Personal
    Verschlüsselung        : CCMP
    Verbindungsmodus       : Automat. Verbindung
    Kanal                  : 11
    Empfangsrate (MBit/s)  : 130
    Übertragungsrate (MBit/s) : 130
    Signal                 : 80%
```

Here's what I have at my home wifi, where the Raspberry with the Wifi dongle is working fine:

```
    Netzwerktyp            : Infrastruktur
    Funktyp                : 802.11ac
    Authentifizierung      : WPA2-Personal
    Verschlüsselung        : CCMP
    Verbindungsmodus       : Automat. Verbindung
    Kanal                  : 36
    Empfangsrate (MBit/s)  : 526.5
    Übertragungsrate (MBit/s) : 526.5
    Signal                 : 76%
```

I guess the main difference is the wifi standard, with my home wifi being on the more recent wifi 5 while the problematic AP still uses wifi 4. On my home network the wifi dongle connects via 2.4 GHz by the way, while on the problematic AP I'm not sure because of the crash it doesn't show up in my wifi router's overview).

lwfinger, 2022-09-18:
The availability of the 5 GHz band does not depend on wifi 5. In general, the 2.4 GHz band will be more crowded than is the 5 GHz band, which could also be a problem.

lemked, 2022-10-09:
So I split the networks now via the router into a 5 GHz and 2.4 GHz band to check whether it's related to that. 

After that, when connecting to the 2.4 GHz I keep receiving a kernel panic. When selecting the 5 GHz band, the Pi starts, but it doesn't connect to the Wifi AP. When I run `wpa_cli`, it returns the following:

```
<3>WPS-AP-AVAILABLE
<3>CTRL-EVENT-NETWORK-NOT-FOUND
```

Not sure what else to try from here, other than trying a different OS or driver version.

lwfinger, 2022-10-10:
As the rtl8188eu only has a 2.4 GHz radio, you can hardly expect it to connect to a 5GHz band on the AP.

I cannot read your posted photos, thus I have no way of seeing why the PI should panic when x86_64 systems do not. It also works on PowerPC 32-bit processors, thus it is not a 32- versus 64-bit problem.

If you can either provide a clearer screen shot, or transcribe the Backtrace information following the register dump, then I might be able to help you.

lemked, 2022-10-13:
Well, that explains why it wasn't connecting to the 5GHz point 😄 I could swear the vendor's website where I bought it some time back listed it as 5GHz compatible, but anyway.

Not sure where you're coming from regarding the processor architecture. The dongle actually works fine on that very same Raspberry when I'm using it at my wifi AP at home. It's just when using it on that other wifi AP that it crashes. Both APs use entirely different routers.

I'll try to get you a bit clearer screenshots next time I have a chance.

Thanks again for your efforts here, really appreciate it.