lwfinger/rtl8188eu #418: KSRC is not working on branch v5.2.2.4

maskinoshita, 2022-08-29:
KSCR is set fixed url, so it use wrong kernel sources when cross compiling.

https://github.com/lwfinger/rtl8188eu/blob/v5.2.2.4/Makefile#L346

lwfinger, 2022-08-29:
I do not do cross compiling. What should it be?

maskinoshita, 2022-08-29:
On master branch, it can change kernel header path using KSRC.
But, branch v5.2.2.4 cannot.

On v5.2.2.4, kernel header path is fixed as host kernel header path.
But cross compile needs other kernel header path that targets the platform like arm, etc.

I want the same behavior for KSRC on v5.2.2.4 branch.

[on master branch] -> KSRC specified kernel path used
KSRC=/linux-src make modules
make ARCH=x86_64 CROSS_COMPILE= -C /linux-src M=/mnt/d/tmp3/rtl8188eu  modules

[on v5.2.2.4 branch (host is WSL2)]  -> KSRC specified a kernel source path, but uses host kernel header path
KSRC=/linux-src make modules
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.102.1-microsoft-standard-WSL2/build M=/mnt/d/tmp3/rtl8188eu  modules

lwfinger, 2022-08-29:
I changed v5.2.2.4 to define KSRC the same as the master branch does.

closed 2022-08-29 by lwfinger

maskinoshita, 2022-08-29:
Thank you for your quick response.
It works well.