lwfinger/rtl8188eu #419: Unable to get RTL8188ETV working with the driver

Mastiffen, 2022-08-30:
Thanks for making a driver for cheap and readily available cards! I was hoping I should be able to get an RTL8188ETV card working with this, so I bought a few of them after I read a posting that this driver should work. But it only partly does. I was going to use it as the access point card in a routed access point setup for my boat, with one Asus USB-AC68 connecting to the remote (which is connected to the cabin wifi) and this card giving stuff on the boat acecss (mainly ESP-32's and two phones working as SignalK dashboards, so I only need 2.4GHz). When I fire it up, I only see "Not associated" when hovering over the network icon in the system tray. I have also tried to set it up as the receiving card, instead of the Asus, but it wouldn't connect to the access point given in the WPA_supplicant config file (and yes, it's 2.4GHz). ifconfig shows me the card, but not much info about it:

`wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 0c:c6:55:3d:fe:79  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`
I have of course done a git clone, and I have my headers up to date. This is my version:
Linux 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

I have not gone to Bullseye, because I use a Tellstick Duo in the boat for a couple of temp sensors, and that won't work under Bullseye.

This is dmesg when I plug in the card:

`[  334.586886] usb 1-1.1: new high-speed USB device number 7 using xhci_hcd
[  334.727757] usb 1-1.1: New USB device found, idVendor=0bda, idProduct=0179, bcdDevice= 0.00
[  334.727767] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  334.727775] usb 1-1.1: Product: 802.11n WLAN NIC
[  334.727782] usb 1-1.1: Manufacturer: Realtek
[  334.732504] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_UNKNOWN_CUT(10)_1T1R_RomVer(0)
[  334.759036] EEPROM ID = 0x8129
[  335.287217] MAC Address = 0c:c6:55:3d:fe:79
`
Could somebody please give me apointer to what I need to do to get the driver to work with this card? I apologize if it's an idiot newbie thing...

lwfinger, 2022-08-31:
If you want to use the chip as an AP, then I suggest you switch to the v5.2.2.4 branch using the command
git checkout origin/v5.2.2.4 -b v5.2.2.4

The code in this branch uses cfg80211 and works with a modern hostapd that uses nl80211 as the driver.

I have not tried this particular form of the chip. Mine is an RTL8188EUS.

Mastiffen, 2022-08-31:
Fantastic, thank you VERY much for the help! Works like a charm now, it seems! 
Only one more thing: I tried to get it persistant as wlan0 (since it seems like this and the Asus USB takes wlan0 rather erratic) with the command:
`ACTION=="add", SUBSYSTEM=="net", DRIVERS=="8188eu", NAME="wlan0"`
This should make it persistant if the driver is actually named 8188eu, as lsmod shows. That way I wouldn't have to mess much if I have to replace one of the USB network cards because it has failed. But it doesn't seem to work. Do you have any idea why that is?

lwfinger, 2022-08-31:
You need to change the "wlan0" to "wlan5", or something else that the Asus will not touch. It would also be best if you wrote a similar rule for the Asus dongle.

lwfinger, 2022-09-01:
That is comething called a udev rule and can not do it in a command line.  You need to create (as root) a file /etc/udev/rules.d/70-8188eu.rules and add the line above in that file.

Mastiffen, 2022-09-02:
I couldn't get that to work, but after a bit of searching I put both lines into the same file, named /etc/udev/rules.d/70-persistent-net.rules, so I now have:

`ACTION=="add", SUBSYSTEM=="net", DRIVERS=="rtl8188eu", NAME="madmaxruter"
ACTION=="add", SUBSYSTEM=="net", DRIVERS=="8812au", NAME="hytta"
ACTION=="add", SUBSYSTEM=="net", DRIVERS=="8814au", NAME="hytta"`

That wa I can use my backup AC card (TP Link) as the link up to the cabin (hytta in Norwegian), while I can use any of the RTL8188ETV-cards I have bought (I bought three, even in crimp hoses they tend to go bad after a couple of years from the corrosive salt water air) as the onboard router card, for ESP-32's, SignalK plotter duplication and other stuff.

Thanks a lot for your help and patience!

Mastiffen, 2022-09-12:
Hi again! Sorry to bug you again, but I tried to install the card as a second USB on a 3B+, and it just won't connect to anything. The dongle shows in lsusb and ifconfig, but "not associated" in the network interface icon on the taskbar. I have created a wpa_supplicant file with the correct info, but it just won't connect to anything. I have tried more than one network, all 2.4 GHz. I tried the second driver too, but that didn't change anything.