lwfinger/rtl8188eu #420: Question: howto ignore deauth reason code 6

rhermsen, 2022-09-01:
I'm having frequent "rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:<bssid>, ignore = 0"
Found that: 
deauth reason code 6 = Class 2 frame received from nonauthenticated station

My WiFi setup.
Two Fritsbox routers in a WiFi MESH using the same SSID.
Also the same SSID is used for the 2.4G and 5G band. 
	- 7590 WiFi Mesh Master
	- 7490 WiFi Mesh Repeater

I assume the "nonauthenticated station" to be the not-associated Mesh system.
Is there an option to make 'ignore = 1'?

lwfinger, 2022-09-02:
To do that, you would need to edit file core/rtw_mlme_ext.c, and find the section at line 2019 that does as follows:

                        }
                }

                DBG_88E_LEVEL(_drv_always_, "sta recv deauth reason code(%d) sta:%pM, ignore = %d\n",
                              reason, GetAddr3Ptr(pframe), ignore_received_deauth);

                if (!ignore_received_deauth)
                        receive_disconnect(padapter, GetAddr3Ptr(pframe), reason);
        }

Before the "if" statement, add a line that says "if (reason == 6) ignore_received_deauth = 1;". As I have no idea what the bad effect might be of  ignoring the error, I cannot make that change for everyone, but you are welcome to change the version on your computer. The reason for ignore code is described in some comments a few lines up.

If the only effect is to improve the stability of the system, I would then consider adding it to the standard driver.

rhermsen, 2022-09-02:
Thanks a lot for the quick answer.
For me it was at line :1625 I hope that is correct...
```
diff -durN core/rtw_mlme_ext.c.orig core/rtw_mlme_ext.c
--- core/rtw_mlme_ext.c.orig    2022-09-02 19:14:25.435806843 +0200
+++ core/rtw_mlme_ext.c 2022-09-02 19:20:02.911062819 +0200
@@ -1622,6 +1622,8 @@
                DBG_88E_LEVEL(_drv_always_, "sta recv deauth reason code(%d) sta:%pM, ignore = %d\n",
                              reason, GetAddr3Ptr(pframe), ignore_received_deauth);
 
+               if (reason == 6) ignore_received_deauth = 1;
+
                if (!ignore_received_deauth)
                        receive_disconnect(padapter, GetAddr3Ptr(pframe), reason);
        }
```
Just installed it and will monitor the next few days.

lwfinger, 2022-09-03:
Yes, that is the correct place.

rhermsen, 2022-09-03:
Unfortunately that didn't work...
This morning found the RPi2 disconnected.
Wasn't able to ping the wlan0 interface at first, only after also connecting eth0 was able to reach both interfaces.
(and failed again after disconnecting eth0. See indeed only eth0 MAC in the arp tables for both interfaces)
From Sep 03 03:05:14 onwards, every second or two I see a "```rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:<bssid_7590_2_4GHz>, ignore = 0```" message.
It was initially connected to the Mesh-Repeater (<bssid_7490_2_4GHz>) which is a bit further away.
So looks it tried to connect to the Mesh-Master (<bssid_7590_2_4GHz>) which triggered the 'deauth loop'.

```
$ cat last_journalctl.log | grep -i "sta recv deauth reason code(6)" | wc -l
24006
Sep 03 11:47:18 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:19 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:20 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:21 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:22 rpi2b dhcpcd[385]: eth0: carrier acquired
Sep 03 11:47:22 rpi2b kernel: smsc95xx 1-1.1:1.0 eth0: Link is Up - 100Mbps/Full - flow control off
Sep 03 11:47:22 rpi2b kernel: IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sep 03 11:47:22 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:22 rpi2b dhcpcd[385]: eth0: IAID -iaid_rpi2b_eth0-
Sep 03 11:47:22 rpi2b dhcpcd[385]: eth0: adding address fe80::3c44:852f:9baa:add5
Sep 03 11:47:22 rpi2b avahi-daemon[324]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::3c44:852f:9baa:add5.
Sep 03 11:47:22 rpi2b avahi-daemon[324]: New relevant interface eth0.IPv6 for mDNS.
Sep 03 11:47:22 rpi2b avahi-daemon[324]: Registering new address record for fe80::3c44:852f:9baa:add5 on eth0.*.
Sep 03 11:47:22 rpi2b dhcpcd[385]: eth0: soliciting an IPv6 router
Sep 03 11:47:22 rpi2b dhcpcd[385]: eth0: soliciting a DHCP lease
Sep 03 11:47:23 rpi2b dhcpcd[385]: eth0: offered 192.168.188.73 from 192.168.188.1
Sep 03 11:47:23 rpi2b dhcpcd[385]: eth0: probing address 192.168.188.73/24
Sep 03 11:47:24 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:25 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:27 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:28 rpi2b dhcpcd[385]: eth0: leased 192.168.188.73 for 864000 seconds
Sep 03 11:47:28 rpi2b avahi-daemon[324]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.188.73.
Sep 03 11:47:28 rpi2b avahi-daemon[324]: New relevant interface eth0.IPv4 for mDNS.
Sep 03 11:47:28 rpi2b dhcpcd[385]: eth0: adding route to 192.168.188.0/24
Sep 03 11:47:28 rpi2b avahi-daemon[324]: Registering new address record for 192.168.188.73 on eth0.IPv4.
Sep 03 11:47:28 rpi2b dhcpcd[385]: eth0: adding default route via 192.168.188.1
Sep 03 11:47:28 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:29 rpi2b kernel: R8188EU: INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 0
Sep 03 11:47:36 rpi2b dhcpcd[385]: eth0: no IPv6 Routers available
Sep 03 11:47:43 rpi2b snmpd[415]: error on subcontainer 'ia_addr' insert (-1)
Sep 03 11:48:13 rpi2b snmpd[415]: error on subcontainer 'ia_addr' insert (-1)
Sep 03 11:48:31 rpi2b sshd[15442]: Accepted password for -user- from 192.168.188.21 port 55971 ssh2
```

Any other options I can try?

lwfinger, 2022-09-03:
I did not think that would help. My interpretation is that the software does not roam correctly.

You might try v5.2.2.4. Use 'git checkout origin/v5.2.2.4 -b v5.2.2.5' to get it.

rhermsen, 2022-09-03:
See your update crossed my edit.
I will wait for the next instability to try the v5.2.2.4 version.

rhermsen, 2022-09-08:
Quick update:
Last night there was a new instability after being stable for more than 5 days.

With the second patch I did see ```INFO sta recv deauth reason code(6) sta:-bssid_7590_2_4GHz-, ignore = 1``` (i.s.o. ignore = 0), but other than that the other behaviours are the same (float of messages of 1 a second, wifi link down until eth0 was connected).
Now I installed ```v5.2.2.4``` and will monitor the stability.

rhermsen, 2022-09-13:
Instead of ```deauth reason code 6``` I have now a ```deauth reason code 8```
(Disassociated because sending station is leaving (or has left) BSS) with 10-15 minute impact. Is this something to look at?

New v5.2.2.4 driver was installed on 2022-09-08 at 21:27:23
```
$ sudo journalctl -S 2022-09-09 | grep "CTRL-EVENT-DISCONNECTED\|Associated with\|CTRL-EVENT-CONNECTED\|Trying to associate\|reason code"
Sep 10 05:11:56 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-DISCONNECTED bssid=-bssid_7590_2_4GHz- reason=2
Sep 10 05:11:57 rpi2b wpa_supplicant[388]: wlan0: Trying to associate with -bssid_7490_2_4GHz- (SSID='-myssid-7' freq=2472 MHz)
Sep 10 05:12:00 rpi2b wpa_supplicant[388]: wlan0: Associated with -bssid_7490_2_4GHz-
Sep 10 05:12:00 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-CONNECTED - Connection to -bssid_7490_2_4GHz- completed [id=0 id_str=]
Sep 11 00:22:42 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-DISCONNECTED bssid=-bssid_7490_2_4GHz- reason=8
Sep 11 00:22:43 rpi2b wpa_supplicant[388]: wlan0: Trying to associate with -bssid_7590_2_4GHz- (SSID='-myssid-7' freq=2462 MHz)
Sep 11 00:22:44 rpi2b wpa_supplicant[388]: wlan0: Associated with -bssid_7590_2_4GHz-
Sep 11 00:22:45 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-CONNECTED - Connection to -bssid_7590_2_4GHz- completed [id=0 id_str=]
```
I had a look at the raw wifi capture during this event and suspect it started with a ```wlan.fc.type_subtype == 0x000a``` (Disassociate).

rhermsen, 2022-10-15:
Closing this issue.

While running on the v5.2.2.4 driver I have not seen any `deauth reason code 6` (only reason=2 and reason=8).

Last events I have seen:
```
$ sudo journalctl -S 2022-09-09 | grep "CTRL-EVENT-DISCONNECTED\|Associated with\|CTRL-EVENT-CONNECTE
D\|Trying to associate\|reason code"
Sep 10 05:11:56 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-DISCONNECTED bssid=-bssid_7590_2_4GHz- reason=2
Sep 10 05:11:57 rpi2b wpa_supplicant[388]: wlan0: Trying to associate with -bssid_7490_2_4GHz- (SSID='-myssid-7' freq=2472 MHz)
Sep 10 05:12:00 rpi2b wpa_supplicant[388]: wlan0: Associated with -bssid_7490_2_4GHz-
Sep 10 05:12:00 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-CONNECTED - Connection to -bssid_7490_2_4GHz- completed [id=0 id_str=]
Sep 11 00:22:42 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-DISCONNECTED bssid=-bssid_7490_2_4GHz- reason=8
Sep 11 00:22:43 rpi2b wpa_supplicant[388]: wlan0: Trying to associate with -bssid_7590_2_4GHz- (SSID='-myssid-7' freq=2462 MHz)
Sep 11 00:22:44 rpi2b wpa_supplicant[388]: wlan0: Associated with -bssid_7590_2_4GHz-
Sep 11 00:22:45 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-CONNECTED - Connection to -bssid_7590_2_4GHz- completed [id=0 id_str=]
Sep 16 18:52:44 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-DISCONNECTED bssid=-bssid_7590_2_4GHz- reason=2
Sep 16 18:52:45 rpi2b wpa_supplicant[388]: wlan0: Trying to associate with -bssid_7490_2_4GHz- (SSID='-myssid-7' freq=2472 MHz)
Sep 16 18:52:49 rpi2b wpa_supplicant[388]: wlan0: Associated with -bssid_7490_2_4GHz-
Sep 16 18:52:50 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-CONNECTED - Connection to -bssid_7490_2_4GHz- completed [id=0 id_str=]
Sep 17 09:17:32 rpi2b wpa_supplicant[388]: wlan0: Trying to associate with -bssid_7590_2_4GHz- (SSID='-myssid-7' freq=2472 MHz)
Sep 17 09:17:33 rpi2b wpa_supplicant[388]: wlan0: Associated with -bssid_7590_2_4GHz-
Sep 17 09:17:33 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-CONNECTED - Connection to -bssid_7590_2_4GHz- completed [id=0 id_str=]
Oct 08 16:35:59 rpi2b wpa_supplicant[388]: wlan0: CTRL-EVENT-DISCONNECTED bssid=-bssid_7590_2_4GHz- reason=3 locally_generated=1
```

closed 2022-10-15 by rhermsen