lwfinger/rtl8188eu #421: Segmentation fault during compilation

thalesmaoa, 2022-09-09:
Hi there, I'm trying to compile the driver for a Raspberry, but I'm getting Segmentation Error during rtw_cmd.o compilation.

```
$ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.95-rt38/build M=/home/pi/rdriver/rtl8188eu  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.95-rt38'
  CC [M]  /home/pi/rdriver/rtl8188eu/core/rtw_ap.o
/bin/bash: line 1:  2412 Segmentation fault      scripts/basic/fixdep /home/pi/rdriver/rtl8188eu/core/.rtw_ap.o.d /home/pi/rdriver/rtl8188eu/core/rtw_ap.o 'gcc -Wp,-MD,/home/pi/rdriver/rtl8188eu/core/.rtw_ap.o.d  -nostdinc -isystem /usr/lib/gcc/arm-linux-gnueabihf/6/include -I./arch/arm/include -I./arch/arm/include/generated  -I./include -I./arch/arm/include/uapi -I./arch/arm/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -mlittle-endian -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -fshort-wchar -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -fno-dwarf2-cfi-asm -fno-omit-frame-pointer -mapcs -mno-sched-prolog -fno-ipa-sra -mabi=aapcs-linux -mfpu=vfp -funwind-tables -marm -Wa,-mno-warn-deprecated -D__LINUX_ARM_ARCH__=6 -march=armv6k -mtune=arm1136j-s -msoft-float -Uarm -fno-delete-null-pointer-checks -Wno-frame-address -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fstack-protector-strong -Wno-unused-but-set-variable -Wno-unused-const-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -pg -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fno-merge-all-constants -fmerge-constants -fno-stack-check -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -D__CHECK_ENDIAN__ -O1 -Wno-unused-variable -Wno-unused-value -Wno-unused-label -Wno-unused-parameter -Wno-unused-function -Wno-unused -Wno-uninitialized -I/home/pi/rdriver/rtl8188eu/include  -DMODULE  -DKBUILD_BASENAME='\''"rtw_ap"'\'' -DKBUILD_MODNAME='\''"8188eu"'\'' -c -o /home/pi/rdriver/rtl8188eu/core/.tmp_rtw_ap.o /home/pi/rdriver/rtl8188eu/core/rtw_ap.c' > /home/pi/rdriver/rtl8188eu/core/.rtw_ap.o.tmp
scripts/Makefile.build:310: recipe for target '/home/pi/rdriver/rtl8188eu/core/rtw_ap.o' failed
make[2]: *** [/home/pi/rdriver/rtl8188eu/core/rtw_ap.o] Error 139
Makefile:1546: recipe for target '_module_/home/pi/rdriver/rtl8188eu' failed
make[1]: *** [_module_/home/pi/rdriver/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.95-rt38'
Makefile:155: recipe for target 'modules' failed
make: *** [modules] Error 2
```
Any help about debugging?

lwfinger, 2022-09-09:
This problem has not been reported, and I do not know enough about the PI to offer a fix. How much RAM? Is swap enabled? Is it running out of space on the SDI card? What version of GCC? What distro?

I think the build on kernel 4.19 should be OK. That file is under 2000 lines of code and should not be difficult to compile.

thalesmaoa, 2022-09-09:
Indeed. This is a compute module. I've tested on x86, and it worked.

The version is `amrv6l`. The distro is Jesse, raspibian. I'm using eMMC and not running out of space. 1GB of RAM.

thalesmaoa, 2022-09-10:
Here is the full information (last night I was erasing everything. With a fresh new system, the error persist):

```
$ lscpu
Architecture:        armv6l
Byte Order:          Little Endian
CPU(s):              1
On-line CPU(s) list: 0
Thread(s) per core:  1
Core(s) per socket:  1
Socket(s):           1
Vendor ID:           ARM
Model:               7
Model name:          ARM1176
Stepping:            r0p7
CPU max MHz:         700.0000
CPU min MHz:         700.0000
BogoMIPS:            697.95
Flags:               half thumb fastmult vfp edsp java tls

$ lsusb
Bus 001 Device 007: ID 2357:010c TP-Link TL-WN722N v2
Bus 001 Device 006: ID 413c:2106 Dell Computer Corp. Dell QuietKey Keyboard
Bus 001 Device 005: ID 0781:5567 SanDisk Corp. Cruzer Blade
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ free -m
              total        used        free      shared  buff/cache   available
Mem:            429          85         101           5         242         287
Swap:             0           0           0
```

I only realize now that I have 512MB of RAM and 101MB free and no swap. I will try to enable swap and see if anything changes.

thalesmaoa, 2022-09-10:
With swap on, same error.

lwfinger, 2022-09-10:
I had a Pi 2B on the shelf. It booted into Raspbian, but the update data base was corrupted. When I tried to write a new version to the SD card, it would not boot! In trying various things, I smoked, at least, the SD card. It is now in the trash!

I then tried to set up a virtual machine using QEMU. With that setup, I could not get headers that matched the running kernel, thus I could not compile the driver.

It does not appear that I will be able to help you. Sorry.

thalesmaoa, 2022-09-10:
I appreciate it anyway. I'm trying cross compilation now. It does compile, however I'm missing something for kernel matching. I'm using `modprobe configs` now. Thanks.

thalesmaoa, 2022-09-10:
Sucess! It is working. Now I'm facing another problem. wpa_supplicant is not connecting. If I specify the driver `-D8188eu`, it does work, but I'm not sure if I can edit the services.

```
$ ll wpa_supplicant*
-rw-r--r-- 1 root root 486 Apr 16  2021 wpa_supplicant-nl80211@.service
-rw-r--r-- 1 root root 286 Apr 16  2021 wpa_supplicant.service
-rw-r--r-- 1 root root 455 Apr 16  2021 wpa_supplicant@.service
-rw-r--r-- 1 root root 478 Apr 16  2021 wpa_supplicant-wired@.service
```

But only wpa_supplicant.service is enabled, in which, driver is not specified. Will it broke if I add -D to the main service?

lwfinger, 2022-09-11:
I have no idea what the wpa_supplicant service is like as I use NetworkManager, which communicates with the supplicant over the D bus. As the service file likely starts the supplicant, adding that switch should work.

thalesmaoa, 2022-09-12:
There is a bug in buster version, I just needed to set:

```
interface wlan0
        metric 100
	env ifwireless=1
	env wpa_supplicant_driver=8188eu,wext
```

Now, I will try to make AP Mode work.

closed 2022-09-12 by thalesmaoa