lwfinger/rtl8188eu #422: hostapd - Kernel 5.10 - RPi Buster

thalesmaoa, 2022-09-12:
Hi @lwfinger, after reading many alike issues, I've tested master, v5.2 and v4.1 branchs.

Only v4.1 branch successfully launched, however, I can't connect. Any idea?
The network doesn't even show for connection.

```
$ sudo ./hostapd -dd /etc/hostapd/hostapd.conf 
random: Trying to read entropy from /dev/random
Configuration file: /etc/hostapd/hostapd.conf
drv->ifindex=5
l2_sock_recv==l2_sock_xmit=0x0x4fe6d0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=149 freq=5745 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=153 freq=5765 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=157 freq=5785 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=161 freq=5805 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=165 freq=5825 MHz max_tx_power=0 dBm
Completing interface initialization
Mode: IEEE 802.11g  Channel: 13  Frequency: 2472 MHz
RATE[0] rate=10 flags=0x1
RATE[1] rate=20 flags=0x1
RATE[2] rate=55 flags=0x1
RATE[3] rate=110 flags=0x1
RATE[4] rate=60 flags=0x0
RATE[5] rate=90 flags=0x0
RATE[6] rate=120 flags=0x0
RATE[7] rate=180 flags=0x0
RATE[8] rate=240 flags=0x0
RATE[9] rate=360 flags=0x0
RATE[10] rate=480 flags=0x0
RATE[11] rate=540 flags=0x0
Flushing old station entries
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.
Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan1 keyidx=0)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan1 keyidx=1)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan1 keyidx=2)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Failed to clear default encryption keys (ifname=wlan1 keyidx=3)
Using interface wlan1 with hwaddr c0:06:c3:d1:6f:fb and ssid 'TesteTeste'
Deriving WPA PSK based on passphrase
SSID - hexdump_ascii(len=10):
     54 65 73 74 65 54 65 73 74 65                     TesteTeste      
PSK (ASCII passphrase) - hexdump_ascii(len=10): [REMOVED]
PSK (from passphrase) - hexdump(len=32): [REMOVED]
rtl871x_set_wps_assoc_resp_ie
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
rtl871x_set_beacon_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set beacon head/tail or DTIM period
wlan1: Setup of interface done.
random: Got 20/20 bytes from /dev/random
```

```
$ cat /etc/hostapd/hostapd.conf 
interface=wlan1
ssid=TesteTeste
wpa_passphrase=TesteTeste
driver=rtl871xdrv
hw_mode=g
channel=13
ctrl_interface=/var/run/hostapd
```

```
$ modinfo 8188eu
filename:       /lib/modules/5.10.120-rt70/kernel/drivers/staging/r8188eu/8188eu.ko
version:        v4.1.4_6773.20130222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     902D710EB21C96BBA2643B2
alias:          usb:v7392pB811d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p18F0d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2C4Ep0102d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0111d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4008d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p331Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3311d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3310d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0076d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDApF179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
depends:        
name:           8188eu
vermagic:       5.10.120-rt70+ preempt_rt mod_unload modversions ARMv6 p2v8 
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_initmac:charp
parm:           rtw_channel_plan:int
parm:           rtw_chip_version:int
parm:           rtw_rfintfs:int
parm:           rtw_lbkmode:int
parm:           rtw_network_mode:int
parm:           rtw_channel:int
parm:           rtw_mp_mode:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_busy_thresh:int
parm:           rtw_led_enable:int
parm:           rtw_ht_enable:int
parm:           rtw_cbw40_enable:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_rf_config:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           debug:Set debug level (1-9) (default 1) (int)
```

lwfinger, 2022-09-13:
Do you mean that you cannot create an AP? If so, see

ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.

and

driver=rtl871xdrv

Version 4.1 and 5.2 use nl80211 as their driver interface in hostapd.

thalesmaoa, 2022-09-13:
If I use nl80211, it says the driver is unknown or invalid.

lwfinger, 2022-09-13:
Your hostapd is not correct for v4,1 or v5.2. They no longer use the rtl871xdrv driver.

thalesmaoa, 2022-09-14:
I'm confused. Let me explain my steps.

First, testing branch v5.2. I can't compile:

```
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/../linux M=/home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4 modules
make[1]: Entering directory '/home/thales/rpi_rtl_compile/linux'
  CC [M]  /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/rtw_cmd.o
In file included from ./include/linux/spinlock.h:328:0,
                 from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:16,
                 from /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/basic_types.h:13,
                 from /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/drv_types.h:15,
                 from /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/rtw_cmd.c:6:
/home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/osdep_service_linux.h: In function ‘_exit_critical_ex’:
/home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/osdep_service_linux.h:248:25: error: passing argument 1 of ‘rt_spin_unlock’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  spin_unlock_irqrestore(plock, *pirqL);
```

After that, I've tried branch v4.1

_with nl80211 driver_
```
~/rtl8188eu_v4.1.8_949/hostapd-0.8/hostapd $ sudo ./hostapd -dd ~/RTL8188-hostapd-2.0/hostapd/forum.conf 
random: Trying to read entropy from /dev/random
Configuration file: /home/pi/RTL8188-hostapd-2.0/hostapd/forum.conf
Line 15: invalid/unknown driver 'nl80211'
1 errors found in configuration file '/home/pi/RTL8188-hostapd-2.0/hostapd/forum.conf'
```

_with rtl871xdrv driver (ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported)_ but it **doesn't crash**. No AP is shown however (checking with Android Phone), but TP-Link TL-WN722N v2 keeps blink like if it was working.
```
~/rtl8188eu_v4.1.8_949/hostapd-0.8/hostapd $ sudo ./hostapd -dd ~/RTL8188-hostapd-2.0/hostapd/forum.conf 
random: Trying to read entropy from /dev/random
Configuration file: /home/pi/RTL8188-hostapd-2.0/hostapd/forum.conf
drv->ifindex=9
l2_sock_recv==l2_sock_xmit=0x0x6fd6e8
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=149 freq=5745 MHz max_tx_power=0 dBm
[... 51 lines cut ...]
Key Counter - hexdump(len=32): [REMOVED]
WPA: group state machine entering state GTK_INIT (VLAN-ID 0)
Get randomness: len=16 entropy=0
GTK - hexdump(len=16): [REMOVED]
WPA: group state machine entering state SETKEYSDONE (VLAN-ID 0)
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_beacon_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set beacon head/tail or DTIM period
wlan1: Setup of interface done.
Wireless event: cmd=0x8b06 len=8
```

Please, notice that this is the version within the branch.

I've start digging and found this:

> It seems that the driver does support hostapd's driver=nl80211 but you have to explicitly enable it in the makefile (it's enabled for several architectures, but left out for some like aarch64, not sure why).
You just have to put this line inside the ifeq block:
`EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT`
https://issuecloser.com/blog/can-i-run-hostapd-with-this-driver

I have a `armv6l`. I'm start to think that I'm the one with a special processor. I can't get it to work. I've been trying it got hours.

thalesmaoa, 2022-09-14:
I can't believe, but I finally got it. Let me share the problem:

The major problem seems to be with the raspberry build. You`re script install the drive at:

```
/lib/modules/5.10.120-rt70/kernel/drivers/staging/
```

But it needed to be at

```
/lib/modules/5.10.120-rt70/kernel/net/wireless/
```

I noticed that a few days ago, but every time I've deleted the module, it appears again. I needed to delete the entire folder, manually move and now it is work.

I hope it survive a reboot.

Thanks!

lwfinger, 2022-09-14:
Most distros want the driver to be installed at /lib/modules/..../kernel/drivers/staging/, not at ..../kernel/net/wireless.

If you want it done in a non-standard way, then you will need to change the Makefile.

thalesmaoa, 2022-09-15:
Same error:

```
$ git pull
remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (1/1), done.
remote: Total 6 (delta 5), reused 6 (delta 5), pack-reused 0
Unpacking objects: 100% (6/6), done.
From https://github.com/lwfinger/rtl8188eu
   8f5cee5..7c6e80d  v5.2.2.4   -> origin/v5.2.2.4
Updating 8f5cee5..7c6e80d
Fast-forward
 osdep_service_linux.h |  5 -----
 rtw_mlme_ext.c        |  2 +-
 rtw_recv.c            | 10 +++++-----
 rtw_xmit.c            |  4 ++--
 4 files changed, 8 insertions(+), 13 deletions(-)
thales@thales-OptiPlex-380:~/rpi_rtl_compile/rtl8188eu_v5.2.2.4$ make clean
#make -C /lib/modules/4.15.0-192-generic/build M=/home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4 clean
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
/bin/sh: 1: cd: can't cd to core
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
thales@thales-OptiPlex-380:~/rpi_rtl_compile/rtl8188eu_v5.2.2.4$ make RPI_build 
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/../linux M=/home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4 modules
make[1]: Entering directory '/home/thales/rpi_rtl_compile/linux'
  CC [M]  /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/rtw_cmd.o
In file included from ./include/linux/spinlock.h:328:0,
                 from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:6,
                 from ./include/linux/umh.h:4,
                 from ./include/linux/kmod.h:9,
                 from ./include/linux/module.h:16,
                 from /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/basic_types.h:13,
                 from /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/drv_types.h:15,
                 from /home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/rtw_cmd.c:6:
/home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/osdep_service_linux.h: In function ‘_enter_critical_bh’:
/home/thales/rpi_rtl_compile/rtl8188eu_v5.2.2.4/osdep_service_linux.h:248:15: error: passing argument 1 of ‘rt_spin_lock’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  spin_lock_bh(plock);
```

Nerixyz, 2022-12-27:
> The major problem seems to be with the raspberry build. You`re script install the drive at:
>
> ```
> /lib/modules/5.10.120-rt70/kernel/drivers/staging/
> ```
>
> But it needed to be at
>
> ``` [...]

Thanks for sharing this, it worked for me as well. **Make sure to remove any blacklist entry for this driver in `/etc/modprobe.d/`**.

I changed 
https://github.com/lwfinger/rtl8188eu/blob/51051d2e4b1335042b85903fcf2b540002d6743e/Makefile#L347
to 
```Makefile
MODDESTDIR := /lib/modules/$(KVER)/kernel/net/wireless/r8188eu/
```

lwfinger, 2022-12-27:
That is not a change that I can make for every distro. You will have to keep that as a private change.

closed 2022-12-27 by lwfinger