lwfinger/rtl8188eu #428: Complete system freeze with TP-Link wi-fi dongle TL-WN725N

simurq, 2022-10-27:
I have git-cloned, made and installed the driver as instructed. Files `8188eu.ko` and `rtl8188eufw.bin` are moved to their respective directories properly. Restarted the computer and can see Wi-Fi connected and working as usual too.

However, the computer freezes completely whenever the browser is loading page, or there is a network connection (e.g. some applications checking for updates on launch). Tested on different browsers (Firefox, Chrome, Brave) with the same result. For some time browsing the web goes well, but the suddenly the whole system freezes for no obvious reason with none of the peripherals (keyboard, mouse, etc.) working/accessible - I cannot even `Ctrl+Alt+Fx` to get into console mode.

PS: It's funny that most of the time the freeze occurs when the browser is loading Reddit.

System specifications:
-------------------------
Linux kernel: **6.0.2**
Operating system: **Pop!_OS 22.04 LTS**

Anything else I can provide?

Any help is appreciated! Thank you!

lwfinger, 2022-10-27:
You did not specify what branch you are running. Try the other.

simurq, 2022-10-27:
I just noticed that this repository had 6 branches, didn't know that.

But I simply established the chipset driver my dongle uses and simply `git clone https://github.com/lwfinger/rtl8188eu`. Then followed your instructions for making and installing the driver: `make all` and `sudo make install`. That's all I did.

simurq, 2022-10-28:
Thank you very much! It works well, no freezes even on Github and Reddit (hopefully in the future as well). 

Two more questions:

1- Can you leave the issue open for another couple of days until I complete test-running the system?
2- Do I need to re-download and update the driver with each kernel update?

lwfinger, 2022-10-28:
At a minimum, you need to 'make' and 'sudo make install' with each kernel update. It is sensible to do a 'git pull' before the make just to be sure you get the latest code. A full download is needed only once!

simurq, 2022-10-31:
Test running is complete, the issue may be closed. No freezing anymore after 3 days of running the driver from branch 5.2.2.4. Thank you very much for your generous support, Larry!