lwfinger/rtl8188eu #429: Can't install driver on Volumio image (Rasbian buster. Kernel: 5.10.92+)

zydariv, 2022-11-20:
Hey guys,

**First:** Thank you for your work on this driver!

**Second:**

Here is some system information:
```
uname -r
5.10.92+
```
```
lsb_release -a 

No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
```

Before installation of the driver I installed the raspberry headers and build essentials
`sudo apt install raspberrypi-kernel-headers`
`sudo apt-get install build-essential`

Then I followed the instructions: `clone`, `make` and `make install`.

Then I got the following error:
```
sudo make all
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.92+/build M=/home/volumio/rtl8188eu  modules
make[1]: *** /lib/modules/5.10.92+/build: No such file or directory.  Stop.
make: *** [Makefile:155: modules] Error 2
```
Even though I installed the header files, there is no build directory. in /lib/modules/5.10.92+ .

With the following symlink I was abble to compile, but I couldn't  `modprobe` or `insmod` the module.
`sudo ln -s /lib/modules/5.10.103+/build /lib/modules/5.10.92+`
```
sudo insmod 8188eu.ko
insmod: ERROR: could not insert module 8188eu.ko: Invalid module format
sudo modprobe 8188eu
modprobe: ERROR: could not insert '8188eu': Exec format error
```

I think the symlink is incorrect, but I also don't know how to build for my kernel version.

Thank you in advance!
cheers!
zydariv

lwfinger, 2022-11-20:
The link needs to go from /lib/...../build to the directory that has your header files. In your case, it is likely /usr/src/linux-5.10.92/
If not that directory, then check others under /usr/src/.

zydariv, 2022-11-24:
Thank you for you answer. Unfortunately, I don't have 5.10.92 in my /usr/src/ .

```
ls /usr/src
linux-headers-5.10.103+  linux-headers-5.10.103-v7+  linux-headers-5.10.103-v7l+
```

```
tree /usr/src/ -L 2
/usr/src/
|-- linux-headers-5.10.103+
|   |-- Documentation
|   |-- Kconfig
|   |-- Makefile
|   |-- Module.symvers
|   |-- arch
|   |-- block
|   |-- certs
|   |-- crypto
|   |-- drivers
|   |-- fs
|   |-- include
|   |-- init
|   |-- ipc
|   |-- kernel
|   |-- lib
|   |-- mm
|   |-- net
|   |-- samples
|   |-- scripts
|   |-- security
|   |-- sound
|   |-- tools
|   |-- usr
|   `-- virt
|-- linux-headers-5.10.103-v7+
|   |-- Documentation
|   |-- Kconfig
|   |-- Makefile
|   |-- Module.symvers
|   |-- arch
|   |-- block
|   |-- certs
|   |-- crypto
|   |-- drivers
|   |-- fs
|   |-- include
|   |-- init
|   |-- ipc
|   |-- kernel
|   |-- lib
|   |-- mm
|   |-- net
|   |-- samples
|   |-- scripts
|   |-- security
|   |-- sound
|   |-- tools
|   |-- usr
|   `-- virt
`-- linux-headers-5.10.103-v7l+
    |-- Documentation
    |-- Kconfig
    |-- Makefile
    |-- Module.symvers
    |-- arch
    |-- block
    |-- certs
    |-- crypto
    |-- drivers
    |-- fs
    |-- include
    |-- init
    |-- ipc
    |-- kernel
    |-- lib
    |-- mm
    |-- net
    |-- samples
    |-- scripts
    |-- security
    |-- sound
    |-- tools
    |-- usr
    `-- virt

```

lwfinger, 2022-11-25:
It has to link to whatever 'uname -r' shows as your present kernel. It looks as though you do not have the latest kernel as the headers are for 5.10.103+, not 5.10.92+.