lwfinger/rtl8188eu #430: modprobe: FATAL: Module 8188eu.ko not found in directory /lib/modules/5.10.0-19-amd64

Pablompg, 2022-11-21:
I have bought a tp-link AC1300 and I wanted to use it on a Debian 11 server with kernel 5.10.0-19-amd64.

I have followed [this instructions](https://vitux.com/install-tp-link-usb-wifi-adapter-to-debian-10/) which basically are:

1. `apt install linux-headers-$(uname -r)`
2. `apt-get install build-essential`
3. Clone this repo and cd into it
4. `sudo make all`
5. `sudo make install`
6. `modprobe 8188eu.ko`

None of the steps gave any error and looked successfull except for the last one, which gives the error provided in the title:

> modprobe: FATAL: Module 8188eu.ko not found in directory /lib/modules/5.10.0-19-amd64

Is there anything I am doing wrong or missing?

Please let me know if further information about my system is required and thank you in advance for your time.

lwfinger, 2022-11-21:
That should have worked. Please show me the output of 'find /lib/modules/ -name 8188eu.ko', 'uname -r', and the line that starts with 'make -C' when you run make.

Pablompg, 2022-11-21:
Hi @lwfinger, thanks for taking the time to have a look at this.

`find /lib/modules/ -name 8188eu.ko` returns 

> lib/modules/5.10.0-19-amd64/kernel/drivers/staging/r8188eu/8188eu.ko

`uname -r` returns

> 5.10.0-19-amd64

These are the results when running:

1. `sudo make all`

> make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-19-amd64/build M=/home/username/rt181888eu modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-19-adm64'
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-19-amd64'

2. `sudo make install`

> install -p -m 644 8188eu.ko /lib/modules/5.10.0-19-amd64/kernel/drivers/staging/r8188eu/
/sbin/depmod -a 5.10.0-19-amd64
cp rt18188eufw.bin /lib/firmware/rtlwifi/.

lwfinger, 2022-11-21:
Your headers match the kernel, AND the .ko file is exactly where it should be. I have no idea why modprobe cannot find it.

One last desperate act. Do the command
sudo ln -s lib/modules/5.10.0-19-amd64/kernel/drivers/staging/r8188eu/8188eu.ko /lib/modules/5.10.0-19-amd64/8188eu.ko

If that works, it is because modprobe is not searching the entire drivers tree.

Pablompg, 2022-11-21:
No luck, it fails to create the link:

> ln: failed to create symbolic link './8188eu-ko': File exists

sodapng, 2022-11-21:
Try https://github.com/aircrack-ng/rtl8188eus

Pablompg, 2022-11-21:
> 

Thanks for the suggestion. Unfortunately, it does not work either

lwfinger, 2022-11-21:
Actually, you should not use the .ko on the modprobe command. With your version, it is looking for 8188eu.ko.ko!

Sorry I missed that before. I booted my only Debian instance, which is a powerpc running Debian 12. With the correct command, it works, but fails with your version.

Pablompg, 2022-11-22:
Thanks for the suggestion again. However it is still not working:

> modprobe: FATAL: Module 8188eu.ko not found in directory /lib/modules/5.10.0-19-amd64

The server has an **_intel i5 6500T_**. Might it be because of incompatibilities with the CPU?

lwfinger, 2022-11-22:
Show me EXACTLY what you are doing starting with make.