lwfinger/rtl8188eu #435: Armbian编译RTL188EU驱动失败

cui8, 2022-12-11:
大佬，我想在这个[系统](https://github.com/ophub/amlogic-s9xxx-armbian)上编译您的驱动，但是出现错误，不知道该怎么办，求大佬解答。
```
root@armbian:~/rtl8188eu# make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.15.79-ophub/build M=/root/rtl8188eu  modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.79-ophub'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: Ubuntu clang version 14.0.0-1ubuntu1
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
  CC [M]  /root/rtl8188eu/core/rtw_ap.o
gcc: error: unrecognized command-line option ‘-Qunused-arguments’
gcc: error: unrecognized command-line option ‘-mno-global-merge’
gcc: error: unrecognized command-line option ‘-fsplit-lto-unit’
make[2]: *** [scripts/Makefile.build:289: /root/rtl8188eu/core/rtw_ap.o] Error 1
make[1]: *** [Makefile:1902: /root/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.79-ophub'
make: *** [Makefile:155: modules] Error 2
```

cui8, 2022-12-11:
[image]

大佬，这个错误怎么解决？百度没有找到解决办法。求大佬回复

lwfinger, 2022-12-12:
Your problem is shown on these two lines:
  The kernel was built by: Ubuntu clang version 14.0.0-1ubuntu1
  You are using:           gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.04

The unrecognozed command-line options arise because you system is set up to use clang, but something is setting symbol "CC" to point to gcc. You will need to contact Ubuntu's ARM experts to find out how to select "clang" rather than "gcc". I am not an Ubuntu user and no nothing of their configuration files.

callMeMakerRen, 2022-12-13:
`make ARCH=arm64 CC=clang LD=ld.lld xxx` ，have a try

cui8, 2022-12-16:
> 您的问题显示在以下两行中： 内核由以下人员构建： Ubuntu clang 版本 14.0.0-1ubuntu1 您正在使用： gcc （Ubuntu 11.3.0-1ubuntu1~22.04） 11.3.04
> 
> 出现未识别的命令行选项是因为您的系统设置为使用 clang，但某些内容正在设置符号“CC”以指向 gcc。您需要联系 Ubuntu 的 ARM 专家，了解如何选择“clang”而不是“gcc”。我不是 Ubuntu 用户，也没有他们的配置文件。

[image]

大佬，这个错误怎么解决？百度没有找到解决办法。求大佬回复

[image]

cui8, 2022-12-16:
> `make ARCH=arm64 CC=clang LD=ld.lld xxx`，试一试

```
root@armbian:/usr/local# cd /opt/rtl8188eu/
root@armbian:/opt/rtl8188eu# export CC="clang"
root@armbian:/opt/rtl8188eu# export LD="ld.lld"
root@armbian:/opt/rtl8188eu# export ARCH="arm64"
root@armbian:/opt/rtl8188eu# MAKE_SET_STRING=" ARCH=${ARCH} CC=${CC} LD=${LD} LLVM=1 LLVM_IAS=1 "

root@armbian:/opt/rtl8188eu# make ${MAKE_SET_STRING} all
```

出现了什么图片的问题，求大佬解答

lwfinger, 2022-12-16:
I am not quite certain why you get the "Exec format error" for scripts/basic/fixdep. My guess is that it was built for x86-64, not arm64. From the kernel source directory, a 'sudo make prepare' should rebuild that utility. From what I have seen, I do not know if the kernel source is installed.

cui8, 2022-12-17:
> 我不太确定为什么你会得到脚本/基本/修复的“执行格式错误”。我的猜测是它是为 x86-64 构建的，而不是 arm64。从内核源代码目录中，“sudo make prepare”应该重建该实用程序。从我所看到的，我不知道是否安装了内核源代码。

[image]
大佬，这个错误怎么解决

lwfinger, 2022-12-17:
That library should have been installed with a 'sudo apt install build-essential', but if not 'sudo apt install libc6' will install that library,

shutup, 2022-12-23:
just modify the makefile
```
modules:
	#$(MAKE) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) -C $(KSRC) M=$(shell pwd)  modules
	$(MAKE) ARCH=$(ARCH) CC=clang LD=ld.lld -C $(KSRC) M=$(shell pwd)  modules
```
and you are ready to go. just use for build on arm64 environment.
[image]

cui8, 2022-12-23:
> just modify the makefile
> 
> ```
> modules:
> 	#$(MAKE) ARCH=$(ARCH) CROSS_COMPILE=$(CROSS_COMPILE) -C $(KSRC) M=$(shell pwd)  modules
> 	$(MAKE) ARCH=$(ARCH) CC=clang LD=ld.lld -C $(KSRC) M=$(shell pwd)  modules
> ```
> 
> and you are ready to go. just use for build on arm64 environment. [image]

我已经修改了makefile文件
```
cd /rtl8188eu
vim Makefile
```
修改Makefile文件为下图：
[image]
但是编译还是报错，我觉得是编译环境问题，求解答
[image]
大佬，ld.lld这个错误怎么解决？

shutup, 2022-12-26:
build一下内核，会自动安装依赖，参考ophub的文档

cui8, 2023-01-01:
> build一下内核，会自动安装依赖，参考ophub的文档

[image]
求大佬帮忙看下这个错误怎么解决？

cui8, 2023-01-01:
> build一下内核，会自动安装依赖，参考ophub的文档

[image]
使用的Armbian_23.02.0_amlogic_s905l3a_bullseye_5.15.82_server_2022.12.12.img镜像出现的错误

lwfinger, 2023-01-01:
Your setup is screwed up. I cannot help you any more.

closed 2023-01-01 by lwfinger