lwfinger/rtl8188eu #441: Crash when switch to  iwd (ubuntu 22.10)

iafilius, 2023-04-09:
Hi Larry ea,

Running ubuntu 22.10 and the v5.2.2.4 to run a hotspot on my realtek micro usb rtl8188
This worked perfectly fine until i switched from wpa_supplicant to more
recent iwd.

The summary of the issue is: while having iwd running fine with one wireless
adapter (AX200), then inserting the RTL8188 usb adapter results in a kernel
crash dump, and also leaves iwd in unusable state.

I'm not sure if this is an iwd or driver issue, or combination.

Using your v5.2.2.4 driver from git
```
$ uname -a
Linux host 5.19.0-38-generic #39-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 17 17:33:16 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

$ git status
On branch v5.2.2.4
Your branch is up to date with 'origin/v5.2.2.4'.

~/git/rtl8188eu$ cat .git/config 
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[remote "origin"]
        url = https://github.com/lwfinger/rtl8188eu.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
        remote = origin
        merge = refs/heads/master
[branch "v5.2.2.4"]
        remote = origin
        merge = refs/heads/v5.2.2.4

user@host:~/git/rtl8188eu$ git pull
Already up to date.

user@host:~/git/rtl8188eu$ sudo make  install
install -p -m 644 8188eu.ko  /lib/modules/5.19.0-38-generic/kernel/drivers/staging/r8188eu/
/sbin/depmod -a 5.19.0-38-generic

had blacklisted original module:
/etc/modprobe.d/50-8188eu.conf:blacklist r8188eu

```

iwd (iwctl) is running/operating just fine
```
user@host:~/git/rtl8188eu$ ps aux|grep iwd
root        2005  0.0  0.0  15824  5040 ?        Ss   19:34   0:00 /usr/libexec/iwd
```

The oops/call trace at the moment of inserting
```
user@host:~/git/rtl8188eu$ dmesg -TW
[za apr  8 19:40:36 2023] usb 1-1.2: new high-speed USB device number 6 using ehci-pci
[za apr  8 19:40:36 2023] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=0179, bcdDevice= 0.00
[za apr  8 19:40:36 2023] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[za apr  8 19:40:36 2023] usb 1-1.2: Product: 802.11n NIC
[za apr  8 19:40:36 2023] usb 1-1.2: Manufacturer: Realtek
[za apr  8 19:40:36 2023] usb 1-1.2: SerialNumber: 00E02D0B3D19
[za apr  8 19:40:37 2023] RTW: rtl8188eu v5.2.2.4_25483.20171222
[za apr  8 19:40:37 2023] RTW: hal_com_config_channel_plan chplan:0x20
[za apr  8 19:40:37 2023] RTW: rtw_regsty_chk_target_tx_power_valid return false for band:0, path:0, rs:0, t:-1
[za apr  8 19:40:37 2023] usbcore: registered new interface driver rtl8188eu
[za apr  8 19:40:38 2023] BUG: kernel NULL pointer dereference, address: 0000000000000000
[za apr  8 19:40:38 2023] #PF: supervisor read access in kernel mode
[... 63 lines cut ...]

```

Check which module was loaded:
```
user@host:~/git/rtl8188eu$ lsmod|grep 8188
8188eu               1748992  0
cfg80211             1040384  4 iwlmvm,8188eu,iwlwifi,mac80211
```

iwctl does not longer see/work with iwd after crash
```
$ sudo iwctl 
Terminate
Waiting for IWD to start...
<nothing>
```

I had the system to reboot always after, either due to console/screen frozen, or no longer reachable though ssh.

The issue is reproducible for 100% for me.

Could you have a look please? 

Regards,

lwfinger, 2023-04-09:
This kind of error usualy kills the system:

[za apr  8 19:40:38 2023] BUG: kernel NULL pointer dereference, address: 0000000000000000
[za apr  8 19:40:38 2023] #PF: supervisor read access in kernel mode
[za apr  8 19:40:38 2023] #PF: error_code(0x0000) - not-present page

I added a statement that should prevent this, but I have no idea if it will work.

iafilius, 2023-04-10:
Hello Larry thanks for quick response.

just tried it with the update, but same seems to happen while inserting.

cmd summary
```
git pull
make clean
make all
sudo make install
```
dmesg -TW
```
[ma apr 10 08:11:32 2023] usb 1-1.2: new high-speed USB device number 6 using ehci-pci
[ma apr 10 08:11:32 2023] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=0179, bcdDevice= 0.00
[ma apr 10 08:11:32 2023] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ma apr 10 08:11:32 2023] usb 1-1.2: Product: 802.11n NIC
[ma apr 10 08:11:32 2023] usb 1-1.2: Manufacturer: Realtek
[ma apr 10 08:11:32 2023] usb 1-1.2: SerialNumber: 00E02D0B3D19
[ma apr 10 08:11:32 2023] RTW: hal_com_config_channel_plan chplan:0x20
[ma apr 10 08:11:32 2023] RTW: rtw_regsty_chk_target_tx_power_valid return false for band:0, path:0, rs:0, t:-1
[ma apr 10 08:11:33 2023] BUG: kernel NULL pointer dereference, address: 0000000000000000
[ma apr 10 08:11:33 2023] #PF: supervisor read access in kernel mode
[ma apr 10 08:11:33 2023] #PF: error_code(0x0000) - not-present page
[ma apr 10 08:11:33 2023] PGD 0 P4D 0 
[ma apr 10 08:11:33 2023] Oops: 0000 [#1] PREEMPT SMP PTI
[ma apr 10 08:11:33 2023] CPU: 3 PID: 1922 Comm: iwd Tainted: G           OE     5.19.0-38-generic #39-Ubuntu
[ma apr 10 08:11:33 2023] Hardware name: LENOVO 2349G7G/2349G7G, BIOS G1ETC2WW (2.82 ) 08/07/2019
[ma apr 10 08:11:33 2023] RIP: 0010:memcmp+0x2e/0x60
[ma apr 10 08:11:33 2023] Code: 06 48 39 07 75 17 48 83 c7 08 48 83 c6 08 48 83 ea 08 48 83 fa 07 77 e6 48 85 d2 74 2b 31 c9 eb 09 48 83 c1 01 48 39 ca 74 0e <0f> b6 04 0f 44 0f b6 04 0e 44 29 c0 74 e9 31 d2 31 c9 31 f6 31 ff
[ma apr 10 08:11:33 2023] RSP: 0018:ffffc06080f676c0 EFLAGS: 00010246
[ma apr 10 08:11:33 2023] RAX: 0000000000000001 RBX: ffff9e6a3accb600 RCX: 0000000000000000
[ma apr 10 08:11:33 2023] RDX: 0000000000000007 RSI: ffffffffc16c5e97 RDI: 0000000000000000
[ma apr 10 08:11:33 2023] RBP: ffffc06080f67a28 R08: 0000000000000000 R09: 0000000000000000
[ma apr 10 08:11:33 2023] R10: 0000000000000000 R11: 0000000000000000 R12: ffffc0608246f000
[ma apr 10 08:11:33 2023] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000005
[ma apr 10 08:11:33 2023] FS:  00007feb7b348740(0000) GS:ffff9e6c2e6c0000(0000) knlGS:0000000000000000
[ma apr 10 08:11:33 2023] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ma apr 10 08:11:33 2023] CR2: 0000000000000000 CR3: 000000011f30c001 CR4: 00000000001706e0
[ma apr 10 08:11:33 2023] Call Trace:
[ma apr 10 08:11:33 2023]  <TASK>
[ma apr 10 08:11:33 2023]  ? cfg80211_rtw_scan+0x311/0xc76 [8188eu]
[ma apr 10 08:11:33 2023]  rdev_scan+0x2d/0xe0 [cfg80211]
[ma apr 10 08:11:33 2023]  cfg80211_scan+0xf6/0x140 [cfg80211]
[ma apr 10 08:11:33 2023]  nl80211_trigger_scan+0x412/0x8d0 [cfg80211]
[ma apr 10 08:11:33 2023]  genl_family_rcv_msg_doit+0x108/0x180
[ma apr 10 08:11:33 2023]  genl_rcv_msg+0xf0/0x200
[ma apr 10 08:11:33 2023]  ? nl80211_send_scan_start+0xc0/0xc0 [cfg80211]
[ma apr 10 08:11:33 2023]  ? genl_get_cmd+0x120/0x120
[ma apr 10 08:11:33 2023]  netlink_rcv_skb+0x57/0x110
[ma apr 10 08:11:33 2023]  genl_rcv+0x28/0x50
[ma apr 10 08:11:33 2023]  netlink_unicast+0x247/0x390
[ma apr 10 08:11:33 2023]  netlink_sendmsg+0x25e/0x4e0
[ma apr 10 08:11:33 2023]  sock_sendmsg+0x6d/0x70
[ma apr 10 08:11:33 2023]  __sys_sendto+0x142/0x1a0
[ma apr 10 08:11:33 2023]  __x64_sys_sendto+0x24/0x40
[ma apr 10 08:11:33 2023]  do_syscall_64+0x5b/0x90
[ma apr 10 08:11:33 2023]  ? do_syscall_64+0x67/0x90
[ma apr 10 08:11:33 2023]  ? syscall_exit_to_user_mode+0x29/0x50
[ma apr 10 08:11:33 2023]  ? do_syscall_64+0x67/0x90
[ma apr 10 08:11:33 2023]  ? do_syscall_64+0x67/0x90
[ma apr 10 08:11:33 2023]  ? sysvec_apic_timer_interrupt+0x4b/0xd0
[ma apr 10 08:11:33 2023]  entry_SYSCALL_64_after_hwframe+0x63/0xcd
[ma apr 10 08:11:33 2023] RIP: 0033:0x7feb7b12081d
[ma apr 10 08:11:33 2023] Code: eb bd 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 80 3d cd dd 0d 00 00 41 89 ca 74 20 45 31 c9 45 31 c0 b8 2c 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 6b c3 66 2e 0f 1f 84 00 00 00 00 00 55 48 83
[ma apr 10 08:11:33 2023] RSP: 002b:00007ffd1726f958 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[ma apr 10 08:11:33 2023] RAX: ffffffffffffffda RBX: 000055e6e452aa20 RCX: 00007feb7b12081d
[ma apr 10 08:11:33 2023] RDX: 0000000000000038 RSI: 000055e6e4545d80 RDI: 0000000000000004
[ma apr 10 08:11:33 2023] RBP: 000055e6e4542f60 R08: 0000000000000000 R09: 0000000000000000
[ma apr 10 08:11:33 2023] R10: 0000000000000000 R11: 0000000000000246 R12: 00007ffd1726f9bc
[ma apr 10 08:11:33 2023] R13: 000055e6e4534600 R14: 000055e6e452a7a0 R15: 000055e6e2e6f3d6
[ma apr 10 08:11:33 2023]  </TASK>
[ma apr 10 08:11:33 2023] Modules linked in: tls 8188eu(OE) snd_seq_dummy snd_hrtimer xt_CHECKSUM rfcomm bnep bridge stp llc btusb btrtl btbcm btintel btmtk ccm algif_aead des_generic libdes algif_skcipher bluetooth cmac ecdh_generic ecc lz4 lz4_compress md4 zram algif_hash af_alg bbswitch(OE) ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 xt_LOG nf_log_syslog xt_multiport nft_limit xt_limit xt_addrtype xt_tcpudp nft_chain_nat xt_MASQUERADE nf_nat xt_comment xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sunrpc nft_compat nf_tables binfmt_misc nfnetlink intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp mei_pxp mei_hdcp iwlmvm kvm_intel mac80211 kvm libarc4 rapl iwlwifi snd_ctl_led intel_cstate snd_hda_codec_realtek snd_hda_codec_generic snd_ [... cut ...]
[ma apr 10 08:11:33 2023] RBP: ffffc06080f67a28 R08: 0000000000000000 R09: 0000000000000000
[ma apr 10 08:11:33 2023] R10: 0000000000000000 R11: 0000000000000000 R12: ffffc0608246f000
[ma apr 10 08:11:33 2023] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000005
[ma apr 10 08:11:33 2023] FS:  00007feb7b348740(0000) GS:ffff9e6c2e6c0000(0000) knlGS:0000000000000000
[ma apr 10 08:11:33 2023] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ma apr 10 08:11:33 2023] CR2: 0000000000000000 CR3: 000000011f30c001 CR4: 00000000001706e0

```

iafilius, 2023-04-10:
Small update with my latest findings:

Added a check if address where *ssids point is is nonzero.
And this prevents the hard crashing.
 
ioctl_cfg80211.c  around section starting at line 2289
```
			if(&*ssids != NULL) {
				if (!memcmp(ssids->ssid, "DIRECT-", 7) &&
...
```

when adding a printk for ssid like:
```
printk(KERN_INFO "ssids  %p\n",ssids );
```
it sprints
```
ssids  0000000000000000
```
in slow pace, but never something else.

lwfinger, 2023-04-10:
Yes, the code was reaching this point before it was fully up. I used a different test than yours, but it should do the same thing.

lwfinger, 2023-04-10:
No, a new issue is not needed. I think this is likely a case where the driver tried to get the station status report before the station was fully available. Frankly, I see no need for the warning, and I will remove it.

I will now close this issue.

closed 2023-04-10 by lwfinger