lwfinger/rtl8188eu #442: kernel 6+

val3ri, 2023-04-30:
Hello, first of all - thank you, your repo is great, you have done a great job!
I have a question - are you going to update this driver to the newer kernel 6+ version? I have updated my raspberry pi 2 model b and it uses now the newest 6.1.21 version, and the dongle tplink wn725 is not working, as you can imagine..

Thanks for advance for the time!

lwfinger, 2023-04-30:
Why do you say it is not working? I have successfully run that dongle on kernels through what is now becoming 6.4.

simurq, 2023-05-01:
I can confirm that the current version of the driver works with kernels 6.x. Here's the flow of commands I usually run from the terminal after each kernel update and cloning the repo:

`git checkout origin/v5.2.2.4 -b v5.2.2.4`
`make`
`sudo make install`
`reboot`

val3ri, 2023-05-01:
So, my mistake, it works! Sorry for bothering!

closed 2023-05-01 by val3ri

lwfinger, 2023-05-01:
@simurq - It should not be necessary to reclone the source every time. A simple 'git pull' should suffice unless you have deleted the source.