lwfinger/rtl8188eu #443: Can v5.2.2.4 branch be used to create an AP with Network Manager?

Mastiffen, 2023-05-03:
I have been using the driver on an RTL8188ETV for a year with an older version of Raspbian (Stretch), but now I have to upgrade because some of my software can't be upgraded to new functions on older versions. I tried using the v5.2.2.4 branch as I did before, but I can't get it to work. Network Manager (which is a lot easier to work with than hostad) says AP mode is not supported, and I did the `git checkout origin/v5.2.2.4 -b v5.2.2.4` and installed the driver that gave me. Is there something I can do, before I spend a few hours setting up hostapd, without even knowing if that will work on Bullseye?

lwfinger, 2023-05-03:
If CONFIG_AP is defined, and it is in the code in the repo, then NM should be able to create an AP with this driver, but you are telling me that this is not the case. Note that this driver uses cfg80211 and nl80211, but no mac80211. NM may require some functionality not coded here.

I do not have any experience with this chip as an AP.

Mastiffen, 2023-05-03:
Wow, you're impressively fast! But thanks, then I will make an attempt with hostapd, maybe I can get that to work.