lwfinger/rtl8188eu #444: v5.2.2.4_25483.20171222 driver not working with 5.10.145-rt74+g1594b25154a2, crash after 5mins

anishkhannayellapu124, 2023-06-30:
Hi,
     Thank you for the driver.
     I am testing the latest driver with 5.10.145 kernel. 
     Device: TL-WN725N
     I am able enable access point using hostapd. I am getting below crash for every 5mins. 
     It is working back only after removing the module and inserting back.
#############################################################################################
    root@s32g399ardb3:/usr/bin/wlan# [  476.472462] RTW: set group key camid:1, addr:00:00:00:00:00:00, kid:1, type:AES
[  476.472650] RTW: send eapol packet
[  477.286352] RTW: send eapol packet
[  817.111239] ------------[ cut here ]------------
[  817.111258] WARNING: CPU: 5 PID: 772 at createbss_hdl+0x120/0x148 [8188eu]
[  817.111502] Modules linked in: 8188eu(O) cfg80211 lib80211 [last unloaded: 8188eu]
[  817.111520] CPU: 5 PID: 772 Comm: RTW_CMD_THREAD Tainted: G        WC O      5.10.145-rt74+g1594b25154a2 #1
[  817.111527] Hardware name: NXP S32G399A-RDB3 (DT)
[  817.111531] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[  817.111537] pc : createbss_hdl+0x120/0x148 [8188eu]
[  817.111672] lr : rtw_cmd_thread+0x454/0x548 [8188eu]
[  817.111806] sp : ffffffc013db3da0
[  817.111808] x29: ffffffc013db3da0 x28: ffffffc0089c9168
[  817.111815] x27: ffffffc008909f74 x26: 0000000000001400
[  817.111821] x25: ffffffc0089c90f8 x24: ffffffc011010320
[  817.111827] x23: ffffffc011010400 x22: 000000010001f8f3
[  817.111833] x21: ffffffc011010350 x20: ffffffc01100f000
[  817.111839] x19: ffffffc01100f000 x18: 0000000000000000
[  817.111845] x17: 0000000000000001 x16: 0000000000000000
[  817.111851] x15: 001fffffffffffff x14: 00000000001d7000
[  817.111857] x13: 0000000000000370 x12: 0000000000000000
[  817.111863] x11: 0000000000000370 x10: 00000000000008e0
[  817.111868] x9 : ffffffc013db3960 x8 : ffffff880230eb00
[  817.111874] x7 : 0000000000000001 x6 : ffffff8804ea9206
[  817.111880] x5 : 0000800020000000 x4 : 0000000000000000
[  817.111886] x3 : 000000000000ffff x2 : 0000000000000006
[  817.111893] x1 : ffffff8804ea9200 x0 : 0000000000000000
[  817.111900] Call trace:
[  817.111903]  createbss_hdl+0x120/0x148 [8188eu]
[  817.112037]  rtw_cmd_thread+0x454/0x548 [8188eu]
[  817.112170]  kthread+0x178/0x180
[  817.112184]  ret_from_fork+0x10/0x30
[  817.112193] ---[ end trace 00f853683c01e7a3 ]---
[  818.113051] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

lwfinger, 2023-06-30:
I just pushed a fix that should take care of the warning. Without testing, I cannot be certain that the AP will stay up when the error condition happens.

furioness, 2023-07-01:
Hello @lwfinger, thank you for your work.

Yesterday I got the update, today booted up my PC and noticed that I had no internet - right after booting, reinserting the stick didn't help.

I downgraded to the previous commit ([f435a106251969a](https://github.com/lwfinger/rtl8188eu/commit/f435a106251969acc588e33143bd774c3a06bf7a)) and now it works again.

The device is TL-WN722N (HVIN TL-WN722 V2)
Kernel: 6.3.9-zen1-1-zen (Arch Linux)

Here are some logs, please ask if you need more (but tell me where to look, please). I can install the latest commit again to help with testing.

```
Jul 01 12:30:37 fs-hpc kernel: WARNING: CPU: 2 PID: 0 at /var/lib/dkms/8188eu/5.2.2.4.r827.573b2f4/build/rtw_cmd.c:415 rtw_enqueue_cmd+0x46/0xd0 [8188eu]
Jul 01 12:30:37 fs-hpc kernel: Modules linked in: wireguard curve25519_x86_64 libchacha20poly1305 chacha_x86_64 poly1305_x86_64 libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel rndis_host cdc_ether usbnet mii xt_nat xt_tcpudp veth xt_conntrack xt_MASQUERADE nf_conntrack_netlink iptable_nat xt_addrtype iptable_filter br_netfilter bridge stp llc snd_seq_dummy snd_hrtimer snd_seq snd_seq_device overlay r8169 realtek mdio_devres libphy joydev mousedev 8188eu(OE) cfg80211 rfkill intel_rapl_msr intel_rapl_common edac_mce_amd snd_hda_codec_realtek kvm_amd snd_hda_codec_generic ledtrig_audio kvm snd_hda_codec_hdmi irqbypass snd_hda_intel snd_intel_dspcfg crct10dif_pclmul usbhid crc32_pclmul snd_intel_sdw_acpi polyval_clmulni snd_hda_codec polyval_generic gf128mul ghash_clmulni_intel snd_hda_core sha512_ssse3 snd_hwdep aesni_intel snd_pcm crypto_simd snd_timer sp5100_tco cryptd snd rapl wmi_bmof gigabyte_wmi pcspkr k10temp soundcore ccp i2c_piix4 gpio_amdpt gpio_generic acpi_cpufreq mac_hid nft_reject_inet nf_reject_ipv4 nf_reject_ipv6
Jul 01 12:30:37 fs-hpc kernel:  nft_reject nft_masq nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 vboxnetflt(OE) vboxnetadp(OE) nf_tables libcrc32c nfnetlink vboxdrv(OE) dm_multipath crypto_user fuse loop bpf_preload ip_tables x_tables dm_mod nvme nvme_core xhci_pci nvme_common xhci_pci_renesas amdgpu i2c_algo_bit drm_ttm_helper ttm video wmi drm_buddy gpu_sched drm_display_helper cec ext4 crc32c_generic crc32c_intel crc16 mbcache jbd2
Jul 01 12:30:37 fs-hpc kernel: CPU: 2 PID: 0 Comm: swapper/2 Tainted: G        W  OE      6.3.9-zen1-1-zen #1 efe7a809d8d6cfab891c1fc96b14365ac47f5aa3
Jul 01 12:30:37 fs-hpc kernel: Hardware name: Gigabyte Technology Co., Ltd. B550M DS3H/B550M DS3H, BIOS F16a 11/03/2022
Jul 01 12:30:37 fs-hpc kernel: RIP: 0010:rtw_enqueue_cmd+0x46/0xd0 [8188eu]
Jul 01 12:30:37 fs-hpc kernel: Code: f3 48 8b 87 88 00 00 00 48 89 06 e8 84 fe ff ff 89 c5 85 c0 74 18 8b 53 18 81 fa 00 04 00 00 76 67 83 3d f0 28 1f 00 03 77 29 <0f> 0b eb 0b 8b 53 18 81 fa 00 04 00 00 77 e8 66 83 7b 08 39 74 2d
Jul 01 12:30:37 fs-hpc kernel: RSP: 0018:ffffb8f8c02fca18 EFLAGS: 00010297
Jul 01 12:30:37 fs-hpc kernel: RAX: 0000000000000001 RBX: ffff97b546573ba0 RCX: 0000000000000008
Jul 01 12:30:37 fs-hpc kernel: RDX: 0000000000000484 RSI: ffff97b546573ba0 RDI: ffffb8f8c2dd659c
Jul 01 12:30:37 fs-hpc kernel: RBP: 0000000000000001 R08: 000000000000000c R09: 000000000000000a
Jul 01 12:30:37 fs-hpc kernel: R10: 0000000000000000 R11: 0000000000000000 R12: ffffb8f8c2dd6520
Jul 01 12:30:37 fs-hpc kernel: R13: ffffb8f8c4027ae8 R14: ffff97b629498808 R15: ffff97b61e207784
Jul 01 12:30:37 fs-hpc kernel: FS:  0000000000000000(0000) GS:ffff97bc5da80000(0000) knlGS:0000000000000000
Jul 01 12:30:37 fs-hpc kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jul 01 12:30:37 fs-hpc kernel: CR2: 00007ff8a521d998 CR3: 0000000592420000 CR4: 0000000000750ee0
Jul 01 12:30:37 fs-hpc kernel: PKRU: 55555554
Jul 01 12:30:37 fs-hpc kernel: Call Trace:
Jul 01 12:30:37 fs-hpc kernel:  <IRQ>
Jul 01 12:30:37 fs-hpc kernel:  ? __warn+0x81/0x1b0
Jul 01 12:30:37 fs-hpc kernel:  ? rtw_enqueue_cmd+0x46/0xd0 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  ? report_bug+0x202/0x270
Jul 01 12:30:37 fs-hpc kernel:  ? handle_bug+0x3c/0x80
Jul 01 12:30:37 fs-hpc kernel:  ? exc_invalid_op+0x19/0xc0
Jul 01 12:30:37 fs-hpc kernel:  ? asm_exc_invalid_op+0x1a/0x20
Jul 01 12:30:37 fs-hpc kernel:  ? rtw_enqueue_cmd+0x46/0xd0 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  ? rtw_enqueue_cmd+0x2c/0xd0 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  report_survey_event+0xc3/0x110 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  ? __pfx__rtw_mi_report_survey_event+0x10/0x10 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  _rtw_mi_report_survey_event+0xe/0x20 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  _rtw_mi_process+0x60/0x80 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  rtw_mi_report_survey_event+0x1d/0x30 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  OnBeacon+0x231/0x7e0 [8188eu 3a447f699d17df5b00708c39fdf682e42e6e6ea2]
Jul 01 12:30:37 fs-hpc kernel:  ? __warn+0xc7/0x1b0
Jul 01 12:30:37 fs-hpc kernel:  ? put_pages_list+0xc8/0xf0
Jul 01 12:30:37 fs-hpc kernel:  ? iommu_v1_map_pages+0x1dc/0x7b0
Jul 01 12:30:37 fs-hpc kernel:  ? select_idle_cpu+0x257/0x550
Jul 01 12:30:37 fs-hpc kernel:  ? select_task_rq_fair+0x163/0x24b0
Jul 01 12:30:37 fs-hpc kernel:  ? update_group_capacity+0x29/0x230
[... cut ...]
Jul 01 12:30:37 fs-hpc kernel: R13: 0000000000000003 R14: 000000917f87f007 R15: 0000000000000000
Jul 01 12:30:37 fs-hpc kernel:  cpuidle_enter+0x2d/0x40
Jul 01 12:30:37 fs-hpc kernel:  do_idle+0x1bf/0x220
Jul 01 12:30:37 fs-hpc kernel:  cpu_startup_entry+0x1d/0x20
Jul 01 12:30:37 fs-hpc kernel:  start_secondary+0x115/0x140
Jul 01 12:30:37 fs-hpc kernel:  secondary_startup_64_no_verify+0xe5/0xeb
Jul 01 12:30:37 fs-hpc kernel:  </TASK>
Jul 01 12:30:37 fs-hpc kernel: ---[ end trace 0000000000000000 ]---
```

lwfinger, 2023-07-01:
I reverted the earlier patch and pushed a new one that works as an STA. It has not been tested in AP mode. That rest is left to you.

anishkhannayellapu124, 2023-07-03:
Hi,

Thank you very much for the quick response.
I took the latest commit, still issue remains same.
root@s32g399ardb3:/usr/bin/wlan# dhclwlan0: AP-STA-DISCONNECTED 52:70:a9:e6:f5:63
wlan0: INTERFACE-DISABLED
wlan0: INTERFACE-ENABLED
[  398.112216] Inside createbss_hdl ...
[  398.112268] ------------[ cut here ]------------
[  398.112271] WARNING: CPU: 4 PID: 338 at createbss_hdl+0x140/0x14c [8188eu]
[  398.112497] Modules linked in: 8188eu(O) cfg80211 lib80211 [last unloaded: r8188eu]
[  398.112518] CPU: 4 PID: 338 Comm: RTW_CMD_THREAD Tainted: G         C O      5.10.145-rt74+g1594b25154a2 #1
[  398.112525] Hardware name: NXP S32G399A-RDB3 (DT)
[  398.112529] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[  398.112535] pc : createbss_hdl+0x140/0x14c [8188eu]
[  398.112668] lr : createbss_hdl+0x34/0x14c [8188eu]
[  398.112803] sp : ffffffc013dfbda0
[  398.112805] x29: ffffffc013dfbda0 x28: ffffffc0089c9168
[  398.112811] x27: ffffffc008909f74 x26: 0000000000001400
[  398.112817] x25: ffffffc0089c90f8 x24: ffffffc011fa6320
[  398.112823] x23: ffffffc011fa6400 x22: ffffffc011fa80b7
[  398.112829] x21: ffffffc011fa5eda x20: ffffff8804f3ba00
[  398.112836] x19: ffffffc011fa5000 x18: ffffffc010cb85f8
[  398.112841] x17: 0000000000000001 x16: 0000000000000000
[  398.112847] x15: 0000000000000028 x14: 0000000000000331
[  398.112853] x13: 0000000000000001 x12: fffffffffffe7517
[  398.112859] x11: ffffffc010cb8610 x10: ffffffc010cc1f70
[  398.112864] x9 : ffffffc013dfbda0 x8 : 2e2e2e206c64685f
[  398.112870] x7 : 7373626574616572 x6 : 00000000fffff9be
[  398.112877] x5 : 0000000000000000 x4 : ffffff885f882888
[  398.112883] x3 : ffffff885f8857c0 x2 : 2ccd2720537f8600
[  398.112889] x1 : 0000000000000000 x0 : 0000000000000000
[  398.112895] Call trace:
[  398.112898]  createbss_hdl+0x140/0x14c [8188eu]
[  398.113031]  rtw_cmd_thread+0x454/0x548 [8188eu]
[  398.113164]  kthread+0x178/0x180
[  398.113176]  ret_from_fork+0x10/0x30
[  398.113186] ---[ end trace 60613964b0be4a2b ]---
[  399.113775] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
wlan0: STA 52:70:a9:e6:f5:63 IEEE 802.11: disassociated

Linux: Linux s32g399ardb3 5.10.145-rt74+g1594b25154a2 #1 SMP PREEMPT Tue Nov 29 10:33:39 UTC 2022 aarch64 GNU/Linux

Mode: AP

Foe every 5mins, i am getting this error. if remove driver module, kill hostapd. insert driver module and run hostapd. Device is working again.

lwfinger, 2023-07-03:
This bug does not make much sense at the moment. I have pushed a change for you to try.

anishkhannayellapu124, 2023-07-05:
Hi,

Thank you for the change.
Post dump stack access point is not working. I took updated driver, still issue is there.
[  388.113153] ------------[ cut here ]------------
[  388.113168] WARNING: CPU: 2 PID: 328 at createbss_hdl+0x28/0xfc [8188eu]
[  388.113393] Modules linked in: 8188eu(O) cfg80211 lib80211 [last unloaded: r8188eu]
[  388.113411] CPU: 2 PID: 328 Comm: RTW_CMD_THREAD Tainted: G         C O      5.10.145-rt74+g1594b25154a2 #1
[  388.113417] Hardware name: NXP S32G399A-RDB3 (DT)
[  388.113421] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[  388.113427] pc : createbss_hdl+0x28/0xfc [8188eu]
[  388.113562] lr : rtw_cmd_thread+0x454/0x548 [8188eu]
[  388.113695] sp : ffffffc013e4bdb0
[  388.113698] x29: ffffffc013e4bdb0 x28: ffffffc0089c9168
[  388.113704] x27: ffffffc008909f74 x26: 0000000000001400
[  388.113712] x25: ffffffc0089c90f8 x24: ffffffc011fa6320
[  388.113718] x23: ffffffc011fa6400 x22: 0000000100005602
[  388.113724] x21: ffffffc011fa6350 x20: ffffffc011fa5000
[  388.113730] x19: ffffffc011fa5000 x18: 0000000000000000
[  388.113736] x17: 0000000000000001 x16: 0000000000000000
[  388.113742] x15: 001fffffffffffff x14: 00000000000000ea
[  388.113747] x13: 0000000000000001 x12: ffffff885f898740
[  388.113753] x11: 0000000000000002 x10: 00000000000008e0
[  388.113759] x9 : ffffffc013e4b960 x8 : ffffff8803833700
[  388.113765] x7 : 0000000000000001 x6 : ffffff880446fa06
[  388.113770] x5 : 8000000000010000 x4 : 0000000000000000
[  388.113776] x3 : 000000000000ffff x2 : 0000000000000006
[  388.113783] x1 : ffffff880446fa00 x0 : 0000000000000000
[  388.113790] Call trace:
[  388.113793]  createbss_hdl+0x28/0xfc [8188eu]
[  388.113927]  rtw_cmd_thread+0x454/0x548 [8188eu]
[  388.114060]  kthread+0x178/0x180
[  388.114072]  ret_from_fork+0x10/0x30
[  388.114081] ---[ end trace 88de2916fdf11a98 ]---
[  389.114731] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

lwfinger, 2023-07-05:
I just pushed another try to see where and why there is a warning. Please try again.

anishkhannayellapu124, 2023-07-06:
Hi,

Thank you for the quick change. With the change, dump_stack is not printed, but access point disconnects after 5mins.
[   33.291328] usbcore: deregistering interface driver rtl8188eu
[   33.396940] RTW: rtw_cmd_thread: DriverStopped(True) SurpriseRemoved(False) break at line 517
[   33.536800] usb 1-1: reset high-speed USB device number 2 using ci_hdrc
[   42.035476] RTW: rtl8188eu v5.2.2.4_25483.20171222
[   42.073509] RTW: hal_com_config_channel_plan chplan:0x22
[   42.073994] RTW: rtw_regsty_chk_target_tx_power_valid return false for band:0, path:0, rs:0, t:-1
[   42.074559] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074565] RTW: unsupported channel: 114 at 5G
[   42.074568] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074571] RTW: unsupported channel: 114 at 5G
[   42.074574] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074577] RTW: unsupported channel: 114 at 5G
[   42.074625] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074628] RTW: unsupported channel: 114 at 5G
[   42.074631] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074634] RTW: unsupported channel: 114 at 5G
[   42.074637] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074640] RTW: unsupported channel: 114 at 5G
[   42.074688] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074692] RTW: unsupported channel: 114 at 5G
[   42.074694] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074699] RTW: unsupported channel: 114 at 5G
[   42.074701] RTW: Invalid Channel 114 of Band 1 in phy_GetChannelIndexOfTxPowerLimit
[   42.074705] RTW: unsupported channel: 114 at 5G
[   42.090804] usbcore: registered new interface driver rtl8188eu
[   42.539151] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.539389] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.539640] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.539889] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.540141] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.540390] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.540640] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.540891] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.541140] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.541391] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.541640] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.541891] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.542143] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.542390] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.542640] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.542891] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.543140] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.543391] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.543640] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   42.543891] RTW: Invalid Channel 0 of Band 0 in phy_GetChannelIndexOfTxPowerLimit
[   53.785114] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   53.785497] RTW: set group key camid:1, addr:00:00:00:00:00:00, kid:1, type:AES
[   55.315034] RTW: set group key camid:1, addr:00:00:00:00:00:00, kid:1, type:AES
[   55.315194] RTW: send eapol packet
[   55.324205] RTW: send eapol packet
[   55.329493] RTW: set pairwise key camid:4, addr:52:70:a9:e6:f5:63, kid:0, type:AES
[  166.296308] RTW: send eapol packet
[  166.315537] RTW: send eapol packet
[  166.330026] RTW: set pairwise key camid:5, addr:ca:8a:f1:1c:2a:ac, kid:0, type:AES
root@s32g399ardb3:/usr/bin/wlan# wlan0: AP-STA-DISCONNECTED ca:8a:f1:1c:2a:ac
wlan0: AP-STA-D[  387.111179] NOT an ad-hoc master
I[  387.111588] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
SCONNECTED 52:70:a9:e6:f5:63
wlan0: INTERFACE-DISABLED
wlan0: INTERFACE-ENABLED
wlan0: STA ca:8a:f1:1c:2a:ac IEEE 802.11: disassociated
wlan0: STA 52:70:a9:e6:f5:63 IEEE 802.11: disassociated

lwfinger, 2023-07-06:
We are getting somewhere now. Unfortunately, using gdb on the traceback was pointing to a different place on my computer. When I actually located the warning, I had it print just a message. It shows up in your log as  "[ 387.111179] NOT an ad-hoc master". That time corresponds to the time when your AP goes down.

Obviously, the code is getting confused between ad-hoc and AP mode. I found only one place where ad-hoc mode was set. It has been disabled. Pull and try again. Let us see what else blows up. :)

anishkhannayellapu124, 2023-07-10:
Hi,

Issue is still coming from same place.

root@s32g399ardb3:/usr/bin/wlan#
root@s32g399ardb3:/usr/bin/wlan# wlan0: AP-STA-DISCONNECTED ca:8a:f1:1c:2a:ac
wlan0: AP-STA-[  390.112218] NOT an ad-hoc master
D[  390.112569] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
ISCONNECTED 52:70:a9:e6:f5:63
wlan0: INTERFACE-DISABLED
wlan0: INTERFACE-ENABLED
wlan0: STA 52:70:a9:e6:f5:63 IEEE 802.11: disassociated
wlan0: STA ca:8a:f1:1c:2a:ac IEEE 802.11: disassociated

root@s32g399ardb3:/usr/bin/wlan#

lwfinger, 2023-07-10:
I just went through the code and removed every place the it could set AD-HOC or AD-HOC-MASTER mode. Let's see where it actually breaks. :)

anishkhannayellapu124, 2023-07-11:
Hi ,

Really thank you for the support. Disabling random MAC addresses resolved my issue

lwfinger, 2023-07-11:
Good catch, but why were you enabling that "feature", or better known as a "bug". It is quite difficult for a driver to get that right.

closed 2023-07-11 by lwfinger