lwfinger/rtl8188eu #446: Version 6.1.38-V7+ supported?

stevewrightnz, 2023-07-17:
Hi,
Getting the following error with bullseye:

`make ARCH=arm CROSS_COMPILE= -C /lib/modules/6.1.36-v7+/build M=/home/pi/rtl8188eu  modules
make[1]: *** /lib/modules/6.1.36-v7+/build: No such file or directory.  Stop.
make: *** [Makefile:155: modules] Error 2
`
Is a fix likely?

lwfinger, 2023-07-17:
Yes, you have not installed the kernel headers.

kdm6389, 2023-07-29:
you need to make change  in 
/os_dep/osdep_service.c

lin no 385

```
#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 17, 0)
	memcpy(pnetdev->dev_addr, padapter->eeprompriv.mac_addr, ETH_ALEN);
#else
	dev_addr_set(pnetdev, padapter->eeprompriv.mac_addr);
#endif
	if (!rtnl_is_locked())
		ret = register_netdev(pnetdev);
	else
		ret = register_netdevice(pnetdev);
	if (ret != 0) {
		RT_TRACE(_module_hci_intfs_c_, _drv_err_,
			 ("register_netdev() failed\n"));
		goto error;
	}
	rtw_proc_init_one(pnetdev);
	return 0;
```