lwfinger/rtl8188eu #447: Concurrent mode not being really concurrent

shinhoshi57, 2023-08-17:
Hi,
Our base hardware is a beaglebone running 4.19.94-ti-r42 kernel, and our 8188eu device is as follows:

root@beaglebone:/opt/rtl8188eu# lsusb
Bus 001 Device 003: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

We've compiled the driver with the concurrent mode option on, which allows us to effectively have 2 wlan interfaces:

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.50  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::36c9:f0ff:fe8c:ec7c  prefixlen 64  scopeid 0x20<link>
        ether 34:c9:f0:8c:ec:7c  txqueuelen 1000  (Ethernet)
        RX packets 259797  bytes 774644 (756.4 KiB)
        RX errors 0  dropped 742  overruns 0  frame 0
        TX packets 2047  bytes 454838 (444.1 KiB)
        TX errors 0  dropped 12 overruns 0  carrier 0  collisions 0

wlan1: flags=-28605<UP,BROADCAST,RUNNING,MULTICAST,DYNAMIC>  mtu 1500
        inet 172.26.10.143  netmask 255.255.0.0  broadcast 172.26.255.255
        inet6 fe80::34c9:f0ff:fe8c:ec7c  prefixlen 64  scopeid 0x20<link>
        ether 36:c9:f0:8c:ec:7c  txqueuelen 1000  (Ethernet)
        RX packets 255407  bytes 32287932 (30.7 MiB)
        RX errors 0  dropped 35737  overruns 0  frame 0
        TX packets 5622  bytes 4332110 (4.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0 is an access point interface for users to connect.

wlan1 is the interface used for the adapter to connect to other SSIDs.

Things seem to work fine, except for when we combine traffic over the 2 interfaces.

For clarity, I am adding a node.js test:

const puppeteer = require('puppeteer');

(async () => {
  const browser = await puppeteer.launch({  headless: "new", ignoreHTTPSErrors: true});
  const page1 = await browser.newPage();
  const page2 = await browser.newPage();

  var promises=[];
  
  promises.push(page1.goto('https://192.168.1.50/panel/login', { waitUntil: 'networkidle2' }));
  promises.push(page2.goto('https://172.26.10.143/panel/login', { waitUntil: 'networkidle2' }));

  await Promise.all(promises);

  await browser.close();
})();

Some times this works, others it produces the adapter to completely crash.

[  257.818325] RTW: wlan1 bBusyTraffic == true
[  265.451662] RTW: rtw_update_ramask => mac_id:0, networkType:0x0b, mask:0x00000000000f0000
               	 ==> rssi_level:5, rate_bitmap:0x0000000000000000, shortGIrate=1
               	 ==> bw:0, ignore_bw:0x1
[  267.996509] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.100578] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.204557] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.308589] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.412623] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.516678] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.620620] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.724641] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.828681] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  268.932686] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.036753] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.140733] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.244752] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.348764] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.452793] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.556870] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.660831] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.764846] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait
[  269.868869] RTW: rtw_sctx_wait timeout: dump_mgntframe_and_wait

...

[  283.532059] RTW: issue_beacon, fail!
[  283.705718] RTW: issue_beacon, fail!
[  283.893890] RTW: issue_beacon, fail!
[  284.076365] RTW: issue_beacon, fail!
[  284.257778] RTW: issue_beacon, fail!
[  284.433639] RTW: issue_beacon, fail!
[  284.606069] RTW: issue_beacon, fail!
[  287.922490] RTW: issue_beacon, fail!
[  300.052775] RTW: issue_beacon, fail!
[  300.243144] RTW: issue_beacon, fail!

Eventually petitions end timeouting. And the client device connecting to the AP ends losing the connection.

Any ideas?

shinhoshi57, 2023-08-17:
Another thing we've observed, stressing the single interface:

[  313.931576] RTW: send_beacon fail! 184 ms
[  313.931591] RTW: issue_beacon, fail!
[  313.931745] RTW: rtl8188e_sreset_xmit_status_check tx hang 
[  313.931753] RTW: sreset_reset
[  313.931764] RTW: sreset_stop_adapter(wlan0)
[  313.931781] RTW: sreset_stop_adapter(wlan1)
[  313.931794] RTW: ==>ips_enter cnts:1
[  313.931799] RTW: nolinked power save enter
[  313.931805] RTW: ===> rtw_ips_pwr_down...................
[  313.931810] RTW: ====> rtw_ips_dev_unload...
[  313.959683] RTW: usb_read_port_cancel
[  313.959731] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
[  313.959756] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
[  313.959774] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
[  313.959797] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
[  313.959820] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
[  313.959839] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
[  313.959859] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
[  313.959878] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
[  313.959890] RTW: usb_write_port_cancel
[  313.959907] RTW: usb_write_port_complete(): TX Warning! bDriverStopped(False) OR bSurpriseRemoved(False) pxmitbuf->buf_tag(0)
[  313.959942] RTW: xmit_tasklet => bDriverStopped or bSurpriseRemoved or bWritePortCancel
[  314.960751] ------------[ cut here ]------------
[  314.960806] WARNING: CPU: 0 PID: 1444 at drivers/usb/musb/musb_host.c:113 musb_h_tx_flush_fifo+0x148/0x14c
[  314.960815] musb-hdrc musb-hdrc.1: Could not flush host TX2 fifo: csr: 0003
[  314.960821] Modules linked in: usb_f_acm u_serial usb_f_ncm usb_f_rndis u_ether libcomposite pru_rproc irq_pruss_intc pruss pm33xx wkup_m3_ipc wkup_m3_rproc remoteproc virtio virtio_ring 8188eu(O) pruss_soc_bus cfg80211 uio_pdrv_genirq uio
[  314.960884] CPU: 0 PID: 1444 Comm: RTW_CMD_THREAD Tainted: G           O      4.19.94-ti-r42 #3
[  314.960889] Hardware name: Generic AM33XX (Flattened Device Tree)
[  314.960930] [<c0114194>] (unwind_backtrace) from [<c010e590>] (show_stack+0x20/0x24)
[  314.960953] [<c010e590>] (show_stack) from [<c0d1fbd4>] (dump_stack+0x8c/0xa0)
[  314.960970] [<c0d1fbd4>] (dump_stack) from [<c013cd3c>] (__warn.part.0+0xcc/0xe8)
[  314.960983] [<c013cd3c>] (__warn.part.0) from [<c013cdd4>] (warn_slowpath_fmt+0x7c/0xa4)
[  314.960999] [<c013cdd4>] (warn_slowpath_fmt) from [<c0a25e10>] (musb_h_tx_flush_fifo+0x148/0x14c)
[  314.961015] [<c0a25e10>] (musb_h_tx_flush_fifo) from [<c0a26dc4>] (musb_cleanup_urb+0x88/0xf0)
[  314.961026] [<c0a26dc4>] (musb_cleanup_urb) from [<c0a26fe0>] (musb_urb_dequeue+0x1b4/0x1f4)
[  314.961046] [<c0a26fe0>] (musb_urb_dequeue) from [<c09c3b58>] (unlink1+0x3c/0x11c)
[  314.961060] [<c09c3b58>] (unlink1) from [<c09c5fa0>] (usb_hcd_unlink_urb+0x80/0xd8)
[  314.961075] [<c09c5fa0>] (usb_hcd_unlink_urb) from [<c09c756c>] (usb_kill_urb.part.0+0x54/0xec)
[  314.961088] [<c09c756c>] (usb_kill_urb.part.0) from [<c09c763c>] (usb_kill_urb+0x38/0x3c)
[  314.962285] [<c09c763c>] (usb_kill_urb) from [<bf178d04>] (usb_write_port_cancel+0x6c/0xd0 [8188eu])
[  314.963775] [<bf178d04>] (usb_write_port_cancel [8188eu]) from [<bf127d44>] (_rtw_write_port_cancel+0x54/0x58 [8188eu])
[  314.964924] [<bf127d44>] (_rtw_write_port_cancel [8188eu]) from [<bf1772b0>] (usb_intf_stop+0x3c/0x4c [8188eu])
[  314.966007] [<bf1772b0>] (usb_intf_stop [8188eu]) from [<bf174fe0>] (rtw_intf_stop+0x28/0x2c [8188eu])
[  314.967053] [<bf174fe0>] (rtw_intf_stop [8188eu]) from [<bf175ea4>] (rtw_ips_dev_unload+0x44/0x7c [8188eu])
[  314.968078] [<bf175ea4>] (rtw_ips_dev_unload [8188eu]) from [<bf175f28>] (rtw_ips_pwr_down+0x4c/0x8c [8188eu])
[  314.969097] [<bf175f28>] (rtw_ips_pwr_down [8188eu]) from [<bf150b74>] (_ips_enter+0xa8/0xc4 [8188eu])
[  314.970107] [<bf150b74>] (_ips_enter [8188eu]) from [<bf16e348>] (sreset_reset+0xa8/0x128 [8188eu])
[  314.971130] [<bf16e348>] (sreset_reset [8188eu]) from [<bf1a10bc>] (rtw_hal_sreset_reset+0x28/0x2c [8188eu])
[  314.972153] [<bf1a10bc>] (rtw_hal_sreset_reset [8188eu]) from [<bf1c427c>] (rtl8188e_sreset_xmit_status_check+0x134/0x16c [8188eu])
[  314.973176] [<bf1c427c>] (rtl8188e_sreset_xmit_status_check [8188eu]) from [<bf1a1104>] (rtw_hal_sreset_xmit_status_check+0x20/0x24 [8188eu])
[  314.974167] [<bf1a1104>] (rtw_hal_sreset_xmit_status_check [8188eu]) from [<bf118268>] (rtw_drvextra_cmd_hdl+0x100/0x6fc [8188eu])
[  314.975141] [<bf118268>] (rtw_drvextra_cmd_hdl [8188eu]) from [<bf1157e8>] (rtw_cmd_thread+0x4b0/0x5a0 [8188eu])
[  314.975644] [<bf1157e8>] (rtw_cmd_thread [8188eu]) from [<c0160f3c>] (kthread+0x140/0x170)
[  314.975665] [<c0160f3c>] (kthread) from [<c01010e8>] (ret_from_fork+0x14/0x2c)
[  314.975673] Exception stack(0xd2e8bfb0 to 0xd2e8bff8)
[  314.975684] bfa0:                                     00000000 00000000 00000000 00000000
[  314.975696] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  314.975706] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  314.975716] ---[ end trace 6dfa521fc6ab6f60 ]---
[  314.975747] RTW: usb_write_port_complete(): TX Warning! bDriverStopped(False) OR bSurpriseRemoved(False) pxmitbuf->buf_tag(0)
[  315.971885] RTW: xmit_tasklet => bDriverStopped or bSurpriseRemoved or bWritePortCancel
[  315.984762] systemd-journald[786]: /dev/kmsg buffer overrun, some messages lost.
[  316.711601] RTW: usb_write_port_complete(): TX Warning! bDriverStopped(False) OR bSurprise

dubhater, 2023-12-03:
> wlan0 is an access point interface for users to connect.
>
> wlan1 is the interface used for the adapter to connect to other SSIDs.

Realtek recommends doing it the other way around:
```
Q4: Which wireless interface can run in station mode? Which wireless interface can
run in AP mode?
A: We recommend that run station mode in wlan0, run AP or P2P mode in wlan1.
```
(from Realtek_WiFi_concurrent_mode_Introduction.pdf)