lwfinger/rtl8188eu #451: Wi-Fi network goes asleep after some period of inactivity

simurq, 2023-11-19:
I [know](https://forums.raspberrypi.com/viewtopic.php?p=2130871#p2130871) that r8188eu was replaced in kernel 6.3 by rtl8xxxu. And since then my wi-fi network sort of "goes asleep" after some period of inactivity, e.g. when I'm not browsing, making me to toggle the wi-fi switch every time I want to go back online to re-connect it.

Since the driver branch 5.2.2.4 I'm using was last updated back in July, can you please advise what I can do to prevent the wi-fi network from falling into a state of drowsiness? Thank you!

Wi-Fi dongle: TP-Link TL-WN725N
Linux Kernel: 6.5.6
OS: Pop!_OS 22.04

dubhater, 2023-12-03:
Next time the wifi falls asleep, please run ``sudo journalctl -b`` and attach the full output here.

simurq, 2023-12-04:
> Next time the wifi falls asleep, please run `sudo journalctl -b` and attach the full output here.

Thank you for your interest! Here is the output of the command:

[Before restart (asleep)](https://gist.github.com/simurq/27e3f3a0687882e3a13b791b1fdd50fb)
[After restart](https://gist.github.com/simurq/c15ceea939e03df9bd9061840557c287)

Hope it helps!

dubhater, 2023-12-04:
Unfortunately I don't see anything strange in the logs.

Does the problem still happen if you move the computer closer to the access point?

Does the driver in this repository work better?

It's probably best to have only one driver at a time loaded, 8188eu or rtl8xxxu.

simurq, 2023-12-05:
> Does the problem still happen if you move the computer closer to the access point?

It's the same, unfortunately. I'm writing this message using USB tethering option on my phone. And this is strange, since both devices use the same network (I turned off mobile data on the phone). But I've never had a similar problem when connecting to the Internet from phone.

> Does the driver in this repository work better?

With every kernel update and up until this problem, I'd usually re-download, build and install the rtl8188eu driver from branch 5.2.2.4.

> It's probably best to have only one driver at a time loaded, 8188eu or rtl8xxxu.

This is the output of **sudo lshw -C netwrok | grep "driver"**. Does it mean I use two separate drivers after the kernel update (for reasons I'm not aware of)?

`configuration: autonegotiation=on broadcast=yes driver=r8169 driverversion=6.5.6-76060506-generic firmware=rtl8168h-2_0.0.2 02/26/15 latency=0 link=no multicast=yes port=twisted pair`
`configuration: broadcast=yes driver=rtl8xxxu driverversion=6.5.6-76060506-generic firmware=N/A ip=192.168.1.9 link=yes multicast=yes wireless=IEEE 802.11`
`configuration: autonegotiation=off broadcast=yes driver=rndis_host driverversion=6.5.6-76060506-generic duplex=half firmware=RNDIS device ip=192.168.11.196 link=yes multicast=yes port=twisted pair`

lwfinger, 2023-12-05:
You are only using r8xxxu. In the configuration, the entries with "twisted pair" are the wiired interfaces.

simurq, 2023-12-12:
> The rtl8188eu driver is an option, but you will need to blacklist rtl8xxxu.

Thank you! I blacklisted it as you advised (added the relevant line to `/etc/modprobe.d/blacklist.conf`), restarted the system, checked if it is replaced with your driver and tested the network performance for a few days. But the problem remains, unfortunately.

So, basically we're on square one again, and I'm even thinking of replacing the dongle as [yet another](https://github.com/lwfinger/rtl8188eu/issues/451#issuecomment-1841043821) radical option...

Thank you for trying to help anyway!

dubhater, 2023-12-12:
There is a newer version of this driver:
https://github.com/aircrack-ng/rtl8188eus/

and even newer still:
https://github.com/ivanovborislav/rtl8188eu/

You should try them both.

If you end up looking for another dongle, something based on MT7601U should have similar price and features, and better performance.

simurq, 2023-12-13:
> There is a newer version of this driver: https://github.com/aircrack-ng/rtl8188eus/
>
> and even newer still: https://github.com/ivanovborislav/rtl8188eu/
>
> You should try them both.
>
> If you end up [...]

Thank you very much! I'll try your suggestions asap.

PS: Just wondering if it somehow might be related to the regression problem reported [here](https://www.phoronix.com/news/Linux-6.6.6-Released)? I'm not a Linux expert, nor a network specialist at all, but the article seems to be sort of in sync with my issue, or is it?...

dubhater, 2023-12-13:
You can check easily by updating to kernel 6.6.6.