lwfinger/rtl8188eu #454: RTW: ap has changed, disconnect now

pbouchand, 2024-02-02:
Hello,

First, thanks a lot for your work!

This is not really an issue, but I would really appreciate it if you could shed some light on the meaning of this kernel message:

```
RTW: start auth
RTW: auth success, start assoc
RTW: send eapol packet
RTW: send eapol packet
RTW: set pairwise key camid:4, addr:7c:16:89:fd:9c:50, kid:0, type:AES
RTW: set group key camid:5, addr:7c:16:89:fd:9c:50, kid:1, type:AES
RTW: ERROR rtw_check_bcn_info: cur beacon key
RTW: ap has changed, disconnect now

```
I've used your driver on a bunch of different APs, and it's the first time I've encountered this issue. I've had it for hours, almost disconnecting every 2 minutes, and suddenly, it works like a charm for one day now.

Any ideas on what was wrong on the AP side?

Thank you!

pbouchand, 2024-04-22:
I'm sorry to bother you again, but could you direct me to some documentation explaining what might cause the driver to determine that the 'AP has changed, disconnect now' ?

Thank you

lwfinger, 2024-04-22:
The only documentation is in the source files. In routine rtw_check_bcn_info() in file core/rtw_wlan_util.c, any of the places where the routine does a 'goto _mismatch' is a candidate. I suggest that you add a 'pr_info()' statement before such goto statements to indicate which one is active.