lwfinger/rtl8188eu #456: hostapd not able to start for r8188 chip

awais11010, 2024-03-07:
Hello @lwfinger,

I am on kernel version 5.10 and trying to use the in-kernel staging driver rtl8188eu to get a dongle working which I know is RTL8188EU. I've provided firmware under /lib/firmware/rtlwifi and the driver does not complain about it. I then took the hostapd from this repo and tried running it with the configuration provided at: https://github.com/lwfinger/rtl8188eu/blob/master/rtl_hostapd.conf

This is what I get:
```
random: Trying to read entropy from /dev/random
Configuration file: new2.conf
drv->ifindex=5
l2_sock_recv==l2_sock_xmit=0x0xa11a00
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=52 freq=5260 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=56 freq=5280 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=60 freq=5300 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=64 freq=5320 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=100 freq=5500 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=104 freq=5520 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=108 freq=5540 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=112 freq=5560 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=116 freq=5580 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=120 freq=5600 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=124 freq=5620 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=128 freq=5640 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=132 freq=5660 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=136 freq=5680 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=140 freq=5700 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=149 freq=5745 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=153 freq=5765 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=157 freq=5785 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=161 freq=5805 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=165 freq=5825 MHz max_tx_power=0 dBm
Frequency 2437 (primary) not allowed for AP mode, flags: 0x0
Primary frequency not allowed
wlan0: IEEE 802.11 Configured channel (6) or frequency (2437) (secondary_channel=0) not found from the channel list of the current mode (1) IEEE 802.11g
wlan0: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: Unable to setup interface.
hostapd_interface_deinit_free(0xa10620)
hostapd_interface_deinit_free: num_bss=1 conf->num_bss=1
hostapd_interface_deinit(0xa10620)
wlan0: interface state DISABLED->DISABLED
hostapd_bss_deinit: deinit bss wlan0
wlan0: Flushing old station entries
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
wlan0: Could not connect to kernel driver
wlan0: Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=3
wlan0: AP-DISABLED 
hostapd_cleanup(hapd=0xa112b8 (wlan0))
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
hostapd_interface_deinit_free: driver=0x58ae90 drv_priv=0xa11950 -> hapd_deinit
hostapd_interface_free(0xa10620)
hostapd_interface_free: free hapd 0xa112b8
hostapd_cleanup_iface(0xa10620)
hostapd_cleanup_iface_partial(0xa10620)
hostapd_cleanup_iface: free iface=0xa10620
ELOOP: remaining socket: sock=6 eloop_data=0xa11aa8 user_data=(nil) handler=0x521909
```

Also looking at the code and the current error I am seeing it seems like the hostapd driver for rtw provided here should ideally set
```
modes[1].channels[i].allowed_bw = HOSTAPD_CHAN_WIDTH_20;
```
because the primary channel allowance check is failing here: https://github.com/lwfinger/rtl8188eu/blob/master/hostapd-2.9/src/common/hw_features_common.c#L624 or maybe I am not following this correctly. Can you please point out what might be wrong here and provide some suggestions? Just wondering how it worked for others because it definitely did :|

lwfinger, 2024-03-07:
All your channels are set with 0 TX power, i.e. they cannot transmit anything. How are you setting your regulatory database - regd, or something else?

awais11010, 2024-03-08:
@lwfinger thanks for giving this attention :)

I have wireless-regdb installed and related kernel configurations enabled.
```
CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=y
CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=y
```

However, I do see the following in dmesg
```
dmesg | grep -i regulatory
cfg80211: Loading compiled-in X.509 certificates for regulatory database
platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
platform regulatory.0: Falling back to sysfs fallback for: regulatory.db
R8188EU: EEPROMRegulatory = 0x0
cfg80211: failed to load regulatory.db
```

Although the regdb loading is showing failure during load here, the r8188eu is reporting global domain (0x0) from eeprom so that should ideally work anywhere, right? i.e. the regdb failure can be ignored IIUC.

awais11010, 2024-03-08:
I was successfully able to use the driver available here in branch v5.2.2 and an unmodified hostapd-2.10 (driver=nl80211).

closed 2024-03-08 by awais11010