lwfinger/rtl8188eu #458: RIP to Larry Finger

w568w, 2024-06-24:
According to [the linux-wireless mailing list](https://lore.kernel.org/linux-wireless/ds6wc3svkyre4p2rwg4f76o67xndazaeoyjwblhuzichscyxoz@5ttdvbymxr55/T/#mfd2f4928e0e013c10375ca766035c1385e18f8e0), @lwfinger a.k.a. Larry Finger, passed away on June 21st.

Still remember how hard the old days making my USB Wifi dongles work on Linux. Really wonderful work. My sincere condolences.

ihipop, 2024-06-25:
RIP :(

F1nch1330, 2024-06-25:
RIP

vitor-alves, 2024-06-27:
Rest In Peace. Finger's work on those wireless firmwares has helped me multiple times and will probably continue to help others for years to come.

Thank you for everything. RIP.

kimocoder, 2024-06-27:
Rest in peace dear friend 🥇   not enough words...

SkyfallWasTaken, 2024-08-15:
Rest in peace :(

freemandealer, 2024-09-12:
RIP Larry, and thanks for your contribution!

haobarry, 2024-09-29:
oh no！RIP

57ar7up-3, 2024-10-06:
RIP. This work will not be meaningless, USB-sticks are alive

ersingencturk, 2024-11-15:
rip dude, your code still lives