lwfinger/rtl8188eu #462: Does TP LINK USB WIFI DONGLE support 150Mb/s bitrate for wifi-hotspot ?

iAboothahir, 2025-02-04:
First **Thanks** for the awesome driver that works well with **TL-WN725N** 150Mbps Wireless Nano usb adapter

I am using https://github.com/lwfinger/rtl8188eu driver with branch _v5.2.2.4_ which working fine TL-WN725N adapter especially for creating wifi-hotspot (already blacklisted r8188eu module which does not work wifi-hotspot properly )
but i am only getting Bitrate 72Mb/s on Android phone and **iwconfig** shows **Bit Rate:72.2 Mb/s**
And i need to know this Dongle is limited 72Mb/s Bitrate for ap mode or driver lack support for it ?
and how to change bitrate to its maximum
### $ uname -r

> 5.19.0-41-generic

### $ cat /etc/os-release 
```
PRETTY_NAME="Ubuntu 22.10"
NAME="Ubuntu"
VERSION_ID="22.10"
VERSION="22.10 (Kinetic Kudu)"
VERSION_CODENAME=kinetic
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=kinetic
LOGO=ubuntu-logo
```

### $ sudo iw phy
```
Wiphy` phy0
	wiphy index: 0
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1962
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 476 Mbps
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412.0 MHz [1] (20.0 dBm)
			* 2417.0 MHz [2] (20.0 dBm)
			* 2422.0 MHz [3] (20.0 dBm)
			* 2427.0 MHz [4] (20.0 dBm)
			* 2432.0 MHz [5] (20.0 dBm)
			* 2437.0 MHz [6] (20.0 dBm)
			* 2442.0 MHz [7] (20.0 dBm)
			* 2447.0 MHz [8] (20.0 dBm)
			* 2452.0 MHz [9] (20.0 dBm)
			* 2457.0 MHz [10] (20.0 dBm)
			* 2462.0 MHz [11] (20.0 dBm)
			* 2467.0 MHz [12] (disabled)
			* 2472.0 MHz [13] (disabled)
			* 2484.0 MHz [14] (disabled)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
	interface combinations are not supported
	Device supports scan flush.
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	Supported extended features:
```
### iwconfig wlx54af97be8747 
```
wlx54af97be8747  IEEE 802.11bgn  ESSID:"<ssid-here>"  Nickname:"<WIFI@REALTEK>"
          Mode:Master  Frequency:2.412 GHz  Access Point: 54:AF:97:BE:87:47   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=1/100  Signal level=-99 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

```
### $ iwlist wlx54af97be8747 bitrate 
```
wlx54af97be8747  4 available bit-rates :
	  1 Mb/s
	  2 Mb/s
	  5.5 Mb/s
	  11 Mb/s
          Current Bit Rate:72.2 Mb/s

```

flitzcore, 2025-05-14:
Hi, I use the same dongle. Uniquely I tried it on windows and ubuntu in the same room doing speed test.  It gives me 24mbps for windows but sadly only 3 mbps on ubuntu. I also have  blacklisted r8188eu. Seems to me that the driver still doesn't fully support the hardware.

How about you? Have you found the optimal solution for it?

ashishs1, 2025-08-09:
I recently built and installed this on Debian, and the speed turned out to be around 0.7mbps. I was fairly away from my router, but my Android phone was showing 20mbps at the same place. So, there is definitely an issue with the driver. The default driver gives me ~2mbps, so I will stick to that.